espressif/arduino-esp32 #787: can't add BLE characteristic with same UUID

ManniAT, 2017-10-29:
I try to add two Humidity characteristics to a service - when adding the second one I get:
E (579) BLEService: << Attempt to add a characteristic but we already have one with this UUID

This should be possible when I add a descriptor to the characteristic.
https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.service.environmental_sensing.xml

Name: ES Measurement
Type: org.bluetooth.descriptor.es_measurement
Requirement: if_multiple_instances_of_same_characteristic_optional_otherwise

Requirement - I have a Meteo Station using NodeMCU esp32 which has values for indoor and outdoor.
So I have the same measurement two times - humidity indoor and humidity outdoor.
From my point of view adding more characteristics with the same UUID should be possible - as long as they provide a different ES Measurement description.

By the way - I just check the BLE_Notify sample.
This uses a lot of memory
Sketch uses 913158 bytes (69%) of program storage space. Maximum is 1310720 bytes.
Global variables use 63888 bytes (21%) of dynamic memory, leaving 231024 bytes for local variables. Maximum is 294912 bytes.
Other examples (WiFi or so) just use a fraction of memory (10-15%) - is this normal?

me-no-dev, 2017-10-30:
Please submit this issue here: https://github.com/nkolban/esp32-snippets/issues

BLE in this repository is pulled from there and @nkolban should be able to help you

ManniAT, 2017-11-08:
OK, made a post there
https://github.com/nkolban/esp32-snippets/issues/175

closed 2017-11-08 by ManniAT