espressif/arduino-esp32 #789: rpi install doesn't seem to install tooltchain

CosmicBandito69, 2017-10-29:
Please fill the info fields, it helps to get you faster support ;)

If you have a Guru Meditation Error, please decode it:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							DOIT ESP32 DEV KIT V1
Core Installation/update date:			?11/jul/2017?
IDE name:							Arduino IDE
Flash Frequency:					?40Mhz?
Upload Speed:						?115200?

### Description:
On a raspberry pi 3, installed arduino IDE, then followed instructions to install esp32 arduino core, which appeared to work.  However, the xtensa gcc binaries did not get installed to the proper location (if at all).  boards show up in the board manager, but attempts to compile sketches fail because the binaries can't be found, because they aren't there.  Everything works fine for esp8266.

There is no xtensa-esp32-elf directory in home/pi/Arduino/hardware/espressif/esp32/tools

### Sketch:
```cpp

//Change the code below by your sketch
uint64_t chipid;  

void setup() {
	Serial.begin(115200);
}

void loop() {
	chipid=ESP.getEfuseMac();//The chip ID is essentially its MAC address(length: 6 bytes).
	Serial.printf("ESP32 Chip ID = %04X",(uint16_t)(chipid>>32));//print High 2 bytes
	Serial.printf("%08X\n",(uint32_t)chipid);//print Low 4bytes.

	delay(3000);

}
```

### Debug Messages:
```
Error:
fork/exec /home/pi/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++: no such file or directory
Error compiling for board DOIT ESP32 DEVKIT V1.

```

me-no-dev, 2017-10-30:
we do not have the toolchain for RPI yet :) maybe you can build crosstool-ng yourself?

CosmicBandito69, 2017-10-30:
Gosh, it's so much easier when someone else does it :) .  Thank you for the response.  I'll give it a go.  I heard it takes 2.5 hours after starting.  It's that starting part that's going to take me a while.

CosmicBandito69, 2017-10-31:
update: After following instructions available online... At 85 minutes... Still going.  Follow instructions to disable parallel processing before "./ct-ng build build.1"  Will update.

CosmicBandito69, 2017-10-31:
Oh no, am I  building it twice?  It says it finished at 241min (!!!!) (running off usb stick) and seems to now be duplicating the same steps.   Crap.  I didn't want the build.1 argument, did I?  How stupid.  I don't see any binaries anywhere.  Sigh.  When it is built, what am I supposed to copy over to the esp32/tools directory?  It looks like I'm 20 min into another 241min build.  :(  Cannot find the binaries.

CosmicBandito69, 2017-10-31:
Success!  Well another 238 minutes later I changed the permissions, as per instructions, and then copied the xtensa-esp32-elf directory over to /home/pi/Arduino/hardware/espressif/esp32/tools/ using copy and paste in the file browsers.  And...  The sketch compiled.  Thank you.  If you change the .config file to restrict parallel processing, I don't think you need the second argument "build.1", so "./ct-ng build" would take half as long.

me-no-dev, 2017-10-31:
Hey :) glad you succeeded :) BTW want to share that toolchain?

CosmicBandito69, 2017-10-31:
How can I share it?  The zip file is 77.5 mb (just the xtensa-esp32-elf directory).  I seem to have a limit of 10mb on attachments.

me-no-dev, 2017-10-31:
maybe something like google drive?

CosmicBandito69, 2017-10-31:
[here](https://drive.google.com/drive/folders/0Bz7Qo_vFhiBmVFYyQlA3ZWxDZ0U?usp=sharing) should link you to my google drive with toolchain in 7z and zip form. unzip them in the tools directory.

CosmicBandito69, 2017-11-03:
Well, my usb stick crashed, so I set everything up on a sd card install, like I was installing the esp32 arduino core on a normal linux system, and used the 7z (or zip) file from link above.  The rpi binaries don't download, so instead of running get.py to download them, unzip the files in the /home/pi/Arduino/hardware/espressif/esp32/tools directory.  It makes a directory, "xtensa-esp32-elf"  Sketches compile now.

Today, I rebuilt the toolchain also (In case someone wanted to do it themselves).  I wanted to see if it would work without changing the .config file to turn off parallel builds, but it indeed crashed with a segmentation fault.  Changing the one line in the .config file allowed it to finish.

Jcspoza, 2018-12-29:
Thanks @CosmicBandito69 for the effort . It works on my RPI 3B  pretty well (after a 1st try with a hook!!). Speed in compilation is acceptable, tested with short examples like ChipID or WIFIscan on a DOIT ESP32 DEV KIT V1.

CosmicBandito69, 2018-12-29:
I’m glad it helped.  Too bad it doesn’t just compile, and they could auto generate binaries that are up to date.  I hope you are not using the binaries I shared.  They are old, and likely have bugs.  Try to compile newer binaries.  I have the same pi and esp32.  It takes hours to compile (10?) the arduino binaries.
Cheers,
— CosmicBandito69

yoursunny, 2019-08-01:
Toolchain for ARMv7 and ARMv8 would be useful for me as well. This issue should remain open.

atanisoft, 2019-08-01:
@me-no-dev can we get the "enhancement" or "to be implemented" tag on this issue?

igrr, 2022-04-05:
This seems to be fixed by https://github.com/espressif/arduino-esp32/commit/5ad468f9dcb5f9cc7249a4fb987aead4f1d9b136 and https://github.com/espressif/arduino-esp32/pull/4117. If you find an issue with the latest 2.0.3-rc1 release, please open a new issue.

closed 2022-04-05 by igrr