espressif/arduino-esp32 #790: How to use both VSPI & HSPI SPI buses simultaneously?

neilpanchal, 2017-10-30:
### Hardware:
Board:							ESP32-DevKit-C
Core Installation/update date:			Latest
IDE name:						IDF Component

### Description:
Trying to use both VPSI & HSPI busses simultaneously to drive two OLED displays. By default, `SPIBegin()` works with VSPI pinouts. 

### Sketch:
I've tried the suggested code here: https://github.com/olikraus/u8g2/issues/377#issuecomment-335052062

Is the usage of HSPI + VSPI through arduino API possible currently? If so, what is the best way to achieve this?

me-no-dev, 2017-10-30:
```cpp
SPIClass SPI2(HSPI);

void setup(){
  SPI2.begin(...);
  //... your code
}
```

In order to use the OLED with the other SPI you might need to alter the lib to take ```SPIClass &spi`` as argument and use that instead of ```SPI```

closed 2017-10-30 by me-no-dev

neilpanchal, 2017-11-05:
@me-no-dev 

I tried replacing all references to SPI and replaced them to SPI2 in this file:
https://github.com/olikraus/U8g2_Arduino/blob/master/src/U8x8lib.cpp#L394

That's the only place SPIClass object is called. I still have the OLED display connected to 
```
SDA = IO 23
SCLK = IO 18
D/C = IO 21
RST = IO 22
CS = IO 5
```
and oled works fine. It didn't switch to HSPI.

theMubashir919, 2020-11-01:
> 
> 
> @me-no-dev
> 
> That's the only place SPIClass object is called. I still have the OLED display connected to the following pins. The code compiles fine and I made sure the SPI2 is instantiated in SPI.cpp in the ESP32-Arduino Library.

Hello @me-no-dev ,
I am having some difficulty on how to instantiate SPI2 in SPI.cpp.
Can you please explain or give an example?
Much appreciated.
Thanks

chegewara, 2020-11-01:
https://github.com/espressif/arduino-esp32/tree/master/libraries/SPI/examples/SPI_Multiple_Buses

theMubashir919, 2020-11-02:
thanks. My issue is resolved