espressif/arduino-esp32 #792: Mac OSX intallation

totovr, 2017-10-30:
I'm trying to set it up in Arduino 1.8.1 version but I got the next error in the installation step

Traceback (most recent call last):
  File "get.py", line 27, in <module>
    current_dir = os.path.dirname(os.path.realpath(unicode(__file__)))
NameError: name 'unicode' is not defined

----------------------------- Remove above -----------------------------

### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			30/oct/2017
IDE name:							Arduino IDE 1.8.1

### Description:
I'm trying to set it up in Arduino 1.8.1 version but I got the next error in the installation step

me-no-dev, 2017-10-30:
interesting... what version OSX?

totovr, 2017-10-30:
Sorry I forgot it, I'm using Sierra

totovr, 2017-10-30:
[image]

me-no-dev, 2017-10-30:
can you run ```python --version``` please? Same OSX here. BTW no need to run the whole command again, just cd into esp32/tools and run ```python get.py``` from there

totovr, 2017-10-30:
Im currently using version 3.6.2.

[image]

me-no-dev, 2017-10-30:
ahh :) try python2 ;) I will look into p3 later

totovr, 2017-10-30:
Done, I return to version 2.7 and is working normally.
Thank you very much!

[image]

I think that Python 3.x doesn't recognize the route of the path.

closed 2017-10-30 by totovr

alustig3, 2017-12-03:
replacing "unicode" with "str", fixes the problem.

line 27 of get.py before:
`current_dir = os.path.dirname(os.path.realpath(unicode(__file__)))
`
after 
`current_dir = os.path.dirname(os.path.realpath(str(__file__)))`