espressif/arduino-esp32 #793: Invalid Library Found

totovr, 2017-10-30:
Hello again, I got the next problem when I was trying to upload a program in the board, I checked the file and it doesnt contain anything.

### Hardware:
Board:							        ESP32 Dev Module
Core Installation/update date:			30/oct/2017
IDE name:							Arduino IDE 1.8.1
Flash Frequency:					        40Mhz
Upload Speed:						115200
[image]
[image]

### Description:
Describe your problem here

### Sketch:
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

totovr, 2017-10-30:
[image]
I added the files manually from your master branch and is working now, I dont know why this library was empty
[image]

me-no-dev, 2017-10-30:
you did not run ```git submodule update --init --recursive``` inside ```hardware/espressif/esp32```

palomarcoder, 2017-11-02:
I get that error as well.  I tried running the command you suggested, and get this error
```
$ git submodule update --init --recursive
fatal: Not a git repository (or any of the parent directories): .git
```

me-no-dev, 2017-11-06:
where did you run this command?

totovr, 2017-11-06:
Youncanncopy directly the files from the folder of master and paste it in BLE folder of your computer

palomarcoder, 2017-11-06:
@me-no-dev  I ran it from my esp32 directory from a git command line window.

```
$ pwd
/c/Users/David/documents/arduino/hardware/espressif/esp32

David@Davidx4 MINGW64 ~/documents/arduino/hardware/espressif/esp32
$ git submodule update --init --recursive
fatal: Not a git repository (or any of the parent directories): .git

David@Davidx4 MINGW64 ~/documents/arduino/hardware/espressif/esp32
$

```

me-no-dev, 2017-11-06:
did you download the repository as zip file?

totovr, 2017-11-07:
https://github.com/totovr/Arduino/blob/master/esp32.md

palomarcoder, 2017-12-27:
I finally figured this out @me-no-dev .  in order for the git submodule command to work, you have to use the recursive option on the original clone command.  This is NOT correct in the windows installation guide.

git clone --recursive -j8 https://github.com/espressif/arduino-esp32.git

imtheman2000, 2018-02-28:
I you have problems with the installation of Wemos Lolin32 ESP32 development board I can send you the files. The problem with BLE i have solved so that I have deleted the folder.

svenne10, 2018-04-16:
Hi ghost, 
I just started to switch from Arduino to ESP32 and have the same problem with the empty BLE folder on my ios. The git command works also the recursive one, but still I get the error with the invalid library BLE. I'm using the Demos Lolin32 ESP32. Maybe you can help me solving the problem?