espressif/arduino-esp32 #794: ESP stop Working UART

lBenji, 2017-10-30:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			11/jul/2017
IDE name:							Arduino IDE
Flash Frequency:					40Mhz
Upload Speed:						115200

### Description:
Sometimes if I reset the esp the last Console Output is "1". But the Serial1 Device is sending data all time.

If I reset the ESP32 5 times. the ESP is working 2 times correctly.

### Sketch:
```cpp
HardwareSerial Serial1(1);
#define SERIAL1_RXPIN 16
#define SERIAL1_TXPIN 17

void setup() {
  Serial1.begin(115200, SERIAL_8N1, SERIAL1_RXPIN, SERIAL1_TXPIN); //Serial Display
  Serial.begin(115200);

  delay(1000);

  Serial.write("1");
  
  delay(1000);

  Serial.write("2");
}

void loop() {
  Serial.println(5.1);
  Serial1.println(5.1);
  
  if (Serial1.available()) {      // If anything comes in Serial (USB),
    Serial.write(Serial1.read());   // read it and send it out Serial1 (pins 0 & 1)
  }

  delay(1000);
}
```

### Debug Messages:
```
no
```

lBenji, 2017-11-06:
I had the same issue as #645

closed 2017-11-06 by lBenji

copercini, 2017-11-06:
This was fixed in https://github.com/espressif/arduino-esp32/commit/e2bd93c0920346cfbe3c792e82b0b3f99f4354a0
Just update your core version and everything should work fine =)

lBenji, 2017-11-06:
Problem fixed.

@copercini yes I have seen it before, that the Issue is fixed