espressif/arduino-esp32 #796: DEEPSLEEP_RESET gets stuck after a few resets

fiskfan, 2017-10-30:
### Hardware:
Board:							ESP32 Dev Module - Sparkfun ESP32 Thing
Core Installation/update date:			30/oct/2017
IDE name:							Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:						921600

### Description:
The board does not correctly restart after deep sleep. Im using the builtin example sketch "TimerWakeUp" without modifications and no other hardware attached and i get random results on the powerup after sleep.

The board usually start successfully a few times after deepsleep, but then it failes to run the actual sketch after startup and a manual reset is needed.

The debug information below shows that the device starts after deepsleep and outputs "Boot number: x", but then in the last boot it doesn't output anything from the sketch and does not return to sleep. A manual press of the reset button restarts the device and the it runs for a few more deepsleep cycles before it gets stuck again.

Any one else seen this problem?

### Sketch:
builtin "TimerWakeUp" example sketch

### Debug Messages:
```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:11904
entry 0x40078a3c
Boot number: 1
Wakeup was not caused by deep sleep
Setup ESP32 to sleep for every 5 Seconds
Going to sleep now
ets Jun  8 2016 00:22:57

rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:11904
entry 0x40078a3c
Boot number: 2
Wakeup caused by timer
Setup ESP32 to sleep for every 5 Seconds
Going to sleep now
ets Jun  8 2016 00:22:57

rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:11904
entry 0x40078a3c

```

iot-man, 2017-10-31:
I'm having the same problem. My sketch goes to sleep for 60 seconds after taking temp/humidity readings etc via I2C. It used to work ok before system upgraded to most recent commit. First two restarts work ok but always hangs on third restart.

**Debug messages**

`rst:0x5` (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:11904
entry 0x40078a3c
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:11904
entry 0x40078a3c
Boot number: 1
Setup ESP32 to sleep for every 60 Seconds

Connecting to SKYE3E0B
MQTT connecting ...connected
19.74
1015.97
Going to sleep now
ets Jun  8 2016 00:22:57

rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:11904
entry 0x40078a3c
Boot number: 2
Setup ESP32 to sleep for every 60 Seconds

Connecting to SKYE3E0B
MQTT connecting ...connected
19.73
1015.91
Going to sleep now
ets Jun  8 2016 00:22:57

rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:11904
entry 0x40078a3c

weizenbock, 2017-11-01:
I'm having a "similar" problem.  In my case it happened with ESP.restart().
It gets successfully restarted a few times, but after several cycles it hangs like below.

```
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:11904
entry 0x40078a3c
```

copercini, 2017-11-01:
Same here, after commit https://github.com/espressif/arduino-esp32/commit/55289a45af1e7898ca0b8da037d92a83baaf0083 this problems starts

degehi, 2017-11-03:
Same here, hangs after 2-3 reboot, just after the 'entry' log line

etimou, 2017-11-04:
Exactly the same problem for me as well with an unmodified "TimerWakeUp" example sketch.

me-no-dev, 2017-11-06:
I would suggest an issue to be opened in IDF (if one does not already exist). It is surely something in that update.

mikey60, 2017-11-06:
I reset the core version back to  https://github.com/espressif/arduino-esp32/tree/95123681d5340defa24472e441758fba88e3a34f  as was suggested in Issue #801 and my sketch with a 20 minute deep sleep that would hang after an hour or so (as described above) now has been working properly for 2 days.

cristianhumelnicu, 2017-11-08:
Any patch for deepsleep issue ???

etimou, 2017-11-09:
I can do the test with older version as well, but I am not very familiar
with git.
I guess the following commands on debian must be adapted:

git clone https://github.com/espressif/arduino-esp32.git esp32

git submodule update --init --recursive

What is the way to follow?

degehi, 2017-11-09:
@etimou: what I did: just apply a checkout after de clone operation, with a version number before 55289a4

on the esp32 directory: git checkout 9512368

running without reboot hangs with this release for me.

etimou, 2017-11-10:
thank you! it works perfectly with this version.

cristianhumelnicu, 2017-11-16:
@etimou, @degehi 
Could you tell me what is the maximum time you've been able to keep ESP32 in deepsleep? All I've done is a deepsleep time of 30 minutes ... if it's more than 30 minutes it's not waking up again ffrom deepsleep...

etimou, 2017-11-16:
I've run the deep sleep example for more a little bit more than one hour
with no problem. It think the counter was around 800.

cristianhumelnicu, 2017-11-16:
@etimou Do you use the deepsleep example from Arduino or other scatch ?
Thk!

mikey60, 2017-11-16:
I am using my own program based on the Arduino deepsleep example an it sleeps for 20 minutes and then sends sensor data to ThingSpeak.  It has run for several days.  When I get a chance I will change the deepsleep to one hour and see if it continues to work.

iotmodules, 2017-11-16:
I'm using the Sparkfun esp32 thing and am having similar issues however its not just sleep, i'm getting hanged reboots when pressing the reset button too, not just in the TimerWakeUp sketch. 

I also find that I get about 5 reboots before a hang when RTC_DATA_ATTR int bootCount = 0; is intialised however if I comment it out and all references to it then I get about 30-40 reboots before a hang. 

Using latest  with 80a7d60 commited 6 days ago

CelliesProjects, 2017-11-16:
Using the MHET Live D1 mini-kit.
Can confirm the problem with hanging reboots.

etimou, 2017-11-18:
I am using the deep sleep example from Arduino, not modified.

euquiq, 2017-11-21:
I'm having the same problem (with latest Arduino - idf):

 I set up a deep-sleep / wake timer for N seconds (5, 2 or even 1 second) and after a low number of wake-ups (2 to 6 times of normal waking up), ESP32 hangs.

So after reading the above posts, I am waiting for the bug to be crushed (on IDF, it seems).

Regards,

Enrique.

copercini, 2017-11-21:
There is a commit that fixes it on IDF https://github.com/espressif/esp-idf/commit/abacf8d2a0347a4bf95315cdb227e9b8b1c93794 seems like just wait @me-no-dev bump the version for Arduino IDE  =)

me-no-dev, 2017-11-21:
need to check if @nkolban has updated the BLE lib and find some time to put a new release.

nkolban, 2017-11-21:
That is job number 1 starting now

cristianhumelnicu, 2017-11-25:
@me-no-dev  Could you estimate when any fix update for deepsleep and reset  issue will be availble ?

me-no-dev, 2017-11-25:
I have updated Arduino. Can you pull the latest changes and give it a try?

cristianhumelnicu, 2017-11-25:
@me-no-dev I already did .. using deepsleep example with timer wake-up and put to be 60 min in deep sleep . Until now have 12 deepsleep cycles without problems for 1h deepsleep

But i still have one problem .. I put another 3 Wroom32 (Rev.1) to sleep 2h , 6h and 8h ant all of them standing in sleep about 45 minutes and after that wake up . All i've done with success is 1h sleep time with first Wroom32 

Could u guide me what is supouse to do for long time deepsleep i mean 8h - 24h ?

Thnak you very much!

me-no-dev, 2017-11-28:
@cristianhumelnicu please post a minimal sketch that shows that issue and I will run it here on a few boards to see the result

cristianhumelnicu, 2017-11-28:
@me-no-dev Is the standard deepsleep example and i use Wroom32 Silicon Rev.1 chipset

#define uS_TO_S_FACTOR 1000000  
#define TIME_TO_SLEEP  7200        

RTC_DATA_ATTR int bootCount = 0;

void setup(){
  Serial.begin(115200);
  delay(200); 

  ++bootCount;
  Serial.println("Boot number: " + String(bootCount));
 
  esp_sleep_enable_timer_wakeup(TIME_TO_SLEEP * uS_TO_S_FACTOR);
  Serial.println("Setup ESP32 to sleep for every " + String(TIME_TO_SLEEP) + " Seconds");

 Serial.println("Going to sleep now");
  esp_deep_sleep_start();  
}
void loop(){ 
}

After last update is working for 1h deep sleep, but if I set TIME_TO_SLEEP>2h , the microconttroler exit from deep sleep in 45min

Until last arduino-esp update  never did exit from deepsleep  for TIME_TO_SLEEP > 30 min 

I did try to define  TIME_TO_SLEEP as unsigned long with same results

Thank you !

me-no-dev, 2017-11-28:
OH!!!! I got it ;)
```(TIME_TO_SLEEP * uS_TO_S_FACTOR)``` this will default to 32 bit integer as opposed to 64bit one (which is what the wakeup timer expects), so in essence you overflow the 32bit number.
Try ```((uint64_t)(TIME_TO_SLEEP) * uS_TO_S_FACTOR)``` instead

euquiq, 2017-11-28:
Hi there ! I got a slightly different issue: I got a wakeup timer, which at this time is set for 3 seconds. When the ESP32 awakes, it produces two beeps in a piezoelectric, then goes to sleep.

After about 2 - 3 hours of working about OK, timing goes wrong and it starts to wake up each 1 second by itself. 

At first I had a screen connected thru SPI to the ESP32 showing battery level when awake, but the screen started to be unresponsive at the time of the "acceleration". So I took it off the system just in case it was something related to it. 

It seems to be not. Bottom line, after some time, timer awakening seems to default to another timing. 

Right now I am in the process of creating a minimal example code and reproduce the problem with it. I will post it ASAP.

Regards,

Enrique.

mikey60, 2017-11-28:
I apologize for the late response.  I actually cloned the esp32-arduino
version 12 hours ago and my sketch worked just fine overnight. It has
executed flawlessly for the past 12 hours or about 36 twenty minute
deep-sleep periods.  Also this version of esp32-arduino seems to have
corrected the Arduino OTA issue I was experiencing with the ESP32 Radio
sketch.   Thank you for all the hard work.

fiskfan, 2017-11-28:
I pulled the latest release now as well and it seems as if the device starts correctly after deepsleep now.

I would also like to say that it's boot is better overall now. Previously it could also fail to load the sketch after i pressed the reset button, but now it seems as if it always successfully starts the sketch.

cristianhumelnicu, 2017-11-28:
Well i confirm to  .. the deepsleep and reset is running normal after latest arduino-esp32 update

I think in this moment we can close the issue !

@me-no-dev  Thank you for all !

euquiq, 2017-11-28:
@cristianhumelnicu not so fast ... ! I am trying this code:

```
#include <esp_deep_sleep.h>

//----Defines 
#define uS_TO_S_FACTOR 1000000  /* Conversion factor for micro seconds to seconds */
#define SLEEP  3 //3 seconds
#define pinBeeper 32

//RTC_DATA_ATTR int bootCount = 0;

void setup()
{
	ledcSetup(0, 2000, 8); //for tone generation
	ledcAttachPin(pinBeeper, 0); //Beeper GPIO to channel 0
	tone(5000, 50); //Happy tone
	tone(6000, 50);
	esp_sleep_enable_timer_wakeup(SLEEP * uS_TO_S_FACTOR);
	esp_deep_sleep_start();
}

void happytone() {
	tone(5000, 50); 
	tone(6000, 50); 
}

//Genero un tono
void tone(int freq, int dur) {
	ledcWrite(0, 127); 
	ledcWriteTone(0, freq);
	delay(dur);
	ledcWrite(0, 0);
}

void loop()
{
	//This is not going to be called
}
```

It starts OK, the beeping occurs on the correct timing. BUT after a couple of hours I cronometrize it again, and instead of 3 seconds, on this opportunity the beeps are spaced about 4 seconds and a half. 

Today in the morning, a similar test, made the beeping go from the programmed 3 seconds down into 1 second. 

I would say that something WEIRD is going on ... maybe it is my particular ESP32 board, so I will try now with a newer board, which also is version 1 silicon inside. (The one I am trying now is Version 0).

euquiq, 2017-11-28:
Maybe it is better for me to open another issue ?

cristianhumelnicu, 2017-11-29:
@euquiq 

#warning esp_deep_sleep.h will be deprecated in the next release. Use esp_sleep.h instead.

euquiq, 2017-11-29:
@cristianhumelnicu thanks for the EYESUP on the deprecation for esp_deep_sleep.h ...

On the delay(3000) issue, this is part of a burn - in test for a system that operates under solar panel / supercapacitors system so I really need to put the ESP32 to sleep.

 I find weird that sleep time may go faster or slower than the programmed value. Sometimes for a significant %. I am at this time performing tests on different sleep times, to see it's behavior.

 I do not think it is "Boot time" related, not at differences drifting from 3 seconds down to 1 second or 4.x seconds, when WIFI is off, etc. 

But if you can throw any idea or light into it, I will try to understand / process it into my tests.

Thank you, Enrique

degehi, 2017-12-19:
I tried the last revision, like @mikey60 @fiskfan and @cristianhumelnicu, those last days and I still have reboot problems. It seems better than before (I use ESP32 with a 1 hour deepsleep and it can run a day without any problem) but finally, sooner or later, it will hang again on wake-up. I tried less than 1 hour deepsleep but same results

If I come back with version 9512368, it works fine even after several days.

What I use:

```
#define uS_TO_S_FACTOR 1000000  /* Conversion factor for micro seconds to seconds */
#define TIME_TO_SLEEP  3600     /* Time ESP32 will go to sleep (in seconds) */
...
// init deep sleep
esp_sleep_enable_timer_wakeup((uint64_t)(TIME_TO_SLEEP) * uS_TO_S_FACTOR);
...
esp_deep_sleep_start();
```
Hardware: Sparkfun ESP32 Thing

Nobody has the same problem than I?

mvadu, 2017-12-25:
I can see the same problem on a Adafruit Huzzah32 feather board. I have a simple modified deep sleep sketch which logs the booth count to a influxdb (which is plotted with grafana). Sleep duration is 120 seconds. As you can see below it randomly gets rest to 0. Max I have seen it grow is 700 times (close to a day). 

[image]

cloudiaworld, 2018-02-08:
Any news regarding this issue? I continue having the same issue on some boards; namely that the esp32 stops waking-up correctly after some (non-deterministic) cycles of sleep/wake up.  I get, in addition to the DEEPSLEEP_RESET a TG0WDT_SYS_RESET.  All the values I have stored as RTC_DATA_ATTR are gone after the problem occurs. 

 Any hints? I am using a ESP-WROOM-32 module. 

Core Installation/update date:	08/02/2018
IDE name:	Arduino IDE 1.8.5
Flash Frequency:	40Mhz / also happens at 80MHz
Upload Speed:	115200

rst:0x5 (DEEPSLEEP_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13076
entry 0x40078a58

misan, 2018-08-01:
Same problem here. It works for a while but eventually does not wake up. Or more exactly, it hangs on wake-up, as the fact the chip is hot makes me think it is no longer in deep-sleep. 

For low-power IoT applications (as it is mine) this issue is a deal-breaker.

wagnerfl85, 2018-08-06:
Same problem here, i also use a ESP32 Dev Module and it get stuck after some time. My sketch runs as follows:

1. wakeup
2. connect to wifi
3. get data from BME280
4. send data via mqtt to openhab
5. go to deep_sleep
6. wakeup after 5 Minutes

The last run was about 30-35 hours without a problem but then stucks again. After a button reset it doe work again.

gianky69, 2018-08-27:
I have a similar sketch like @wagnerfl85 (ESP32, wifi, BME data...) . I send data to Home Assistant via MQTT.
Using latest Arduino build the  ESP32 battery module hangs after 2-3 hours!

mkihr, 2018-08-28:
Same issue here. With both adafruit feather and WEMOS ESP32. Hangup when wakeup from deep sleep.  I only tried with 1 to 10 minutes. After some wakeup, it stuck. Can be after 10 or after 700.
As I see here this is an ongoing issue for long time. Will there be a solution?

axelisart, 2018-09-27:
Hi everybody. I don't know if it can help but here are some observations again : I tried with a raw esp32 (just the chip, not mount in any dev board) and I am near 1 month without any stall. It represent 3786 wakes as it wakes every 10mn. The esp32 is just connected to an SHT31 sensor to get temp/hdty. It is powered through a tsp-03 step down buck (220v to 3,3v). I have lot of different ESP32 and ESP8266 Chinese dev boards working in parallel in the same wake up frequency and every of them stalled lot of times during the same period. So for the moment my conclusion is that it is maybe not due to the ESP chip itself but more to the board it is attached. So if you absolutely need to run a project with ESP, look a my workaround.  I used a Board Plate Pinboard Convertor to solder the esp32 (50 cents) and an ESP-WROOM32 (3$). I just soldered required pins for my sht sensor and tx/rx to program it. It is just a bit difficult to solder chip pin to the board but it is feasible. I will keep posting here if I see any stall for my "naked" ESP32.

misan, 2018-09-27:
That is a very good point @axelisart I am going to monitor what happens to the power supply during the wake-up from sleep. It may well be that many of the existing boards are not properly design to handle well this sudden load change. Thanks for reporting it, as many of us are trying to figure out why we could not get a reliable result when apparently others have no complaints.

misan, 2018-09-27:
No wonder there is wake-up trouble, supply rail looks like sh*t on restart. Next stop is to get a decent 3v3 supply.

[image]

stickbreaker, 2018-09-27:
@misan I see those droops (5 I think) are off scale, How far down does they go?

Chuck.

misan, 2018-09-27:
It appears they go almost 1Volt down.

[image]

stickbreaker, 2018-09-27:
@misan Wow, that regulator is useless, If I am reading you scope correctly, 800mV droop, 800mV overshoot.  So the 3.3V drops down to 2.5V.  A 24% reduction! and then jumps to 4.1V on the rebound.  That is ridiculous.

Chuck.

stickbreaker, 2018-09-27:
@misan It's almost like you have an inductor on the power line. 

Try changing your USB cable,  maybe you have a one with a builtin choke.

Chuck.

misan, 2018-09-27:
@stickbreaker Thanks Chuck, similar results powering it from Raspberry PI 2A power brick. AMS117 LDO. I'll try a larger capacitor after the regulator.

stickbreaker, 2018-09-27:
The specs for a AMS1117 show only a 25mV variance from 0 to 800mA load, but that is with input held at 4.75V.
[image]

You are definitely going to have to beef up that power circuit.

Chuck.

stickbreaker, 2018-09-27:
@misan 
It could even be board layout issues. Sometimes there is no consideration for instantaneous current flow when traces are laid out.  a 10mil trace can't handle much power!

Chuck.

misan, 2018-09-27:
@stickbreaker good point, maybe a tricker wire is all that is needed. I will keep on searching ,thanks a lot.

blackbrayn, 2018-10-03:
Having similar issues trying to deep sleep a wemos lolin32 v1 clone board, after a few cycles of deep sleep it just prints dots in the serial , left the esp connected on the laptop for serial monitoring.
Code also doesn't run , i see no connections to the server it sends the data.

chegewara, 2018-10-03:
This week ive been doing test with this example and it worked for 30-40 cycles with no issue (short test): https://github.com/nkolban/ESP32_BLE_Arduino/blob/7951347ed68313d75c367e1f2cce763cb56d1eb2/examples/BLE_iBeacon/BLE_iBeacon.ino

misan, 2018-10-03:
@blackbrayn Not sure about your code, but many programs will print dots while WiFi access is being secured. If that is the case, it may well be printing forever is access is never granted for any reason.

That was one of the problems I experienced that looked like a fault of the deep sleep but it was actually an endless loop in my code once Wifi authentication was not successful. After a closer consideration, it was user error (me) and not the ESP fault. But the fact that data logging stopped at a given point looked like the chip was at fault.

That said, I can easily trigger a fault condition in the TimerWakeUp example if I commented out the delay in

`void setup(){

Serial.begin(115200);

delay(1000); //Take some time to open up the Serial Monitor`

It is my understanding that delay helps getting things settled, as it will fail after a while if commented out. Also, some boards will change the serial port every time I upload new code (while others do not do that).

blackbrayn, 2018-10-03:
@misan Thx for the tips , i'll increase the time to see if it helps(thinking it's maybe too agressive for the wireless).
Just noticed that running on battery seems to be more stable for now (maybe the voltage doesn't drop that much) , i'll do some more tests on the electric side once i get some time , maybe solder a capacitor right to the pins of the board that conect directly to the esp module.
Running the battery thru the multimeter on the low amps side(100mA) causes instant problems (no boot) i supose , using the 10 amp range that has a beefy shunt and a lower resistance works fine but can't really measure deep sleep curent.

mvadu, 2018-10-03:
I always had it running on a lipo battery, and still had the problem. Not sure about the spoke dropping voltage resulting in a reset.

Regards,
Adarsha

sent from handheld, excuse typos.

blackbrayn, 2018-10-04:
Just ran a test on lipo battery with no other modifications(just a multimeter in series) to really see if it's more stable on my board , it just stopped after a while, it worked for only one hour and after that it started drawing 120mA all the time.
[image]

[59 more replies]

closed 2021-09-06 by stale[bot]