espressif/arduino-esp32 #798: Cannot obtain IP address through LAN8720 PHY Ethernet Module (ESP32)

James-sjec, 2017-10-31:
### Hardware:
Board:							?ESP32 Dev Module?
Core Installation/update date:			?11/jul/2017?
IDE name:							?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
Upload Speed:						?115200?

### Description:
Describe your problem here

```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```
#include <AWS_IOT.h>
#include <WiFi.h>
#include <ETH.h>

AWS_IOT hornbill;

char HOST_ADDRESS[]="a292implg79vjw.iot.us-east-2.amazonaws.com";
char CLIENT_ID[]= "client id";
char TOPIC_NAME[]= "/things/securekit/shadow/";

static bool connect_aws ;
static bool eth_connected ;

int tick=0,msgCount=0,msgReceived = 0;
char payload[512];
char rcvdPayload[512];

void mySubCallBackHandler (char *topicName, int payloadLen, char *payLoad)
{
    strncpy(rcvdPayload,payLoad,payloadLen);
    rcvdPayload[payloadLen] = 0;
    msgReceived = 1;
}

void WiFiEvent(WiFiEvent_t event)
{
  switch (event) {
    case SYSTEM_EVENT_ETH_START:
      Serial.println("ETH Started");
      //set eth hostname here
      ETH.setHostname("esp32-ethernet");
      break;
    case SYSTEM_EVENT_ETH_CONNECTED:
      Serial.println("ETH Connected");
      connect_aws  = true;
      break;
    case SYSTEM_EVENT_ETH_DISCONNECTED:
      Serial.println("ETH Disconnected");
      connect_aws = false;
      break;
    case SYSTEM_EVENT_ETH_STOP:
      Serial.println("ETH Stopped");
      //eth_connected = false;
      break;
    default:
      break;
  }
}

void setup()
{
    Serial.begin(115200);
    WiFi.onEvent(WiFiEvent);
    ETH.begin();

    Serial.println("Connecting");
    while(connect_aws != true)
    {
      Serial.print('.');
      delay(1000);
    }

     Serial.print("ETH MAC: ");
      Serial.print(ETH.macAddress());
      Serial.print(", IPv4: ");
      Serial.print(ETH.localIP());
      if (ETH.fullDuplex()) {
        Serial.print(", FULL_DUPLEX");
      }
      Serial.print(", ");
      Serial.print(ETH.linkSpeed());
      Serial.println("Mbps"); 
    
    if(hornbill.connect(HOST_ADDRESS,CLIENT_ID)== 0)
    {
        Serial.println("Connected to AWS");
        delay(1000);

        if(0==hornbill.subscribe("$aws/things/securekit/shadow/update",mySubCallBackHandler))
        {
            Serial.println("Subscribe Successfull to the topic : $aws/things/securekit/shadow/update");
        }
        else
        {
            Serial.println("Subscribe Failed, Check the Thing Name and Certificates");
            while(1);
        }
    }
    else
    {
        Serial.println("AWS connection failed, Check the HOST Address");
        while(1);
    }

    delay(2000);

}

void loop()
{     
      Serial.println("");
      Serial.print("ETH MAC: ");
      Serial.print(ETH.macAddress());
      Serial.print(", IPv4: ");
      Serial.print(ETH.localIP());
      if (ETH.fullDuplex()) {
        Serial.print(", FULL_DUPLEX");
      }
      Serial.print(", ");
      Serial.print(ETH.linkSpeed());
      Serial.println("Mbps"); 

    if(msgReceived == 1)
    {
        msgReceived = 0;
        Serial.print("Received Message:");
        Serial.println(rcvdPayload);
    }
    
     sprintf(payload,"Hello from hornbill ESP32 : %d",msgCount++);
        
        if(hornbill.publish("$aws/things/securekit/shadow/update",payload) == 0)
        {        
            Serial.print("Publish Messageed to :");
            Serial.println("$aws/things/securekit/shadow/update");
        }
        else
        {
            Serial.println("Publish failed");
        }
    
    delay(1000);
   
}

everslick, 2018-04-19:
Is this issue still unsolved for you? Or did you find a solution in the meantime?

pipi61, 2018-04-19:
Hi!
Try examples from arduino library first
c:\\.........\hardware\espressif\esp32\libraries\WiFi\examples\
or  [https://github.com/espressif/arduino-esp32/tree/master/libraries/WiFi/examples](url)

closed 2019-08-15 by stale[bot]