espressif/arduino-esp32 #800: Intermittant crash using WifiSecure and x509 certificates  on POST

steeeeeeved, 2017-10-31:
### Hardware:
Board:							Wemos ESP32 Dev Board
Core Installation/update date:			30/10/2017
IDE name:							Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:						115200

### Description:
Intermittent problems connecting to AWS IOT via HTTPS calls using WifiSecure and TLS Certificates.
When attempting to connect tot he AWS IOT endpoint with x509 certificates (which have been redacted in sample code), it fails about 80% of the time. 

Yesterday I updated the version of the https://github.com/espressif/arduino-esp32 to latest master, from a version several months old (I dont have an exact date unfortunately).  This code was working successfully previously. 

Resetting the board will occasionally allow the HTTP POST to complete, where there will be anything from 1 to 5 successful posts, before the crash. 

(Apologies if i have missed something easy!)

### Sketch:
```cpp

//Change the code below by your sketch
#include <WiFi.h>
#include <WiFiClient.h>
#include <WiFiGeneric.h>
#include <WiFiMulti.h>
#include <WiFiServer.h>
#include <WiFiSTA.h>
#include <WiFiType.h>
#include <WiFiUdp.h>

/*
  Connects to AWS IOT over HTTPS and pushes sensor data
  Apache 2.0 Licence. Steve Drew <steve@sdrew.net>
  Beta Quality - May eat hampster
*/

#include <WiFiClientSecure.h>
#include <ArduinoJson.h>

#include <DHT.h>

//Device
const char* deviceid = "test";
#define SLEEP_DELAY_IN_SECONDS  600
const char* devicelocation = "gh";

#define DHTTYPE DHT11
const int DHTPin = 22;
const int soilpin = 32;
const int POWER_PIN = 34;
const int LIGHT_PIN = 33;

//WIFI
const char* ssid     = "iot-things";     // your network SSID (name of wifi network)
const char* password = "XXXXXXXXXXXX"; // your network password

//IOT Endpoint
const int httpsPort = 8443;
const char* server = "XXXXXXXXXXXX.iot.eu-west-1.amazonaws.com";  // Server URL
const char* topic = "/topics/ESP32/DHT11"; // TOPIC to post to
// ASW IOT CA CERT

const char* test_ca_cert = \
                           "-----BEGIN CERTIFICATE-----\n" \
                           "-----END CERTIFICATE-----\n";

// You can use x.509 client certificates if you want
const char* privateKeyBuff = \
                             "-----BEGIN RSA PRIVATE KEY----- \n" \
                             "-----END RSA PRIVATE KEY-----\n";
[... 233 lines cut ...]
headers received

{"message":"OK","traceId":"eee7b8a2-21ec-c265-4915-4411716a5c5e"}[I][ssl_client.cpp:194] stop_ssl_socket(): Cleaning SSL connection.
Entering sleep mode

Starting connection to server...
[I][ssl_client.cpp:45] start_ssl_client(): Free heap before TLS 165692
[I][ssl_client.cpp:47] start_ssl_client(): Starting socket
[I][ssl_client.cpp:75] start_ssl_client(): Seeding the random number generator
[I][ssl_client.cpp:84] start_ssl_client(): Setting up the SSL/TLS structure...
[I][ssl_client.cpp:97] start_ssl_client(): Loading CA cert
[I][ssl_client.cpp:115] start_ssl_client(): Loading CRT cert
[I][ssl_client.cpp:122] start_ssl_client(): Loading private key
[I][ssl_client.cpp:153] start_ssl_client(): Performing the SSL/TLS handshake...
[I][ssl_client.cpp:165] start_ssl_client(): Protocol is TLSv1.2 Ciphersuite is TLS-ECDHE-ECDSA-WITH-AES-256-GCM-SHA384
[I][ssl_client.cpp:167] start_ssl_client(): Record expansion is 29
[I][ssl_client.cpp:173] start_ssl_client(): Verifying peer X.509 certificate...
[I][ssl_client.cpp:183] start_ssl_client(): Certificate verified.
[I][ssl_client.cpp:186] start_ssl_client(): Free heap after TLS 120012
Connected to server!
a2alduq2b5zo9j.iot.eu-west-1.amazonaws.com
headers received

{"message":"OK","traceId":"2453ffdc-9a92-00a4-532f-5febb23e4407"}[I][ssl_client.cpp:194] stop_ssl_socket(): Cleaning SSL connection.
```

copercini, 2017-11-01:
> Debug exception reason: Stack canary watchpoint triggered 
Out of stack....

try move these lines to setup:
`
  client.setCACert(test_ca_cert);
  client.setCertificate(certificateBuff); 
  client.setPrivateKey(privateKeyBuff);  
`

steeeeeeved, 2017-11-01:
Moving the three lines gives the same out of stack error.. (Thanks for identifying that for me, I'll get googling about workarounds/fixing)

steeeeeeved, 2017-11-03:
Hi,
Yes thats what I'm looking to do.. (Using a HiGrow temp / moisture sensor board)
So tracing back, I did not have the latest checkout of arduinio-esp32 and was missing the increase in stack size: https://github.com/espressif/arduino-esp32/commit/4495659ac53bcdc4f8e59d6918b3e472903d63a7#diff-9b0caa90c75b300921be10a08d10ff95

Thanks for pointing me in the right direction.

closed 2017-11-03 by steeeeeeved