espressif/arduino-esp32 #801: pinMode() causes watchdog reset and CPU freezes afterwards!

vishnumaiea, 2017-11-01:
### Hardware:
Board:							**NodeMCU-32S**
USB Chip: **CH340G**
Core Installation/update date:			**A week ago**
IDE name:							**Arduino IDE 1.8.5**
CPU Frequency:                                               **240MHz**
Flash Frequency:					 **80Mhz**
Upload Speed:						**921600**
OS: **Windows 7 64bit**
Power Source: **USB**

### Description:
I'm trying to interface a 3.2" TFT LCD display to ESP32 (NodeMCU-32S). I need 8 + 4 pins for parallel interface. So defined the pins and set all of them as OUPUT using the pinMode() in setup() function. The code compiles and uploads successfully. But as soon as the execution reaches the pinMode() function, the system gets reset and after a few resets, the CPU freezes. You can see in the serial output window that the next event after printing the "Initializing..." message is a reset. The loop() function never gets executed after the reset.

If I comment out the pinMode() line, everything works fine.

### Sketch:

A part of the code is given below

```cpp

#define CS_PIN 13
#define RST_PIN 12
#define DC_PIN 14
#define WR_PIN 27

#define PD0 15 //parallel data pins
#define PD1 2
#define PD2 0
#define PD3 4
#define PD4 16
#define PD5 17
#define PD6 21
#define PD7 22

void setup() {
  Serial.begin (115200);
  Serial.print("CPU Clock = ");
  Serial.println(ESP.getCpuFreqMHz());
  Serial.println ("Initializing...");

  pinMode (CS_PIN, OUTPUT); //set all pins as outputs
  // pinMode (RST_PIN, OUTPUT);
  // pinMode (DC_PIN, OUTPUT);
  // pinMode (WR_PIN, OUTPUT);
  // pinMode (PD0, OUTPUT);
  // pinMode (PD1, OUTPUT);
  // pinMode (PD2, OUTPUT);
  // pinMode (PD3, OUTPUT);
  // pinMode (PD4, OUTPUT);
  // pinMode (PD5, OUTPUT);
  // pinMode (PD6, OUTPUT);
  // pinMode (PD7, OUTPUT);
}

void loop() {
  Serial.println ("Working..");
  Serial.print ("CPU Clock = ");
  Serial.println (ESP.getCpuFreqMHz());
  delay(500);
}
```
### Serial output:

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:812
load:0x40078000,len:0
load:0x40078000,len:10212
entry 0x40078a00
CPU Clock = 240
Initializing...
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:812
load:0x40078000,len:0
load:0x40078000,len:10212
entry 0x40078a00
CPU Clock = 240
Initializing...
ets Jun  8 2016 00:22:57
```
### Build console output when uploading:

```
Sketch uses 140170 bytes (10%) of program storage space. Maximum is 1310720 bytes.
Global variables use 11084 bytes (3%) of dynamic memory, leaving 283828 bytes for local variables. Maximum is 294912 bytes.
esptool.py v2.1
Connecting........__
Chip is ESP32D0WDQ6 (revision 0)
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 8192 bytes to 47...

Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds...
Hash of data verified.
Flash params set to 0x022f
Compressed 11120 bytes to 7467...

Writing at 0x00001000... (100 %)
Wrote 11120 bytes (7467 compressed) at 0x00001000 in 0.1 seconds (effective 967.0 kbit/s)...
Hash of data verified.
Compressed 141312 bytes to 79541...

Writing at 0x00010000... (20 %)
Writing at 0x00014000... (40 %)
Writing at 0x00018000... (60 %)
Writing at 0x0001c000... (80 %)
Writing at 0x00020000... (100 %)
Wrote 141312 bytes (79541 compressed) at 0x00010000 in 1.2 seconds (effective 942.9 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 122...

Writing at 0x00008000... (100 %)
Wrote 3072 bytes (122 compressed) at 0x00008000 in 0.0 seconds...
Hash of data verified.

Leaving...
Hard resetting...

adonizer, 2017-11-01:
Same issue on pin 6 and 7.
Error Log:

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:572
load:0x40078000,len:0
load:0x40078000,len:9880
entry 0x400789d8
Guru Meditation Error of type LoadProhibited occurred on core  1. Exception was unhandled.
Register dump:
PC      : 0x40080e15  PS      : 0x00060530  A0      : 0x800d0f17  A1      : 0x3ffcf150  
A2      : 0x00000002  A3      : 0x00000002  A4      : 0x00000013  A5      : 0x00000017  
A6      : 0x000000ff  A7      : 0x00000000  A8      : 0x3f40141c  A9      : 0x00000000  
A10     : 0xffffffff  A11     : 0x00000040  A12     : 0x08000000  A13     : 0x00000000  
A14     : 0xffffffff  A15     : 0x00000006  SAR     : 0x0000001a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xffffffff  LBEG    : 0x400d1b6c  LEND    : 0x400d1bce  LCOUNT  : 0x00000002  

Backtrace: 0x40080e15:0x3ffcf150 0x400d0f14:0x3ffcf170 0x400d0fcf:0x3ffcf190 0x400e1163:0x3ffcf1b0

Rebooting...

me-no-dev, 2017-11-01:
@adonizer pins 6-11 are used by the flash chip. can not use them.

@vishnumaiea that is rather really strange! I am currently using pin13 for many things. Is it possible that there is something wrong with the board/module?

vishnumaiea, 2017-11-02:
@me-no-dev Not sure if it's a hardware problem. Couldn't find anything wrong with the hardware after initial inspection. I do not have another board to test this. May be I'll seek some help from my friends to check the same.

@adonizer Also, GPIOs 34 to 39 can only be used as INPUTs.

copercini, 2017-11-03:
@vishnumaiea This issue is very similar to this one: https://github.com/espressif/arduino-esp32/issues/796

My guess is that is something broken after commit https://github.com/espressif/arduino-esp32/commit/55289a45af1e7898ca0b8da037d92a83baaf0083 that when the board reset so often it stuck in the start (with any code)

Could you test it with an old core version, like https://github.com/espressif/arduino-esp32/tree/95123681d5340defa24472e441758fba88e3a34f  ?

vishnumaiea, 2017-11-03:
@copercini Will try that.

everslick, 2018-04-19:
Is this issue still relevant? @vishnumaiea Can you close it pls, if not?

vishnumaiea, 2018-04-20:
The cause of the problem is suspected to be the hardware bugs associated with revision 0 of the ESP32 chip (https://www.espressif.com/sites/default/files/documentation/eco_and_workarounds_for_bugs_in_esp32_en.pdf). This issue was not found in revision 1 chips when tested.

closed 2018-04-20 by vishnumaiea

nikisalli, 2019-06-01:
Hi! i'm working on an esp32 Revision 1 and i got the same problem when trying to set gpio13 as output through pinMode().
at 240Mhz clock it stamps this error at boot:
`Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.`
in my code i'm just using the serial port and two pwm channels:
`void setup() {
  ledcSetup(0,40000,8);
  ledcSetup(1,40000,8);
  ledcAttachPin(22,0);
  ledcAttachPin(21,1);
  
  pinMode(22,OUTPUT);
  pinMode(21,OUTPUT);
  
  ledcWrite(0,0);
  ledcWrite(1,0);
  
  //pinMode(13,OUTPUT);
  
  Serial.begin(9600);
}
void loop() {
}`
anyone got the same problem and/or a solution?

epareja2, 2023-06-16:
I'm having the same problem with an ESP32 DEV module and I think it's not related to the use of the pin.
The problem I'm facing is that after waking up from the deep sleep using the timer, it runs the setup() function but when it leaves setup() it freeze.

This is my setup() function and I've placed a lot of logs there to check if it was because of accessing some RTC_DATA_ATTR, but it doesn't have anything to do with that.

I post this here regardless of this topic being old, but still happening in 2023

Using this framework
Processing esp32dev (platform: espressif32 @ 6.3.1; board: esp32dev; framework: Arduino)
 framework-arduinoespressif32 @ 3.20009.0 (2.0.9)
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - tool-mkfatfs @ 2.0.1
 - tool-mklittlefs @ 1.203.210628 (2.3)
 - tool-mkspiffs @ 2.230.0 (2.30)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5

`void setup()
{
  char message_buf[100];

  // put your setup code here, to run once:
  Serial.begin(115200);
  configureIO();
  myPower.begin(PWR_SENSE_IN, BAT_SENS_AD, BAT_SENS_OUT);
  Counter = 0;
  // timerSetup();
  configureRTC();
  sprintf(message_buf, "Inicio despues de WakeUp\n");
  Serial.println(message_buf);
  sprintf(message_buf, "Inicio de configuracion time_t=(%ld)\n", (long)RTCTime);
  Serial.println(message_buf);
  sprintf(message_buf, "datetime =%s\n", ctime(&RTCTime));
  Serial.println(message_buf);
  myPower.setVoltageSettings(4.28, (2700.0 / (7500.0 + 2700.0)), 0.2);
  myPower.getBatteryADC();
  sprintf(message_buf, "After Configuring voltage ADC = %d and Voltage = %f\n\r", myPower.batteryADC, myPower.batteryVoltage);
  Serial.println(message_buf);
}`