espressif/arduino-esp32 #802: enable DAC CW?

baradhili, 2017-11-02:
### Hardware:
Board:							?ESP32 WROOM?
Core Installation/update date:			?11/jul/2017?
IDE name:							?Arduino IDE?
Flash Frequency:					?40Mhz?
Upload Speed:						?115200?

### Description:
Wishlist item I guess.. but can we get access to enabling the CW generator associated with the DAC? https://github.com/krzychb/dac-cosine appears to have working code, just need Arduino integration

### Sketch:
NA
### Debug Messages:
NA

everslick, 2018-04-19:
What is stopping you from using the code you linked to? What do you expect the Arduino Core team to do to solve a specific problem that is relevant for the target user group of the Arduino platform? Can you elaborate on the issue you are having? Thanks.

baradhili, 2018-04-19:
Sorry forgot this issue was open its so long ago... please close, 
problem solved

closed 2018-04-19 by everslick