espressif/arduino-esp32 #808: WiFi authentication issue since core update

englishman-hu, 2017-11-04:
### Hardware:
Board:							doit dev module
Core Installation/update date:			25/Oct/2017?
IDE name:							Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:						921600

### Description:
Since core update my routine cannot connect to wifi, keeps dropping AUTH_FAIL events

It works fine with earlier core (pulled on 19 July).

### Sketch:
...
// init wifi and connect
void wifiInit() {
  WiFi.mode(WIFI_STA);
  WiFi.disconnect();
  WiFi.setAutoConnect(true);
  WiFi.setAutoReconnect(true);
  delay(10);
  WiFi.begin(SSID, PASS);
  WiFi.printDiag(Serial);
}

// non-blocking wifi re-connect, returns connected status, assumes WiFi.setAutoConnect(true)
bool wifiConnect() {
  // do not re-connect within WIFI_RECONNECT_LATENCY msecs of last connect attempt
  static unsigned long _lastattempt = 0;
  unsigned long tm = millis();
  if (WiFi.status() != WL_CONNECTED && tm > _lastattempt + WIFI_RECONNECT_LATENCY) {
    Serial.println("\nConnecting to WiFi...");
    Serial.println(WiFi.status());
    WiFi.printDiag(Serial);
    WiFi.reconnect();
    _lastattempt = tm;
  }
  return WiFi.status() == WL_CONNECTED;
}
  
...

### Debug Messages:

[D][WiFiGeneric.cpp:265] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:270] _eventCallback(): Reason: 201 - AUTH_FAIL

stickbreaker, 2017-11-04:
@englishman-hu check out #807 

Chuck.

englishman-hu, 2017-11-04:
Pulled it and tried...
Unfortunately did not fix the issue:

[D][WiFiGeneric.cpp:265] _eventCallback(): Event: 2 - STA_START
Mode: STA
Channel: 3
Auto connect: 1
SSID (3): ***
Passphrase (12): ************
BSSID set: 0
...
[D][WiFiGeneric.cpp:265] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:270] _eventCallback(): Reason: 201 - AUTH_FAIL
Connecting to WiFi...
1
Mode: STA
Channel: 1
Auto connect: 1
SSID (3): ***
Passphrase (12): ************
BSSID set: 0
[D][WiFiGeneric.cpp:265] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:270] _eventCallback(): Reason: 201 - AUTH_FAIL
Connecting to WiFi...

aboulfad, 2017-11-14:
I have the same exact issue... w same errors, STA_DISC: AUTH_FAIL. Any ideas ???

**Hardware**:
Board:	ESP32S (WROOM32)
Core Installation/update date:	yesterday (pulled master)
IDE name:	Arduino IDE 1.8.5
Flash Frequency:	80Mhz
Upload Speed:	115200

Router: Netgear R7000, 2.4Ghz, WPA2-PSK [AES]

aboulfad, 2017-11-14:
ok I tried an older router Trendnet TEW-639GR, and the esp32 wifi does connect. So there must be something that it does NOT like with my Netgear's R7000  2.4Ghz flavour :-(

aboulfad, 2017-11-15:
ok very weird behaviour but I have finally solved it. I was always using static IP config but without specifying dns (which I didn't care as I was doing a test). I removed the static IP config entries and it worked !

If I use the form: _WiFi.config(local_IP, gateway, subnet)_, I always get the same errors as the OP.
If I add to above DNS entry: _WiFi.config(local_IP, gateway, subnet, **primaryDNS**)_, then it ALWAYS works... Probably unrelated to this issue as the OP did not specify if he uses static IP or not.

I don't understand why the lack of the dns entry with static IP worked on the old router and not the R7000.

englishman-hu, 2017-11-15:
I use dynamic IP - but the issues is even before it would come to IP assignment.
It simply cannot complete the AP association and authentication.
The main thing is that it works fine with the core I initially pulled in July, but keeps failing with the current core.

lonerzzz, 2017-12-11:
@englishman-hu If you have not done so already, you should raise the issue in the IDF forums.

closed 2018-04-19 by everslick