espressif/arduino-esp32 #810: Smart config will not reconnect automatically after restart

bill-orange, 2017-11-05:
### Hardware:
Board:							ESP32 Wemos
Core Installation/update date:		not sure, latest, I think
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:					115200

### Description:
SmartConfig behave differently on ESP32 than ESP8266.  Both connect using the phone APP just fine.  Upon a restart the ESP8266 will reconnect.  The ESP32 will not reconnect and will search until I use the phone APP again to provide credentials.

Shouldn't they behave the same and reconnect?  Could there be something in my ESP32 code inhibiting re-connection?  

### Sketch:
Standard examples embedded in my lengthy sketch.

/

copercini, 2017-11-05:
here is the fix: https://github.com/espressif/arduino-esp32/pull/807
=)

bill-orange, 2017-11-05:
I added:

sta_conf->bssid_set = 0;

to the .cpp file and recompiled. It made no differece. If I reset the ESP32 is searches forever until I use the phone app. Is there any else I need to do to effect the change?

Here's my connect function:

void setupWiFi () {
/* Set ESP32 to WiFi Station mode /
WiFi.mode(WIFI_AP_STA);
/ start SmartConfig */
WiFi.beginSmartConfig();

/* Wait for SmartConfig packet from mobile */
Serial.println("Waiting for SmartConfig.");
while (!WiFi.smartConfigDone()) {
delay(500);
Serial.print(".");
}
Serial.println("");
Serial.println("SmartConfig done.");

/* Wait for WiFi to connect to AP */
Serial.println("Waiting for WiFi");
while (WiFi.status() != WL_CONNECTED) {
delay(500);
Serial.print(".");
}
Serial.println("WiFi Connected.");
Serial.print("IP Address: ");
Serial.println(WiFi.localIP());
}

Here's verbose debug:

entry 0x40078a3c
[D][WiFiGeneric.cpp:265] _eventCallback(): Event: 14 - AP_STACONNECTED
[D][WiFiGeneric.cpp:265] _eventCallback(): Event: 14 - AP_STACONNECTED
[D][WiFiGeneric.cpp:265] _eventCallback(): Event: 14 - AP_STACONNECTED
Waiting for SmartConfig.
.........[D][WiFiSTA.cpp:621] _smartConfigCallback(): Status: FIND_CHANNEL
.....................................................................

bill-orange, 2017-11-10:
I think we can close this issue.  I still think, that as an objective, smartConfig should behave the same on an ESP 32 as it does on an ESP8266 for software portability.

closed 2017-11-10 by bill-orange

johnty, 2018-02-13:
any updates on this? would be great if this could work the same as the esp8266!

NareshDealer, 2018-03-27:
I m having similar problem.. where the ssid and password is not getting stored and i need to resupply the ssid and password.