espressif/arduino-esp32 #814: touchRead() gives occasionally wrong values

beegee-tokyo, 2017-11-06:
Anybody has the same problems?

### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:		Home 0.3.2·Core 3.5.0b2
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:					OTA

### Description:
I poll the touch pin status with a timer function every 500ms. Occasionally the value returned by **`touchRead()`** is changing, even the touch "pin" is not touched at all nor environment temperature/humidity changed.
I poll it, because I want to detect both touch and release situations to be able to detect a "long-press" inside my application (in the future).
I tried **`portENTER_CRITICAL(&touchMux);`** and **`portEXIT_CRITICAL(&touchMux);`** around the **`touchRead()`**, but it doesn't change anything.

The whole app is using the **`loop()`** for updating a TFT display and 3 background tasks (2 on CPU1, 1 on CPU0) to read light sensors (every 10 seconds) , temperature sensors (every 20 seconds) and get weather info from the internet (every 30 minutes).
In **`loop()`** I am as well listening for UDP broadcasts to receive information from local sensors.
Another task is used to send all the collected information through MQTT to an external server.

Could it be that occasionally other tasks disturb the **`touchRead()`**? But that should be avoided by using the **`portENTER_CRITICAL(&touchMux);`** and **`portEXIT_CRITICAL(&touchMux);`** or am I wrong?

As you can see in the code below, I get around the problem atm by calling **`touchRead()`** a second time, but I hope I can avoid this.

### Sketch:
(Only the touch related functions)
```cpp
#include "setup.h"

/** Status of last touch read */
bool isTouched = false;
/** Lock for touch read */
portMUX_TYPE touchMux = portMUX_INITIALIZER_UNLOCKED;

void touchFunction();

/**
  initTouch
  Initialize timer to read touch value every 500ms
*/
void initTouch() {
  startTimerMsec(500, touchFunction, true); // Starts a HW timer to call touchFunction() repeatedly every 500ms.
}

long touchCounter = 0;

/**
  touchFunction
  Check touch value and restart all tasks if touched
*/
void touchFunction() {
  // portENTER_CRITICAL(&touchMux);
  int touchValue = touchRead(T3);
  // portEXIT_CRITICAL(&touchMux);

  if (touchValue < 50) {
    Serial.println(String(touchCounter) + "\t\tTouch value = " + String(touchValue));
  } else {
    Serial.println(String(touchCounter) + " Touch value = " + String(touchValue));
  }
  touchCounter++;

  if (touchValue < 20) {
    touchValue = touchRead(T3);
    if (touchValue < 20) {
      if (!isTouched) {
        isTouched = true;
        addMeeoMsg("touch", "1", false);
        // xTaskResumeFromISR(lightTaskHandle);
        // xTaskResumeFromISR(tempTaskHandle);
        // xTaskResumeFromISR(weatherTaskHandle);
      }
    }
  }
  if (touchValue > 50) {
    if (isTouched) {
      isTouched = false;
      addMeeoMsg("touch", "0", false);
    }
  }
}
```

### Debug Messages:
Results from the touchRead(), first number is count of touchRead() done.
```
0 Touch value = 64
1 Touch value = 65
2		Touch value = 32
3 Touch value = 65
4 Touch value = 65
5 Touch value = 65
6		Touch value = 16
7 Touch value = 65
8 Touch value = 65
9 Touch value = 64
10 Touch value = 64
11 Touch value = 64
12 Touch value = 64
13 Touch value = 64
14		Touch value = 36
15 Touch value = 64
...
38 Touch value = 65
39		Touch value = 27
40 Touch value = 65
...
129 Touch value = 65
130		Touch value = 7
131 Touch value = 65
...
225 Touch value = 64
226		Touch value = 45
227 Touch value = 64
...
267 Touch value = 65
268		Touch value = 44
269 Touch value = 65
...
313 Touch value = 64
314		Touch value = 31
315 Touch value = 64
...
338 Touch value = 65
339		Touch value = 12
340 Touch value = 64
...
541 Touch value = 65
542		Touch value = 13
543 Touch value = 64
...
666 Touch value = 65
667		Touch value = 20
668 Touch value = 64
...
690 Touch value = 64
691		Touch value = 39
692 Touch value = 64
693 Touch value = 64
694 Touch value = 64
695		Touch value = 43
696 Touch value = 64
...
730 Touch value = 65
731		Touch value = 30
732 Touch value = 64
733 Touch value = 64
734 Touch value = 65
735		Touch value = 0
736 Touch value = 64
...
```

me-no-dev, 2017-11-06:
That is rather strange. Since you are reading it only from that timer function, you do not need to lock/unlock it. Why not use touch interrupt (beware it runs in ISR in IRAM so you need to use some RTOS technique to transfer the result to your logic)

beegee-tokyo, 2017-11-06:
I read the description for touch interrupt and maybe I misunderstand, but it seems the touch interrupt is only fired when the value drops below the treshold but not if it goes above again. That way I would not be able to detect long touches. 
Correct me if I am wrong.

me-no-dev, 2017-11-06:
as far as I remember, you will get multiple interrupts, or better said, interrupt every cycle of the read operation. You can detect touch end by not getting interrupt in a given period

me-no-dev, 2017-11-06:
or long touch by counting the number of interrupts you get.

beegee-tokyo, 2017-11-06:
Will give it a try and report back tomorrow.

beegee-tokyo, 2017-11-07:
Changed the code to use **`touchAttachInterrupt()`** and detecting release and long touch in a timer ISR. This seems to be ok. 
### Sketch:
(Only the touch related functions)
```cpp
#include "setup.h"

/** Touch status of T3 pad */
bool isTouchedT3 = false;
/** Long touch status of T3 pad */
bool longTouchT3 = false;
/** Time that pad T3 was touched */
long touchTimeT3 = 0;

void checkT3Status();
void touchT3ISR();

/**
	initTouch
	Initialize timer to read touch value every 500ms
*/
void initTouch() {
  startTimerMsec(500, checkT3Status, true);
  touchAttachInterrupt(T3, touchT3ISR, 20);
}

/**
  touchT3ISR
  Called when touch pin value goes below treshold
*/
void touchT3ISR() {
  if (!isTouchedT3) {
    touchTimeT3 = millis();
    isTouchedT3 = true;
    addMeeoMsg("touch", "1", false);
    xTaskResumeFromISR(lightTaskHandle);
    xTaskResumeFromISR(tempTaskHandle);
    xTaskResumeFromISR(weatherTaskHandle);
  }
}

/**
  checkT3
  Checks if T3 pad is still touched
*/
void checkT3Status() {
  if (isTouchedT3) {
    if (touchRead(T3) > 50) {
      isTouchedT3 = false;
      longTouchT3 = false;
      addMeeoMsg("touch", "0", false);
      addMeeoMsg("longtouch", "0", false);
    } else {
      if ((millis()-touchTimeT3) >= 3000) {
        longTouchT3 = true;
        addMeeoMsg("longtouch", "1", false);
      }
    }
  }
}
```

closed 2017-11-07 by beegee-tokyo