espressif/arduino-esp32 #818: How to install it?

suelio1, 2017-11-08:
Please fill the info fields, it helps to get you faster support ;)

If you have a Guru Meditation Error, please decode it:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?ESP32 Dev Module?
Core Installation/update date:			?11/jul/2017?
IDE name:							?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
Upload Speed:						?115200?

### Description:
Describe your problem here

### Sketch:
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

beegee-tokyo, 2017-11-08:
Maybe I misunderstand your question but did you do a [search](https://duckduckgo.com/?q=platformIO+install+linux&atb=v42-7ds&ia=web)
and then first result leads to [Docs » PlatformIO Core » Installation](http://docs.platformio.org/en/latest/installation.html)

suelio1, 2017-11-08:
@beegee-tokyo thanks. I already installed

closed 2018-04-19 by everslick