espressif/arduino-esp32 #819: Short circuit boolean evaluation?

stickbreaker, 2017-11-08:
Am I just old and slow?  I just spent the hours tracking down a Short Circuit Boolean Evaluation problem.
Now I have to go through all of my code looking for similar logic issues!

```c++
bool checkShort(uint8_t a, uint8_t line){
// a function that displays a column title on multiple lines is necessary
bool complete = //title fits on this line;
return complete;
} 

bool complete_title = true;
uint8_t line=0;
do{ // loop repeats until all titles have been displayed
  uint8_t a=0;
  while(a<10){
    complete_title = complete_title & checkShort(a,line);
    a++;
    }
  line++;
  }while(!complete_title);

```
I was assuming (ass u me) that `checkShort()` would be evaluated for each value of a.  But alas, as soon as `complete_title` is false, `checkShort()` is never evaluated again! 

What other Ordering **got you's** are there? 

I just discovered that this build defaults to SCBE, what else ?

Chuck.

me-no-dev, 2017-11-08:
wooot? are you saying that the ```while(a<10)``` is actually running less than 10 times?

stickbreaker, 2017-11-08:
I can't be certain about the while(), but effectively yes. Once `complete_title` is false nothing else in the loop has any effect.  If I added another statement to the loop it should execute it, but with this simple example, the value `a` is never used again, so the compiler could short circuit the loop completely when `complete_title` becomes false.

Chuck.

atanisoft, 2017-11-08:
have you tried with && instead of & inside the while loop? ie:
`
while(a<10) {
    complete_title = complete_title && checkShort(a, line);
    a++;
}
`

stickbreaker, 2017-11-08:
The & was just a type, the actual code used logical and(&&) not the bit wise (&). 

Now if the statement was reordered
`complete_title = checkShort(a,line) && complete_title;` It should function as expected. but that is dependent upon the optimization configurations.  If the compiler decides that it is easier to evaluate a boolean than a function it may for speed optimization do the boolean first.  Since as soon as the equation is false, nothing can change it back to true.

I'm going to have to review all of my code for this type of expectation failure.

Chuck.

ManniAT, 2017-11-08:
I tend to say you should (don't know the compiler so  well) get a warning that you assing an int to a boolean.
You use & (single) which is binary / bitwise AND which combines the two binary values of completet_title and the result of checkShort(...). 
like aranisoft wrote - use && instead and it should not call checkShort as soon as complete_title is false.
But I don't understand the loop.
You start with completet_title true - checkShort gets called.
As long as it returns true it gets call in the next inner loop again.
But then the outer loop (do...while) will end.
On the other side - if checkShort results false - complete_title will become false.
In the next inner loop shortcut will take place (complete_tiltle is already false) and checkShort will no longer be called.
The outer loop will keep running - the inner also without any change in the result.
If you want to call checkShort in every loop (no matter what compelete_title is) just reverse the and.
...=checkShort() && complete_title;

This would still keep complet_title false once checkShort returns false - but checkShort will get called for the rest of the remaining a++
But - it will also result in an endless outer loop since this will run while(!comp...)

ManniAT, 2017-11-08:
Uuuups - while commenting you wrote too.
About change of order and optimization.
How about 
**complete_tilte&=checkShort();**
Although it is binary AND - C uses the same values for true / false -- so if complete_title is true -- a binary AND with the same result returned by checkShort should bee true too.
And I think the compiler can't optimize this away

stickbreaker, 2017-11-08:
@ManniAT too many 'if's' I'm just going to change my coding style.  relying on complete evaluation is a non starter. I'll just use if's to compare variable, not functions. If a function evaluates to a boolean, I'll just code a local variable and use that variable in the succeeding equation.  That would guarantee my function is evaluated.  The compiler will optimize the code.  At least by setting a local variable it **should** actual evaluate the function.

Chuck.

closed 2017-11-14 by stickbreaker