espressif/arduino-esp32 #821: SNTP code for esp32 in arduino

jaggi, 2017-11-10:
Please fill the info fields, it helps to get you faster support ;)

If you have a Guru Meditation Error, please decode it:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?ESP32 Dev Module?
Core Installation/update date:			?11/jul/2017?
IDE name:							?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
Upload Speed:						?115200?

### Description:
Describe your problem here

### Sketch:
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

jaggi, 2017-11-10:
hiii i need a code to get a date and time from sntp server for esp32 in arduino ...can anyone suggest me???

beegee-tokyo, 2017-11-10:
@jaggi [NTPClient by Fabrice Weinberg](https://github.com/arduino-libraries/NTPClient) works fine for me.

MarkyAD, 2017-11-10:
```javascript
// example using usa eastern standard/eastern daylight time
// edt begins the second sunday in march at 0200
// est begins the first sunday in november at 0200

const char* NTP_SERVER = "ntp.mydomain.com";
const char* TZ_INFO    = "EST5EDT4,M3.2.0/02:00:00,M11.1.0/02:00:00";

struct tm timeinfo;

void setup() {
  Serial.begin(115200);
  configTzTime(TZ_INFO, NTP_SERVER);
  // start wifi here
  if (getLocalTime(&timeinfo, 10000)) {  // wait up to 10sec to sync
    Serial.println(&timeinfo, "Time set: %B %d %Y %H:%M:%S (%A)");
  } else {
    Serial.println(&timeinfo, "Time not set");
  }
}

void loop() {
  if (getLocalTime(&timeinfo)) {
    Serial.println(&timeinfo, "%Y-%m-%d %H:%M:%S");
  }
 delay(1000);
}

MarkyAD, 2017-11-10:
Just adding that struct tm is declared in tools/sdk/include/newlib/time.h its basically the same as unix struct tm, so look there for what elements are contained within it in order to use them in your sketch. On unix, "man ctime" shows the same information.

beegee-tokyo, 2017-11-10:
@MarkyAD I tried this before, but this does not update the timer that is used when calling the Arduino time lib functions hour(), minute(), ... And it looks like (maybe I do not understand getLocalTime() well) that everytime I call getLocalTime() the NTP server is called. I don't want to call the NTP timer every time I need a timestamp somewhere in my code.

MarkyAD, 2017-11-10:
Hey @beegee-tokyo, good to see you again. getLocalTime() is simply a wrapper around the function calls time() (like now() in arduino timelib) and localtime_r() which updates the struct tm elements. The functions configTime() or configTZTime() init sntp (once in setup) but after that the ntp server is called at hourly intervals. The functionality of timelib is available by the variables in the struct tm structure (tm_hour, tm_min, tm_sec, etc) and by unix-like time function calls (asctime(), ctime(), gmtime(), localtime(), strftime(), etc).

beegee-tokyo, 2017-11-10:
So I can call 
```
time(&now);
localtime_r(&now, info);
```
To update the tm struct instead of calling `getLocalTime()`?

MarkyAD, 2017-11-10:
@beegee-tokyo yes exactly, except that localtime_r takes both arguments as pointers, so if "info" is 
declared as struct tm, then its...
```javascript
time_t now;
struct tm info;

time(&now);
localtime_r(&now, &info);

zafrirron, 2018-02-03:
Since I've wasted A LOT of time on getting NTP time to work (paying for my stupidity), I hope this may save some time for others.

When calling configtime with NTP server name (like "pool.ntp.org"), you better make sure you have a valid DNS settings (to resolve the domain name).

A valid DNS will be available if you started the network connection with default parameters (like Wifi.begin(ssid,sspass)), but will not be available in case you started your network connection with WiFi.config(IP,gateway,subnet) (used to get fix IP) without giving it valid DNS -> WiFi.config(IP,GW,SUB,DNS1,DNS2)!!!

beegee-tokyo, 2018-02-03:
Nice finding. I added it to the [Wiki](Missing DNS settings when using static IP)

RottenVanSuti, 2018-03-20:
Hi, I have a little issue with classes configTzTime, setTime. In my code appears they are not declared, but I suppose they are inluded into the "NTP.client.h" right? If don't, sorry, I'm no so experienced. Did you make these functions?

everslick, 2018-04-19:
This issue is closed, because it looks as if it is not a bug or problem with the ESP32 Arduino core or its support libraries. For general API usage questions or help on specific coding challenges, please visit the [arduino-esp32 Gitter channel](https://gitter.im/espressif/arduino-esp32). If you feel this issue was closed in error, reopen it and comment, why you think this is a bug in the Arduino-Core.

closed 2018-04-19 by everslick

CarlosEspinosaBrazil, 2019-12-18:
Congratulations to all of you. Thank you very much. God bless