espressif/arduino-esp32 #823: Cannot Capture image using ov7725 in esp32

James-sjec, 2017-11-10:
I used the example program from https://github.com/igrr/esp32-cam-demo
 
I get the following Error message:  

I am using WROOM-32 , below is the picture of the development board
img_20171109_001809 1

I enabled test pattern ,after which I got the following error
apc - 2017 11 09 00 10 - 001 3d
apc - 2017 11 09 00 10 - 002 3d

I am using following pin map
apc - 2017 11 09 00 32 - 003 3d
@James-sjec

me-no-dev, 2017-11-10:
Why not post this into it's repository? :)

What I see above is that the queue that is used to acquire the image is emptied too slow and you are probably running out of memory. What are you capturing? Which camera?

igrr, 2017-11-22:
Closing as the same was posted in esp32-cam-demo.

closed 2017-11-22 by igrr