espressif/arduino-esp32 #824: Getting error while using BT stack.

Nakul93, 2017-11-10:
I am getting core panic issue while using BT stack, when BLE is used individually it works fine but when I integrated it with other code it panics the core and gives this following msg
/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./timers.c:799 (prvCheckForValidListAndQueue)- assert failed!
abort() was called at PC 0x4008761f on core 0
Guru Meditation Error: Core  0 panic'ed (abort)

Backtrace: 0x40085837:0x3ffe3ba0 0x00000000:0x3ffe3bc0 0x40087632:0x3ffe3be0 0x40086c80:0x3ffe3c10 0x40082b12:0x3ffe3c40 0x40082ccc:0x3ffe3c70 0x4007867d:0x3ffe3cb0 0x4007879c:0x3ffe3ce0 0x40078a68:0x3ffe3d50 0x400800f9:0x3ffe3e70 0x40007c34:0x3ffe3eb0 0x40000740:0x3ffe3f20

CPU halted.

Also the size of the code got increased while using BT stack so I have updated the size in boards.txt

me-no-dev, 2017-11-10:
Increased how much? What board? Minimal code that can reproduce? Decoded exception?

Nakul93, 2017-11-10:
Ideally it was 1310720 bytes I have changed it to 1410720. I am using ESP32 WROOM board.
I have provided the backtrace as well.

me-no-dev, 2017-11-10:
backtrace is decoded using this: https://github.com/me-no-dev/EspExceptionDecoder

the size that you changed also corresponds to the size of the partitions here: https://github.com/espressif/arduino-esp32/blob/master/tools/partitions/default.csv

APP0 and APP1 is where sketch go

Nakul93, 2017-11-10:
So do i have to increase the size of APP0 and APP1?  Is that the reason I am getting that error?
And I've tried installing the exception decoder plugin, it doesn't show up in arduino.

me-no-dev, 2017-11-10:
did you restart Arduino? Installed in the correct place?

Nakul93, 2017-11-10:
Yes now I am getting the exception decoder, this is what it displayed after pasting the backtrace

Decoding 11 results
0x40085837: xPortInIsrContext at ?? line ?
0x40087632: prvCheckForValidListAndQueue at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./timers.c line 839
0x40086c80: vTaskStartScheduler at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./tasks.c line 4420
0x40082b12: start_cpu0_default at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./cpu_start.c line 257
0x40082ccc: call_start_cpu0 at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./cpu_start.c line 165
0x400800f9: _WindowUnderflow8 at ?? line ?

Nakul93, 2017-11-13:
How to increase the partition size?
I have tried editing the default.csv file but it gives error.

HelloDB, 2017-12-17:
I also met this problem，He couldn't run after I changed default.csv 

# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x150000,
app1,     app,  ota_1,   0x160000,0x150000,
eeprom,   data, 0x99,    0x310000,0x1000,
spiffs,   data, spiffs,  0x311000,0x14F000,

------------------------------------------
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:812
load:0x40078000,len:0
load:0x40078000,len:11404
entry 0x40078aa0
user code done
---------------------------------
### Hardware:
Board: ESP32 Dev Module
Core Installation/update date: 20171217
IDE name: Platform.io
Flash Frequency: 40Mhz
Upload Speed: 115200

HelloDB, 2017-12-17:
When I delete spiffs it can run, but I need spiffs, what should I do
```
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x150000,
app1,     app,  ota_1,   0x160000,0x150000,
eeprom,   data, 0x99,    0x310000,0x1000,
# spiffs,   data, spiffs,  0x311000,0x14F000,
```

HelloDB, 2017-12-17:
When I reduce spiffs it can run, 0x311000 + 0x14F000 = 4060kb, 4060k <4 M Why can not run
```-------------
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x150000,
app1,     app,  ota_1,   0x160000,0x150000,
eeprom,   data, 0x99,    0x310000,0x1000,
spiffs,   data, spiffs,  0x311000,640K,

closed 2018-04-19 by everslick