espressif/arduino-esp32 #825: Wierd value from analogRead output after Wifi connection

docloulou, 2017-11-10:
### Description:
Hello,
I want to use the code from [there](https://learn.sparkfun.com/tutorials/esp8266-thing-hookup-guide/example-sketch-ap-web-server) to build a small api for my ESP32.

The issue is when i get value from my ANALOG_PIN before `  initWifi();  server.begin();` its works but after the value is ALWAYS `4095`

TRY1: If i dont start wifi i got RIGHT value from serial
TRY2: If i add `  initWifi();  server.begin();`  then the ANALOG_PIN value is stuck to `4095 ` in serial output and WEBPAGE, the fact is the `DIGITAL_PIN  ` show the RIGHT value in both serials and web page.

Thanks for your help guys

### Sketch:
```cpp

#include <WiFi.h>

int val = 0;           // variable to store the value read

const uint8_t LED_PIN = 5; // Thing's onboard, green LED
const uint8_t ANALOG_PIN = 27; // The only analog pin on the Thing
const uint8_t DIGITAL_PIN = 12; // Digital pin to be read

const char* ssid     = "Livebox-51E0"; // Your ssid
const char* password = "xxxxxxxxxx"; // Your Password

WiFiServer server(80);

void setup() {

  initHardware();

val = analogRead(ANALOG_PIN);    // read the input pin

Serial.print(val);             // <<<---- the right value

  initWifi();
  server.begin();

val = analogRead(ANALOG_PIN);    // read the input pin

Serial.print(val);             // <<---- Wrong value (WHY ?!) : always 4095

}

void loop() {

WiFiClient client = server.available();

if (!client) {
    return;
  }

  // Read the first line of the request
  String req = client.readStringUntil('\r');
  Serial.println(req);
  client.flush();

  // Match the request
  int val = -1; // We'll use 'val' to keep track of both the
                // request type (read/set) and value if set.
  if (req.indexOf("/led/0") != -1)
    val = 0; // Will write LED low
  else if (req.indexOf("/led/1") != -1)
    val = 1; // Will write LED high
  else if (req.indexOf("/read") != -1)
    val = -2; // Will print pin reads
  // Otherwise request will be invalid. We'll say as much in HTML

  // Set GPIO5 according to the request
  if (val >= 0)
    digitalWrite(LED_PIN, val);

  client.flush();

  // Prepare the response. Start with the common header:
  String s = "HTTP/1.1 200 OK\r\n";
  s += "Content-Type: text/html\r\n\r\n";
  s += "<!DOCTYPE HTML>\r\n<html>\r\n";
  // If we're setting the LED, print out a message saying we did
  if (val >= 0)
  {
    s += "LED is now ";
    s += (val)?"on":"off";
  }
  else if (val == -2)
  { // If we're reading pins, print out those values:
    s += "Analog Pin = ";
    s += String(analogRead(ANALOG_PIN));
    s += "<br>"; // Go to the next line.
    s += "Digital Pin 12 = ";
    s += String(digitalRead(DIGITAL_PIN));
  }
  else
  {
    s += "Invalid Request.<br> Try /led/1, /led/0, or /read.";
  }
  s += "</html>\n";

  // Send the response to the client
  client.print(s);
  delay(1);
  Serial.println("Client disonnected");

delay(100); //delay for reread
}

void initHardware()
{
  Serial.begin(115200);
  pinMode(DIGITAL_PIN, INPUT_PULLUP);
  pinMode(LED_PIN, OUTPUT);
  digitalWrite(LED_PIN, LOW);
  // Don't need to set ANALOG_PIN as input, 
  // that's all it can be.
}

void initWifi(){
  // Connect to WiFi network
WiFi.mode(WIFI_STA);
Serial.println();
Serial.println();
Serial.print("Connecting to ");
Serial.println(ssid);

WiFi.begin(ssid, password);

while (WiFi.status() != WL_CONNECTED) {
delay(500);
Serial.print(".");
}
Serial.println("");
Serial.println("ESP-32 is online");

// Start the server
server.begin();
Serial.println("Server started");

// Print the IP address
Serial.println(WiFi.localIP());
}
```

### serial output:
```
0

Connecting to Livebox-51E0
..
ESP-32 is online
Server started
192.168.1.31
4095 
```

me-no-dev, 2017-11-10:
use pins 32 and above when WiFi is ON

closed 2017-11-10 by me-no-dev

thecvely, 2018-10-24:
The esp 32 integrates two 12-bit ACD registers. ADC1 whit 8 channels attached to  GPIOs 32-39 ande ADC2 whit 10 channels in another pins. The thing is the ESP32 uses the ADC2 to manage wifi functions, so if you use Wifi, you can´t use that register. 
See that here https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/adc.html

0x13ens, 2019-10-30:
If 27 is the only analog pin, how can we transmit sensor data? pins 32 and above do not work only 27...