espressif/arduino-esp32 #826: Memory issues for very large program

shofixti244, 2017-11-10:
### Description:
Hey everyone, please do forgive me if I've come to the wrong neighborhood with this question, but I've been struggling with it for a few days now and I'm pretty much stumped.

So I've been working with the ESP32 (Adafruit HUZZAH32 to be exact) for quite some time now, working on a basic IoT project that involves connecting to Amazon Web Services. I've managed to get Wifi, BLE and the AWS connection all working in separate projects, but now that I'm bringing them together I'm hitting some issues.

After integrating my BLE library, my ported AWS library, and my Wifi code, I was surprised when all of a sudden my serial output started to look like this:

[image]

And of course, asides from the garbage Serial output, the program stopped functioning in general.

2 days of trial and error later, **I find that so long as the final compressed size of my codebase is larger than 750,000 bytes, I encounter this problem.**

Not being an experienced embedded programmer myself, the only conclusion that I've been able to arrive at so far is that the programs code is so large that I'm seeing a memory over-run issue. So I turned my attention to the Partitions table of the board, and made some modifications:

[image]

If I'm understanding this all correctly, this should allow for more than enough memory for my program to operate. However I'm still seeing the garbage output and lack of functionality mentioned above.

All the details about my setup are described below... Any input at all here would be a fantastic help. I'm only just learning this whole new world of development, so it may be something really obvious that I'm just not aware of.

### Setup:
Board:							Adafruit HUZZAH32
IDE name:						Atom IDE, Platform.io, Espressif 32 (Stage)
Libraries: 						awsiot for embedded C (ported), ESP32 BLE Arduino
arduino-esp32 commit: 			aba1d7d

me-no-dev, 2017-11-10:
Hi :) So shorter sketches with the same configuration work? Can you post a minimal sketch that shows the problem?

shofixti244, 2017-11-10:
I'm out of the office for a bit, but I'll attempt to make a simplified repeatable sketch on Sunday. I'll need to find a way to write a program that uses 750,00 bytes in a "simple" way, but I'm sure I can figure something out.

Also, Thank you for the assistance... been stressing abo

everslick, 2017-11-11:
I bet my life that you have a memory corruption bug in your sketch that only reveals itself, because your changes also alter the memory layout of your binary. especially the garbled output hints on a Serial.print() with a stray pointer. a static code analyses tool like cppcheck might help.

My binary is currently a tiny bit short of 900K without those issues. (i got different ones, not related to size though) ;-)

```
Linking ../build/ESP32/3.0.0/alpha/ger/genesys.bin
  Versions: 2.4.0-252-g571a7a5-dirty, 80a7d60
esptool.py v2.1

Memory usage
  Ram:    45104 bytes
  Flash: 892191 bytes

Flash size: 4MB (32Mb)
```

shofixti244, 2017-11-13:
2 more days of debugging have come up with nothing here. I tried out CPPCheck and resolved all of the warnings it produced, however I've still got this issue.

I CAN confirm that this is not simply a code size issue. It seems it's just coincidence that things started to go awry past 700,000 bytes. I'm trying to figure out how to best proceed under everslick's theory that it is some sort of memory corruption bug, but since I can confirm that my issue appears to be about my code, and is unrelated to the ESP32 environment or setup, I'd understand if closing this thread was appropriate.

shofixti244, 2017-11-13:
Just in conclusion here,

Doing more debugging and research, I've come to the conclusion that the issue that I'm having is the stack size being exceeded. It turns out that Wifi and Bluetooth both take up significant stack space, and my own work with the AWS framework takes up an additional 30k bytes. From what I can tell, I'm exceeding my limit. And this is backed up by some of the other error messages I am receiving while debugging specific points:

`assertion "res == pdTRUE" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./dport_access.c"`

A google search led me here: https://github.com/espressif/arduino-esp32/issues/727 and I figure the problems I was explaining above COULD be related to the stack being overrun.

Thank you all for the assistance. If I can ask one last question... Is it possible to increase the stack size on the ESP32?

closed 2018-04-19 by everslick