espressif/arduino-esp32 #828: "HTTPClient::end(void)" runs are too long

xiruilin, 2017-11-11:
In /libraries/HTTPClient/src/HTTPClient.cpp, function "void HTTPClient::end(void)", line 231:

        if(_tcp->available() > 0) {
            log_d("still data in buffer (%d), clean up.", _tcp->available());
            while(_tcp->available() > 0) {
                _tcp->read();
            }
        }

If the unread data is very large, it will take a long time. If changed to:

        if(_tcp->available() > 0) {
            log_d("still data in buffer (%d), clean up.", _tcp->available());
            _tcp->flush();
            }
        }

You can achieve what you want.

Uksa007, 2018-02-06:
Hi,
I can confirm that I have this issue also, takes around 60 seconds to close the http connection when trying to close a streaming radio station, like http://streaming.shoutcast.com/80sPlanet?lang=en-US
There was no issue with the original code when using an ESP8266, so not sure what has changed.

https://github.com/xiruilin above change to library does resolve the issue for me.

h3ndrik, 2018-02-06:
can we make a pull request for this?

commit e3a5ae43 (2018-03-04): clean up faster (fixes #828) (#1087)

flush tcp buffer instead of reading it byte by byte.
closed 2018-03-04 by me-no-dev