espressif/arduino-esp32 #829: No serial via USB output on ESP32 FireBeetle. Not hardware issue.

barsznica, 2017-11-12:
### Hardware:
Board:							        DFRobot ESP-WROOM-32
Core Installation/update date:			BN: Unknown board, VID: 2341, PID: 00F3
Chip:                                                                SP32D0WDQ6 (revision 0)
IDE name:							Arduino IDE 1.8.4, Win10 Home, IDF component: N/A
Flash Frequency:					        80, 40Mhz
Upload Speed:						9600, 115200, 921600

### Description:

No serial-over-USB displayed on 'Serial Monitor' when using the Espressif 'Boards' :(

What could I be doing wrong? 

I tried the DFRobot Board, [Board](https://git.oschina.net/dfrobot/FireBeetle-ESP32/raw/master/package_esp32_index.json), from [https://www.dfrobot.com/wiki/index.php/FireBeetle_ESP32...](https://www.dfrobot.com/wiki/index.php/FireBeetle_ESP32_IOT_Microcontroller_(Supports_Wi-Fi_%26_Bluetooth)_SKU:_DFR0478#More), and it works!

The Espressif Board setting compiles code all the code I have tried. THANKS HEAPS!!!
The DFRobot Board has difficulties compiling the very few, although different, bits of code I have tried ;(

### Sketch:
```cpp

#include <Arduino.h>

void setup() 
{
  delay(1000);
  Serial.begin(115200);
}
 
void loop() 
{
  Serial.println("Hello from FireBeetle ESP32 board");
  delay(2000);
}
```

### Debug Messages:
```
No errors or warnings, with Max Power at the debug level controls (I be givin' it all I got cap'n).
```

barsznica, 2017-11-12:
Must be this 'boot:0x3' business.
Hmm, what can I do?

barsznica, 2017-11-12:
I have serial out via TX pin. Woohoo!

copercini, 2017-11-21:
Is your pin 0 connected to something?

DzikuVx, 2017-11-24:
I confirm the same issue with 23cf233ecad1121d56a3a93adbb9de2e5889e665 , no nothing on pin 0

After rolling back to 706bf48708e6668077187ba3f5543d0f7b49e132 , everything works fine again.
Hardware: DOit ESP32 dev kit V1

Looks like bug was introduced somewhere between those 2 commits

me-no-dev, 2017-11-24:
pin 0? what is the issue now? no serial output?

DzikuVx, 2017-11-24:
> The same issue
> No serial-over-USB displayed on 'Serial Monitor' when using the Espressif 'Boards' :(

me-no-dev, 2017-11-24:
nothing at all or does it stop before the sketch starts? any bootloader output?

DzikuVx, 2017-11-24:
@me-no-dev bootloader output present,
 no content available over serial from sketch like this, for example
```
#include <Arduino.h>

void setup() 
{
  delay(1000);
  Serial.begin(115200);
}
 
void loop() 
{
  Serial.println("Hello from FireBeetle ESP32 board");
  delay(2000);
}
```

me-no-dev, 2017-11-24:
I had this happen to me once last night, but then it started working an no fail... will investigate further if I can reproduce it again. Asked around IDF team if they have a clue also.

DzikuVx, 2017-11-24:
Like I wrote, rolling back to 706bf48708e6668077187ba3f5543d0f7b49e132 made that work again

me-no-dev, 2017-11-24:
quite an old commit though ;) it's something between last night and a couple of weeks ago

DzikuVx, 2017-11-24:
I needed EEPROM so just checked out to the oldest commit I was able to use

me-no-dev, 2017-11-24:
can you please pull the last change and give it a go?

DzikuVx, 2017-11-24:
@me-no-dev Looks like it helped. Serial working again in 

For the record, bootloader output
```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13076
entry 0x40078ad0
```

I'm not sure it solved problem of @barsznica but mine was solved. Thanks

barsznica, 2017-11-24:
I'll try it out when I get a chance

everslick, 2018-04-19:
New activity in this issue since months, looks as solved. -> closing

closed 2018-04-19 by everslick

iZhangHui, 2018-07-20:
@me-no-dev 
This issue happen again with FireBeetle board running 1.0.0-rc3. 

After push RST button, Serial Monitor only display the message below:

rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download

When replug the usb, nothing output on Serial Monitor.

iZhangHui, 2018-07-20:
@me-no-dev Thanks for your explanation. 
I didn't connect anything to GPIO0. 

I can upload firmware to the board, when uploading finished, it should be re-plug usb power line.
Then, the board works well with led blink sketch.

But after re-plug usb power line,  Serial Monitor display nothing. In my code I add some codes like Serial.println("******") for debug, and the baud rate is same with Serial Monitor.

kalvis84, 2018-08-15:
I have this problem on DF Fire Beetle. After programming, GPIO0 is held low. It looks like a problem with the software on STM chip that is used as UART-USB converter on that boards.

HendrikVE, 2018-08-18:
I have also problems getting serial output on my FireBeetle (V2.0). Nothing is connected to the board but I measured 3.3 V on GPIO 0. At least I've got the same output as @iZhangHui when I used Arduino IDE to flash, but nothing when using make monitor.

kalvis84, 2018-08-19:
Currently I am using separate uart-usb converter and I have to force reset on stm chip to make it pull up gpio0 again.

HendrikVE, 2018-08-19:
Luckily I've found a UART USB converter at home and I can confirm I'm getting serial output with it. Anyway, it would be cool if I wouldn't need additional hardware. For me its working fine without forcing reset btw.

iZhangHui, 2018-08-20:
@HendrikVE How did you wire the UART USB convetor? Thank you.

HendrikVE, 2018-08-20:
I've put TXD to RXD (GPIO 3) and RXD to TXD (GPIO 1). That's all. My converter appears as ttyUSB0 while the board itself is ttyACM0. You can setup ttyACM0 in `make menuconfig` as default serial port and monitor with `ESPPORT=/dev/ttyUSB0 make monitor`. This way you dont need to rebuild every time in cause of different flash/monitor ports. It's also comfortable to have a USB port with single switches for every USB port as the board will be only flashed if the converter is disconnected from USB.

barsznica, 2018-08-20:
RX, TX & "GND", of course!

murali-dotworld, 2021-05-10:
Arduino Serial monitor not showing the output. Use Putty software, it shows the output