espressif/arduino-esp32 #830: ESP32 stalls after waking up from deep sleep

cloudiaworld, 2017-11-12:
### Hardware:
Board:							?ESP32 DEVKIT V1?
Core Installation/update date:			?12/Nov/2017?
IDE name:							?Arduino IDE 1.8.5?
Flash Frequency:					?40Mhz?
Upload Speed:						?115200?

### Description:
The micro stalls after repeatedly sleeping-waking-up. It happens after a non-deterministic number of reboots (sleep - wake up cycles). Sometimes 20, 50, 100 times. I have the same issue in several boards; however, I was not able to reproduce it on this board https://www.adafruit.com/product/3405 (this one works as expected ---never stalls---)

### Sketch:
The behavior I describe can be  reproduced  with the DeepSleep examples 
https://github.com/espressif/arduino-esp32/tree/master/libraries/ESP32/examples/DeepSleep

### Debug Messages:
rst:0x5 (DEEPSLEEP_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:11904
entry 0x40078a3c
Boot number: 41
Wakeup caused by timer
Setup ESP32 to sleep for every 5 Seconds
Going to sleep now
ets Jun  8 2016 00:22:57

rst:0x5 (DEEPSLEEP_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:11904
entry 0x40078a3c
Boot number: 42
Wakeup caused by timer
Setup ESP32 to sleep for every 5 Seconds
Going to sleep now
ets Jun  8 2016 00:22:57

rst:0x5 (DEEPSLEEP_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:11904
entry 0x40078a3c

### Debug Messages:

nsollars, 2017-11-15:
Board:	Wemos Lolin32 v1.0.0
Core Installation/update date:	14/Nov/2017
IDE name:	Arduino IDE 1.8.5
Flash Frequency:	40Mhz
Upload Speed:	115200

I have a sketch that uses the timer wakeup method as per the example,  watching the sketch after a number of hours ive noticed that the RTC_ATTR int counter is being reset ( seems to be at random ), so something is not quite right as it would seem a reset is being called after waking up from deep sleep.

cristianhumelnicu, 2017-11-16:
@nsollars 
What is the maximum deepsleep time you got ??
I also tried to put the microcontroller in sleep but all I did was a deepsleep of 30 minutes maximum. After this period, there is no longer a deepsleep. I use WROOM32 silicon rev.1

cloudiaworld, 2017-11-16:
@nsollars 
I suspected as well that the problem might have to do with the RCT_ATTRs, but seems not to be the case. I have the same symptoms without defining RTC_ATTR variables.

cristianhumelnicu, 2017-11-16:
@nsollars Can u give me more deails about how to sertting or use RTC_ATTR var? 

I just using deepsleep exemple from Arduino without any modification , except deepsleep duration ... 
Working under 30 minutes ... Never wake up if  DS isset over 1h

nsollars, 2017-11-16:
The only value I changed was the duration in seconds,

#define uS_TO_S_FACTOR 1000000  /* Conversion factor for micro seconds to seconds */
#define TIME_TO_SLEEP  10        /* Time ESP32 will go to sleep (in seconds) */

Once my setup loop is done collecting data from sensors etc etc.

the last 2 lines are:

esp_sleep_enable_timer_wakeup(TIME_TO_SLEEP * uS_TO_S_FACTOR);
esp_deep_sleep_start();

The highest wakeup count ive seen was up in the 700's but normal numbers range between 100's to 400's

Now the final aim is to set deepsleep for an hour+ as this project is going to use a lipo battery with a solar charging solution.

Nige

nsollars, 2017-11-16:
Oh and the variable of course

RTC_DATA_ATTR int wakeupCount = 0;

Im using this count as part of a maintenance / monitoring message.

nsollars, 2017-11-16:
One other minor observation that has come up is, I introduced a BME280 temp / humidity / pressure sensor.

I wrapped up the begin for the object like so
if(!bme280.begin() {
   serial print out did not detect the sensor
   while(;;)
}

And im guessing the pull of the reset is glitching the i2c bus enough to cause my code not to detect the sensor.  Hitting the reset button on the lolin32 fixes the issue and normal operations commence.

NIge

robotrovsky, 2017-11-23:
Someone knows the last revision in which deep sleep was working well?

copercini, 2017-11-23:
@robotrovsky Still no, there is a fix in IDF, but the guys are working hard to implement it on arduino too, you can check the status here: https://github.com/espressif/arduino-esp32/issues/796

Will close this issue to  concentrate this bug on https://github.com/espressif/arduino-esp32/issues/796

closed 2017-11-23 by copercini

robotrovsky, 2017-11-23:
@copercini Thank you very much!