espressif/arduino-esp32 #831: SPIFFS.begin() crash in library use

merlinschumacher, 2017-11-12:
### Hardware:
Board:							Wemos Lolin 32 V1.0.0
Core Installation/update date:			12. Nov 17
IDE name:							Arduino IDE
Flash Frequency:					40Mhz/80Mhz
Upload Speed:						115200/921600

### Description:
Trying to use the SPIFFS.begin() inside a library triggers an exception on the ESP32 which leads to a reboot.

### Sketch:
```cpp

//Change the code below by your sketch

#include "test.h"

test test;

void setup() {
    Serial.begin(115200);  
}

void loop() {
  // put your main code here, to run repeatedly:

}
```
### test.cpp
```cpp
#include "test.h"

#include "SPIFFS.h"

test::test()
{

  SPIFFS.begin(true);
}
```

###test.h
```cpp
#ifndef test_h
#define test_h

class test {
  public:
    test();

};

#endif

### Debug Messages:
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:11904
entry 0x40078a3c
/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./queue.c:1438 (xQueueGenericReceive)- assert failed!
abort() was called at PC 0x400845e9 on core 0

Backtrace: 0x40086fb0:0x3ffe3920 0x400870af:0x3ffe3940 0x400845e9:0x3ffe3960 0x400df975:0x3ffe39a0 0x400830a7:0x3ffe39c0 0x40083a6a:0x3ffe39e0 0x40083c83:0x3ffe3a10 0x400df9d1:0x3ffe3a40 0x400dfb3e:0x3ffe3a70 0x400dfb8f:0x3ffe3a90 0x400dffd2:0x3ffe3ab0 0x400e0884:0x3ffe3af0 0x400d0af8:0x3ffe3b70 0x400d0a45:0x3ffe3ba0 0x400d0a96:0x3ffe3bc0 0x400dcc5b:0x3ffe3be0 0x40081238:0x3ffe3c00 0x4008140d:0x3ffe3c30 0x40078892:0x3ffe3c50 0x400788f9:0x3ffe3c80 0x40078a36:0x3ffe3cb0 0x40078b5b:0x3ffe3e70 0x40007c31:0x3ffe3eb0 0x4000073d:0x3ffe3f20

Rebooting...

### Exception Decoder ###

Decoding 24 results
0x40086fb4: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./panic.c line 553
0x400870b3: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./panic.c line 553
0x400845ed: xQueueGenericReceive at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./queue.c line 1992
0x400df475: spi_flash_op_lock at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spi_flash/./cache_utils.c line 56
0x40083003: spi_flash_disable_interrupts_caches_and_other_cpu at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spi_flash/./cache_utils.c line 95
0x400839c6: spi_flash_mmap_pages at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spi_flash/./flash_mmap.c line 143
0x40083bdf: spi_flash_mmap at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spi_flash/./flash_mmap.c line 120
0x400df4d1: load_partitions at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spi_flash/./partition.c line 225
0x400df63e: esp_partition_find at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spi_flash/./partition.c line 225
0x400df68f: esp_partition_find_first at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spi_flash/./partition.c line 225
0x400dfad2: esp_spiffs_init at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spiffs/./esp_spiffs.c line 387
0x400e0384: esp_vfs_spiffs_register at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spiffs/./esp_spiffs.c line 414
0x400d0a48: fs::SPIFFSFS::begin(bool, char const*, unsigned char) at /home/merlin/Arduino/hardware/espressif/esp32/libraries/SPIFFS/src/SPIFFS.cpp line 92
0x400d09ad: test::test() at /tmp/arduino_build_420497/sketch/test.cpp line 8
0x400d09fe: _GLOBAL__sub_I_test at /home/merlin/Arduino/test/test.ino line 3
:  (inlined by) _GLOBAL__sub_I_test at /home/merlin/Arduino/test/test.ino line 15
0x400dc75b: do_global_ctors at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./cpu_start.c line 412 (discriminator 3)
0x40081194: start_cpu0_default at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./cpu_start.c line 335
0x40081369: call_start_cpu0 at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./cpu_start.c line 210

me-no-dev, 2017-11-12:
You can not call it in the class constructor ;) you need to call it later (like begin() which is called in setup())

closed 2017-11-12 by me-no-dev

twisniew1976, 2018-03-28:
@me-no-dev : Why calling SPIFFS.begin is prohibited inside class constructors ?

stickbreaker, 2018-03-28:
@twisniew1976 When the object is defined as a global object, it is initiated before all of the runtime support is ready.  In the class constructor, you can only initialize static values,  `SPIFFS.begin()` sets up dynamic memory buffers.

If you changed the example object from a global object you could have your object execute `SPIFFS.begin()` in its constructor.

```c++
void setup(){
test test;
}
```
Would work just fine,  because `setup()` is called after all of the runtime support is correctly configured.
Or if you need global visibility, create a global pointer and then initialize it in setup;
```c++
test *test;

void setup(){
test = new test;
}
```
This way, it is not initialized until after the runtime support is live.  But, since it is a pointer you would have to access it `test->` instead of `test.`

Chuck.

i4things, 2018-12-26:
There is a proper C+ way to ensure that a class/ function is called before it is used ( usually used when a logging library is used) the ESP32 runtime should use it make to ensure the full runtime is properly initialized before anything else