espressif/arduino-esp32 #836: Can't connect to wifi access point with space in SSID

GeeF, 2017-11-15:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			14/nov/2017
IDE name:							Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:						115200

### Description:
Connection to a WiFi access point fails, if the SSID contains a space.

### Sketch:
use the example client sketch

me-no-dev, 2017-11-16:
I don't think that SSID should contain space at all

GeeF, 2017-11-16:
Have a look at standards.ieee.org/getieee802/download/802.11-2012.pdf
Section 8.4.2.2: "The length of the SSID field is between 0 and 32 octets".
So it is basically just a 32 byte buffer that can be interpreted as UTF-8.

If it can't/shouldn't/wouldn't work for the esp32 implementation, it should at least be documented somewhere so people don't run into it.

me-no-dev, 2017-11-16:
all in all, Arduino calls the API exposed in IDF. Arduino itself does not limit you to "no space in name". I will try to contact the WiFi engineers and see what they can do.

zhangyanjiaoesp, 2017-12-01:
@GeeF Arduino calls the API exposed in IDF. Arduino itself does not limit you to "no space in name". So I tested in IDF,  I can connect to a WiFi access point successfully, if the SSID contains a space.

GeeF, 2017-12-01:
I have just tried again with a different AP and it works. Seems to be something fishy about the other AP I’ve been using. 
Sorry to have wasted your time

Fitap, 2018-03-25:
Hi, I was the same issue like @GeeF and in my case I have setup the tkip flag in wireless security profile.
May helps some one.

Regards.

closed 2018-04-19 by everslick