espressif/arduino-esp32 #837: Missing pgm_read_ptr macro

nsollars, 2017-11-15:
Board:	Wemos Lolin32 v1.0.0
Core Installation/update date:	14/Nov/2017
IDE name:	Arduino IDE 1.8.5
Flash Frequency:	40Mhz
Upload Speed:	115200

-- Description --

as per this conversation with the maintainer of neoGPS.
https://github.com/SlashDevin/NeoGPS/issues/58

The ESP32 core does not have the pgm_read_ptr macro defined yet. This deficiency has been corrected in all the other Arduino cores (e.g. SAM and SAMD). You could move this line in NeoGPS_cfg.h outside of the conditional #if:

//------------------------------------------------------------------------
// The PROGMEM definitions are not correct for Zero, MKR1000 and
//    earlier versions of Teensy boards

//#if defined(ARDUINO_SAMD_MKRZERO) | defined(ARDUINO_SAMD_ZERO) |   \  <-- comment out
//    (defined(TEENSYDUINO) && (TEENSYDUINO < 139))                     <-- comment out
  #undef pgm_read_ptr
  #define pgm_read_ptr(addr) (*(const void **)(addr))
//#endif                                                             <-- comment out

This will define the macro for your build. If this works, it should be reported as a bug to the ESP32 maintainers.

This change did fix the build and the library works as expected for the Ublox m8q.

Nige

everslick, 2018-04-19:
@nsollars Is this issue still relevant with latest master, or can we close it?

nsollars, 2018-04-19:
@everslick   Il update you on the result as soon as possible,  I would imagine this is most likely fixed..

nsollars, 2018-04-19:
@everslick   Yup confirmed this one can be closed, reverted the file and added a define for the lolin32 and no more problem.

Thanks
Nige

closed 2018-04-19 by everslick