espressif/arduino-esp32 #843: Getting error and reboot while using BT adv scan

sciencecow, 2017-11-17:
I'm debugging BLE adv scan code. But esp32 reboot sometimes.

I ported code of this link to arduino-esp32.
https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/ble_eddystone

Any ideas?
The following is log data of my esp32.

Tasks currently running:
CPU 0: Btc_task
CPU 1: IDLE
Task watchdog got triggered. The following tasks did not feed the watchdog in time:
 - IDLE (CPU 0)
Tasks currently running:
CPU 0: Btc_task
CPU 1: IDLE
Task watchdog got triggered. The following tasks did not feed the watchdog in time:
 - IDLE (CPU 0)
Tasks currently running:
CPU 0: Btc_task
CPU 1: IDLE
ASSERT_PARAM(-218959118 0), in arch_main.c at line 293ASSERT_PARAM(-218959118 0), in arch_main.c at line 293ASSERT_PARAM(-218959118 0), in arch_main.c at line 293ASSERT_PARAM(-218959118 0), in arch_main.c at line 293ASSERT_PARAM(-218959118 0), in arch_main.c at line 293ASSERT_PARAM(-218959118 0), in arch_main.c at line 293ASSERT_PARAM(-218959118 0), in arch_main.c at line 293ASSERT_PARAM(-218959118 0), in arch_main.c at line 293Guru Meditation Error of type InstrFetchProhibited occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0xf2f2f2f2  PS      : 0x00060433  A0      : 0x8001a1f5  A1      : 0x3ffc0030  
A2      : 0x00000000  A3      : 0xf2f2f2f2  A4      : 0x00000000  A5      : 0x0000003c  
A6      : 0x3ffb93cc  A7      : 0x3ffb8360  A8      : 0x80019fb8  A9      : 0x000064c0  
A10     : 0xf2f2f2f2  A11     : 0x0000a55a  A12     : 0x00000000  A13     : 0x00000020  
A14     : 0x000022e4  A15     : 0xfffffffc  SAR     : 0x00000004  EXCCAUSE: 0x00000014  
EXCVADDR: 0xf2f2f2f0  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffc  

Backtrace: 0x72f2f2f2:0x3ffc0030 0x4001a1f2:0x3ffc0060 0x40055fa8:0x3ffc0080 0x400818a2:0x3ffc00a0 0x40129671:0x3ffc00c0 0x40019fb5:0x3ffc00e0 0x4001a1f2:0x3ffc0110 0x40055fa8:0x3ffc0130 0x400818a2:0x3ffc0150 0x40129671:0x3ffc0170 0x40019fb5:0x3ffc0190 0x4001a1f2:0x3ffc01c0 0x40055fa8:0x3ffc01e0 0x400818a2:0x3ffc0200 0x40129671:0x3ffc0220 0x40019fb5:0x3ffc0240 0x4001a1f2:0x3ffc0270 0x40055fa8:0x3ffc0290 0x400818a2:0x3ffc02b0 0x40129671:0x3ffc02d0 0x40019fb5:0x3ffc02f0 0x4001a1f2:0x3ffc0320 0x40055fa8:0x3ffc0340 0x400818a2:0x3ffc0360 0x40129671:0x3ffc0380 0x40019fb5:0x3ffc03a0 0x4001a1f2:0x3ffc03d0 0x40055fa8:0x3ffc03f0 0x400818a2:0x3ffc0410 0x40129671:0x3ffc0430 0x40019fb5:0x3ffc0450 0x4001a1f2:0x3ffc0480 0x40055fa8:0x3ffc04a0 0x400818a2:0x3ffc04c0 0x40129671:0x3ffc04e0 0x40019fb5:0x3ffc0500 0x4001a1f2:0x3ffc0530 0x40055fa8:0x3ffc0550 0x400818a2:0x3ffc0570 0x40129671:0x3ffc0590 0x40019fb5:0x3ffc05b0 0x40046683:0x3ffc05e0 0x40047515:0x3ffc0600 0x40048533:0x3ffc0620 0x40048672:0x3ffc0640 0x40083a1e:0x3ffc0660 0x40083e33:0x3ffc0680 0x400855d9:0x3ffc06a0 0x400d6d2e:0x00000000

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:11904
entry 0x40078a3c

brannan, 2018-01-20:
Me too. Did you get any result on this? I'm just getting started with ESP32.

Thanks

everslick, 2018-04-19:
@sciencecow @brannan Is this issue still unsolved for you? If yes, pls provide small code examples and a decoded stack backtrace.

closed 2019-08-15 by stale[bot]