espressif/arduino-esp32 #845: [E][WiFiUdp.cpp:183] endPacket(): could not send data: 12

rtwfroody, 2017-11-18:
### Hardware:
Board:							Adafruit HUZZAH32 Feather
Core Installation/update date:			?11/jul/2017?
IDE name:							IDF component

esp-idf: 7e8c2a9c00a6fb05cd5da306b62c4474a999b1a2
arduino-esp32: 80a7d6084bc0297d371198cbf4d0eab0b98ddd0c

### Description:
Sending UDP packets doesn't work. I get the following error on serial:
[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12

errno 12 is ENOMEM, so maybe I need to configure a buffer somewhere?

Very rarely I do see a handful of bytes come through to my PC, but it effectively never works.

This same code worked fine on an ESP8266.

### Sketch:
```cpp
// size is typically a handful of bytes
size_t UdpPrint::write(const uint8_t *buffer, size_t size)
{
    if (WiFi.status() != WL_CONNECTED) {
        return 0;
    }

    udp.beginPacket(ip_address, port);
    size_t result = udp.write(buffer, size);
    udp.endPacket();
    return result;
}
```

beegee-tokyo, 2017-11-18:
This code works for me on ESP32:
```
/**
 * udpSendMessage
 * Send a message over UDP
 *
 * @param ipAddr
 *				receiver IP address
 * @param udpMsg
 *				payload to be send
 * @param udpPort
 *				port to be used to send
 * @return <code>bool</bool>
 *				true if payload could be sent
 *				false if error occurs
 */
bool udpSendMessage(IPAddress ipAddr, String udpMsg, int udpPort) {
	/** WiFiUDP class for creating UDP communication */
	WiFiUDP udpClientServer;

	// Start UDP client for sending packets
	int connOK = udpClientServer.begin(udpPort);

	if (connOK == 0) {
		Serial.println("UDP could not get socket");
		addMeeoMsg("", "[ERROR] " + digitalTimeDisplaySec() + "UDP could not get socket", true);
		return false;
	}
	int beginOK = udpClientServer.beginPacket(ipAddr, udpPort);

	if (beginOK == 0) { // Problem occured!
		udpClientServer.stop();
		Serial.println("UDP connection failed");
		addMeeoMsg("", "[ERROR] " + digitalTimeDisplaySec() + "UDP connection failed", true);
		return false;
	}
	int bytesSent = udpClientServer.print(udpMsg);
	if (bytesSent == udpMsg.length()) {
		// Serial.println("Sent " + String(bytesSent) + " bytes from " + udpMsg + " which had a length of " + String(udpMsg.length()) + " bytes");
		udpClientServer.endPacket();
		udpClientServer.stop();
		return true;
	} else {
		Serial.println("Failed to send " + udpMsg + ", sent " + String(bytesSent) + " of " + String(udpMsg.length()) + " bytes");
		addMeeoMsg("", "[ERROR] " + digitalTimeDisplaySec() + "Failed to send " + udpMsg + ", sent " + String(bytesSent) + " of " + String(udpMsg.length()) + " bytes", true);
		udpClientServer.endPacket();
		udpClientServer.stop();
		return false;
	}
}
```
As you can see I do a udpClientServer.begin(udpPort) and udpClientServer.stop() everytime I send a package.
I use this for both sending an UDP packet to a single receiver and broadcasting (not multicasting) an UDP packet within the LAN.

everslick, 2017-11-18:
I do NOT begin() stop() on each message I send over UDP, but for me it works as well.

I create the WiFiUDP instance once, call begin() once and reuse it over its whole lifetime.

rtwfroody, 2017-11-18:
I've played around a bit more, and it appears that the first several sends just always have problems. As long as I call begin() again after each error, pretty quickly things settle down and work normally. I tried adding a 10s delay after WiFi has connected, but that doesn't help.

Here's my code that works:
```
size_t UdpPrint::write(const uint8_t *buffer, size_t size)
{
    if (WiFi.status() != WL_CONNECTED) {
        return 0;
    }

    udp.beginPacket(ip_address, port);
    size_t written = udp.write(buffer, size);
    int ok = udp.endPacket();

    if (!ok || written != size) {
        Serial.print("ok=");
        Serial.print(ok);
        Serial.print(", size=");
        Serial.print(size);
        Serial.print(", written=");
        Serial.println(written);
        Serial.println("Reinitialize UDP");
        udp.begin(port);
        return -1;
    }

    return size;
}
```

When I use this, here's what my serial console says:
```
[D][WiFiGeneric.cpp:265] _eventCallback(): Event: 7 - STA_GOT_IP

wifi status: 3
I (8739) wifi: pm start, type:0

setup_radio()
RFM69 radio init OK!
RFM69 radio @915 MHz
setup state
0 15650 [E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
ok=0, size=1, written=1
Reinitialize UDP
[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
ok=0, size=5, written=5
Reinitialize UDP
[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
ok=0, size=1, written=1
Reinitialize UDP
EEPROM starts with: b1 00 15 01
[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
ok=0, size=32, written=32
Reinitialize UDP
1 15680 [E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
ok=0, size=1, written=1
Reinitialize UDP
setup GUI
2 15940   succeeded
3 15940 connecting to heat pump
4 18940   failed
5 18940 connecting to temperature/humidity sensor
6 18940 connecting to RTC
7 19210 temperature_dc=206 set_point_low_dc=177 set_point_high_dc=277
[W][esp32-hal-i2c.c:334] i2cRead(): Ack Error! Addr: 68
8 19220 It is 2165/165/165 165:165:85
9 19220 temperature changed
10 20230 It is 2017/11/18 11:28:58
11 21490 temperature_dc=207 set_point_low_dc=177 set_point_high_dc=277
```

(Note that each line with a number is printed to serial and sent over UDP. So when there's a line like that with no error message it was transmitted successfully.)

korstiaanS, 2018-03-19:
Hi,

I notice that sending UDP works fine in the beginning but after some minutes of doing nothing sending does not work anymore?

beegee-tokyo, 2018-03-19:
I am using [example code](http://desire.giesecke.tk/index.php/2018/02/08/communication-between-devices-that-doesnt-require-a-server-udp-code-examples/) (scroll down for ESP32 code) for UDP broadcasting. The ESP32 is now online for weeks without problems.

korstiaanS, 2018-03-19:
Hi,

I' m not using broadcast.
I' just waiting for a UDP packet to receive and when it comes in I send an OK back.
After a reboot it all works fine but if I wait some minutes and if I then receive a packet my send package is not arriving.
So after some minutes of inactivity the receiving of UDP keeps working but the sending doesn't?
It is like the sending is in a kind of sleep mode?

`void loop() {
    int packetSize = udpEsp.parsePacket();
    if (packetSize) {
        // read the packet into packetBufffer
        int len = udpEsp.read(packetBuffer, UDP_TX_PACKET_MAX_SIZE);
        if (len >= 2) {
            replyinterfaceAddr = udpEsp.remoteIP();
            replyport = udpEsp.remotePort();
            Serial.println(replyinterfaceAddr);
            Serial.println(replyport);
            packetBuffer[len] = 0;
            if (strstr(packetBuffer, "SIGNAL") != NULL) { 
                Serial.println("Signal");
                Serial.println(WiFi.RSSI());
                long rssi = WiFi.RSSI();    
                sprintf(responsebuf, "%iOK", rssi);          
                replyStringWifi(responsebuf, replyinterfaceAddr, replyport);
            }
...

the reply function:

void replyStringWifi(char* answer, IPAddress interfaceAddr, uint16_t port) {
    char responsestring[21] = "";
    strcpy(responsestring, responseprefix);
    strcat(responsestring, answer);
    Serial.println(responsestring);
    udpEsp.beginPacket(interfaceAddr, port);
    udpEsp.write((uint8_t *)responsestring, strlen(responsestring));
    udpEsp.endPacket();
}
`

If I wait some minutes to send a packet (from Packet Sender) it STILL comes in the ESP32 but the reply (sending an UDP packet) does not come to the Packet Sender anymore.

beegee-tokyo, 2018-03-19:
Same command for broadcast and direct send. Difference is only the IP address. For broadcast it is a mask (192.168.0.255) for direct send it is an IP address (192.168.0.12)

korstiaanS, 2018-03-19:
Ok, I made a small test program based on @beegee-tokyo :

```
#include <WiFiUdp.h>
#include <WiFi.h>

bool udpSendMessage(IPAddress ipAddr, String udpMsg, int udpPort) {
  /** WiFiUDP class for creating UDP communication */
  WiFiUDP udpServer;
 
  // Start UDP client for sending packets
  int connOK = udpServer.begin(udpPort);
 
  if (connOK == 0) {
    Serial.println("UDP could not get socket");
    return false;
  }
 
  int beginOK = udpServer.beginPacket(ipAddr, udpPort);
 
  if (beginOK == 0) { // Problem occured!
    udpServer.stop();
    Serial.println("UDP connection failed");
    return false;
  }
 
  int bytesSent = udpServer.print(udpMsg);
  if (bytesSent == udpMsg.length()) {
    udpServer.endPacket();
    udpServer.stop();
    return true;
  } else {
    Serial.println("Failed to send " + udpMsg + ", sent " + String(bytesSent) + " of " + String(udpMsg.length()) + " bytes");
    udpServer.endPacket();
    udpServer.stop();
    return false;
  }
}

/** UDP port */
static const int udpBcPort = 9750;
/** WiFiUDP class for listening to UDP  */
WiFiUDP udpListener;
char esid[20] = "yourssid";
char epass[20] = "yourpassword";
 
void setup(void) {
  // Some initialization stuff
  WiFi.begin(esid, epass);
  Serial.begin(115200);
  // Start UDP listener
  udpListener.begin(udpBcPort);
  int c = 0;
  while ( c < 30 ) {
      if (WiFi.status() == WL_CONNECTED) { 
        break;
      } 
      delay(500);
      c++;
  }
  Serial.println(WiFi.localIP());
  Serial.println("Started");
}
 
void loop(void) {
  // here is your code that is processed in a loop
 
  // Check if message arrived
  int udpMsgLength = udpListener.parsePacket(); 
  if (udpMsgLength != 0) {
    Serial.println("Receiving");
    byte udpPacket[udpMsgLength+1]; // reserve a local buffer
    IPAddress udpIP = udpListener.remoteIP();
 
    // Read the received payload into the local buffer
    udpListener.read(udpPacket, udpMsgLength);
    udpPacket[udpMsgLength] = 0;
 
    udpListener.flush(); // empty UDP buffer for next packet
 
    String debugMsg = "UDP package from ";
    debugMsg += "Sender IP: " + String(udpIP[0]) + "." + String(udpIP[1]) + "." + String(udpIP[2]) + "." + String(udpIP[3]);
    debugMsg += "\n Msg: " + String((char *)udpPacket);
    Serial.println(debugMsg);
   // now send an OK back to Packet Sender. This stops working after > 1 minute of INACTIVITY...
    udpSendMessage(udpIP, "OK", 9750);
  }
}

```

To test:

Fill in SSID and password.
After reboot send UDP packet to the correct IP of the ESP (can see this in Serial monitor) to port 9750.
Make sure that the UDP listener is active on the Packet Sender (port 9750).
Shortly after the reboot, the packet is received on the ESP and it sends an "OK" back to the Packet sender. This works every time.
Then wait 1 minute.
Try again, send a message to ESP.
Message is received on ESP.(check this via serial console).
NO MESSAGE arrives at Packet Sender. It only will work again after a reboot.

Conclusion; UDP sending stops working after 1 minute (or more) of inactivity?

korstiaanS, 2018-03-19:
I opened a new issue #1237 ...

beegee-tokyo, 2018-03-19:
Weird. Just checked my full code on [Github](https://github.com/beegee-tokyo/ESP32-Weatherstation/tree/master/src). And (beside of some JSON parsing) it is the same and working.

I see 3 differences
1. I use broadcast listener and sender
2. listener receives messages around every 10 seconds
3. sending is every 1 minute independant of received messages

Not sure which of the differences makes it working for me.

korstiaanS, 2018-03-19:
Ah, you send every minute, probably the reason it keeps on working ... Try to change this to 3 or 5 minutes..

korstiaanS, 2018-03-19:
My problem is found (but not solved) see #1237

closed 2019-08-15 by stale[bot]

elklaaso, 2023-12-08:
@rtwfroody the error errno 12 is ENOMEM is correct in my case, i freed up memory by moving esp32 sqlite to psram, and the error dissapeared. the error came after i made a large query that took a lot of memory to compile