espressif/arduino-esp32 #846: WiFiClientSecure issue: SSL - A fatal alert message was received from our peer

anteph, 2017-11-18:
### Description:
Hi! I'm experiencing some troubles while using the SSL functionalities of the Arduino core.

Basically, I'm trying to connect to a test online API using HTTPS.
https://jsonplaceholder.typicode.com/posts?userId=1

I'm trying to connect using the validation of the whole certificate chain. I've grabbed the certificate of the root CA using Firefox and exporting the certificate, which can be seen below on the sketch code.

The sketch is based on the example from the WiFiClient secure, I've also removed some stuff to keep things simple. Basically, my first objective is to successfully connect to the server.

Nonetheless, I'm not able to do it and I obtain the error shown below, with the debug level set to verbose:
"SSL - A fatal alert message was received from our peer"

I'm directly using the IP of the server because I was having some problems with the hostname resolution.

Thanks in advance,
Nuno Santos

PS: the example sketch for WiFiClient lacks the "WiFi.h" include, which prevents it from compiling.

### Sketch:
```#include <WiFi.h>
#include <WiFiClientSecure.h>

const char* ssid = "MySSIS";
const char* password =  "MyPass";

const char* test_root_ca= \
"-----BEGIN CERTIFICATE-----\n" \
"MIICiTCCAg+gAwIBAgIQH0evqmIAcFBUTAGem2OZKjAKBggqhkjOPQQDAzCBhTEL\n" \
"MAkGA1UEBhMCR0IxGzAZBgNVBAgTEkdyZWF0ZXIgTWFuY2hlc3RlcjEQMA4GA1UE\n" \
"BxMHU2FsZm9yZDEaMBgGA1UEChMRQ09NT0RPIENBIExpbWl0ZWQxKzApBgNVBAMT\n" \
"IkNPTU9ETyBFQ0MgQ2VydGlmaWNhdGlvbiBBdXRob3JpdHkwHhcNMDgwMzA2MDAw\n" \
"MDAwWhcNMzgwMTE4MjM1OTU5WjCBhTELMAkGA1UEBhMCR0IxGzAZBgNVBAgTEkdy\n" \
"ZWF0ZXIgTWFuY2hlc3RlcjEQMA4GA1UEBxMHU2FsZm9yZDEaMBgGA1UEChMRQ09N\n" \
"T0RPIENBIExpbWl0ZWQxKzApBgNVBAMTIkNPTU9ETyBFQ0MgQ2VydGlmaWNhdGlv\n" \
"biBBdXRob3JpdHkwdjAQBgcqhkjOPQIBBgUrgQQAIgNiAAQDR3svdcmCFYX7deSR\n" \
"FtSrYpn1PlILBs5BAH+X4QokPB0BBO490o0JlwzgdeT6+3eKKvUDYEs2ixYjFq0J\n" \
"cfRK9ChQtP6IHG4/bC8vCVlbpVsLM5niwz2J+Wos77LTBumjQjBAMB0GA1UdDgQW\n" \
"BBR1cacZSBm8nZ3qQUfflMRId5nTeTAOBgNVHQ8BAf8EBAMCAQYwDwYDVR0TAQH/\n" \
"BAUwAwEB/zAKBggqhkjOPQQDAwNoADBlAjEA7wNbeqy3eApyt4jf/7VGFAkK+qDm\n" \
"fQjGGoe9GKhzvSbKYAydzpmfz1wPMOG+FDHqAjAU9JM8SaczepBGR7NjfRObTrdv\n" \
"GDeAU/7dIOA1mjbRxwG55tzd8/8dLDoWV9mSOdY=\n" \
"-----END CERTIFICATE-----\n";

WiFiClientSecure client;

void setup() {
  
  Serial.begin(115200);
  
  Serial.println(test_root_ca);
  delay(100);

  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(1000);
  }

  client.setCACert(test_root_ca);
  
  Serial.println("\nStarting connection to server...");
  if (!client.connect(IPAddress(104,31,86,157), 443))
    Serial.println("Connection failed!");
  else {
    Serial.println("Connected to server!");
    
  }  
}

void loop() {
  
}
```

### Debug Messages:
```
Starting connection to server...
[V][ssl_client.cpp:48] start_ssl_client(): Free heap before TLS 132116
[V][ssl_client.cpp:50] start_ssl_client(): Starting socket
[V][ssl_client.cpp:86] start_ssl_client(): Seeding the random number generator
[V][ssl_client.cpp:95] start_ssl_client(): Setting up the SSL/TLS structure...
[V][ssl_client.cpp:108] start_ssl_client(): Loading CA cert
[V][ssl_client.cpp:143] start_ssl_client(): Setting hostname for TLS session...
[V][ssl_client.cpp:158] start_ssl_client(): Performing the SSL/TLS handshake...
[E][ssl_client.cpp:28] handle_error(): SSL - A fatal alert message was received from our peer
[E][ssl_client.cpp:30] handle_error(): MbedTLS message code: -30592
[E][WiFiClientSecure.cpp:107] connect(): lwip_connect_r: 11
[V][ssl_client.cpp:209] stop_ssl_socket(): Cleaning SSL connection.
Connection failed!
```
### Hardware:
Board:							Generic ESP32 NodeMCU board
Core Installation/update date:			around 1/nov/2017
IDE name:							Arduino IDE

anteph, 2017-11-18:
Just as an additional note, if I use the HTTP client library and pass the CA certificate as additional input of the begin method, it works.

copercini, 2017-11-19:
This is because an SSL certificate is issued for a particular domain name. If the certificate name doesn't match the visited domain, it will cause an error

If you try open https://104.31.86.157/posts?userId=1 in your web browser it also will not connect and show some SSL errors

The strangest thing here is the HTTP client working in this case

copercini, 2017-11-20:
In this particular case, the Cloudflare is blocking access to the site with direct IP in any way, even without SSL

anteph, 2017-11-23:
Hi! Thanks for the feedback.

Indeed I cannot access directly by IP using the browser. I've resorted to hardcoding the IP at the time because I was not being able to get a proper domain name translation, but I did not checked in the browser.

In the case of the HTTP Client, I was using the domain name rather than the IP, since I was not having problems with the name resolution.

So that should be the reason why it works with the HTTP Client. 

I will leave the HTTP code below for future reference. 

```
#include <WiFi.h>
#include <HTTPClient.h>
 
const char* ssid = "yourNetworkName";
const char* password =  "yourNetworkPass";
 
void setup() {
 
  Serial.begin(115200);
  delay(1000);
 
  WiFi.begin(ssid, password); 
 
  while (WiFi.status() != WL_CONNECTED) {
    delay(1000);
    Serial.println("Connecting to WiFi..");
  }
 
  Serial.println("Connected to the WiFi network");
}
 
const char* root_ca= \
"-----BEGIN CERTIFICATE-----\n" \
"MIICiTCCAg+gAwIBAgIQH0evqmIAcFBUTAGem2OZKjAKBggqhkjOPQQDAzCBhTEL\n" \
"MAkGA1UEBhMCR0IxGzAZBgNVBAgTEkdyZWF0ZXIgTWFuY2hlc3RlcjEQMA4GA1UE\n" \
"BxMHU2FsZm9yZDEaMBgGA1UEChMRQ09NT0RPIENBIExpbWl0ZWQxKzApBgNVBAMT\n" \
"IkNPTU9ETyBFQ0MgQ2VydGlmaWNhdGlvbiBBdXRob3JpdHkwHhcNMDgwMzA2MDAw\n" \
"MDAwWhcNMzgwMTE4MjM1OTU5WjCBhTELMAkGA1UEBhMCR0IxGzAZBgNVBAgTEkdy\n" \
"ZWF0ZXIgTWFuY2hlc3RlcjEQMA4GA1UEBxMHU2FsZm9yZDEaMBgGA1UEChMRQ09N\n" \
"T0RPIENBIExpbWl0ZWQxKzApBgNVBAMTIkNPTU9ETyBFQ0MgQ2VydGlmaWNhdGlv\n" \
"biBBdXRob3JpdHkwdjAQBgcqhkjOPQIBBgUrgQQAIgNiAAQDR3svdcmCFYX7deSR\n" \
"FtSrYpn1PlILBs5BAH+X4QokPB0BBO490o0JlwzgdeT6+3eKKvUDYEs2ixYjFq0J\n" \
"cfRK9ChQtP6IHG4/bC8vCVlbpVsLM5niwz2J+Wos77LTBumjQjBAMB0GA1UdDgQW\n" \
"BBR1cacZSBm8nZ3qQUfflMRId5nTeTAOBgNVHQ8BAf8EBAMCAQYwDwYDVR0TAQH/\n" \
"BAUwAwEB/zAKBggqhkjOPQQDAwNoADBlAjEA7wNbeqy3eApyt4jf/7VGFAkK+qDm\n" \
"fQjGGoe9GKhzvSbKYAydzpmfz1wPMOG+FDHqAjAU9JM8SaczepBGR7NjfRObTrdv\n" \
"GDeAU/7dIOA1mjbRxwG55tzd8/8dLDoWV9mSOdY=\n" \
"-----END CERTIFICATE-----\n";
 
void loop() {
 
  if ((WiFi.status() == WL_CONNECTED)) { //Check the current connection status
 
    HTTPClient http;
 
    http.begin("https://jsonplaceholder.typicode.com/posts?userId=1", root_ca); //Specify the URL and certificate
    int httpCode = http.GET();                                                  //Make the request
 
    if (httpCode > 0) { //Check for the returning code
 
        String payload = http.getString();
        Serial.println(httpCode);
        Serial.println(payload);
      }
 
    else {
      Serial.println("Error on HTTP request");
    }
 
    http.end(); //Free the resources
  }
 
  delay(10000);
}
```

Thanks for the help,
Nuno Santos

closed 2017-11-23 by anteph