espressif/arduino-esp32 #848: Exception decoder not decoding for some infrequent low level errors.

lonerzzz, 2017-11-18:
### Hardware:
Board:							ESP-WROOM-32
Core Installation/update date:		2017/11/10
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
Upload Speed:					115200

### Description:
I am getting the following different errors after a number of hours but unfortunately, the Exception decoder doesn't provide details when I copy the back trace in. Normally the back traces can be decoded. Any suggestions from the rest of the information or other settings to enable to point me in a direction for debugging?

Guru Meditation Error of type InstrFetchProhibited occurred on core  1. Exception was unhandled.
Register dump:
PC      : 0x030df48f  PS      : 0x00050033  A0      : 0x030df48f  A1      : 0x3ffd50f0
A2      : 0xf803d03f  A3      : 0xf48f3ffe  A4      : 0x503ffd00  A5      : 0x3ffefc03
A6      : 0x020114d0  A7      : 0x24d14417  A8      : 0x11451249  A9      : 0x44144924
A10     : 0x134d34d1  A11     : 0x49245144  A12     : 0x00000013  A13     : 0x00000001
A14     : 0x00060d23  A15     : 0x3ffd3400  SAR     : 0x0000000c  EXCCAUSE: 0x00000014
EXCVADDR: 0x030df48c  LBEG    : 0x400d294e  LEND    : 0x400d2955  LCOUNT  : 0x00000000

Backtrace: 0x030df48f:0x3ffd50f0 0x030df48c:0x05045040

Guru Meditation Error of type IllegalInstruction occurred on core  1. Exception was unhandled.
Register dump:
PC      : 0x4a28a964  PS      : 0x00050033  A0      : 0x4a28a964  A1      : 0x3ffd50f0
A2      : 0x1d75d4d1  A3      : 0x6dc4d146  A4      : 0xa4d1461a  A5      : 0xd1461c6d
A6      : 0x2f1c69a4  A7      : 0x55449145  A8      : 0x44d14410  A9      : 0x11451151
A10     : 0x44175544  A11     : 0x12550451  A12     : 0x000000ae  A13     : 0x00000001
A14     : 0x00060b23  A15     : 0x3ffd3400  SAR     : 0x0000000c  EXCCAUSE: 0x00000000
EXCVADDR: 0x00000000  LBEG    : 0x400d294e  LEND    : 0x400d2955  LCOUNT  : 0x00000000

Backtrace: 0x4a28a964:0x3ffd50f0 0x4a28a961:0x1c7dc411

Guru Meditation Error of type InstrFetchProhibited occurred on core  1. Exception was unhandled.
Register dump:
PC      : 0x8f40020d  PS      : 0x00050033  A0      : 0x8f40020d  A1      : 0x3ffd50f0
A2      : 0x3ffff863  A3      : 0xfefce3cf  A4      : 0x0134d13e  A5      : 0xf38f3f01
A6      : 0x6d40fff8  A7      : 0x154d9445  A8      : 0x51645145  A9      : 0x74c14517
A10     : 0xd1441655  A11     : 0x45144934  A12     : 0x00000023  A13     : 0x00000001
A14     : 0x00060723  A15     : 0x3ffd3400  SAR     : 0x0000000c  EXCCAUSE: 0x00000014
EXCVADDR: 0x8f40020c  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffff3

Backtrace: 0x4f40020d:0x3ffd50f0 0x4f40020a:0x40ff0834

me-no-dev, 2017-11-18:
probably did not find an address to decode? No output at all?

lonerzzz, 2017-11-18:
None at all following my standard cut and paste of the 'Backtrace:' line

lonerzzz, 2017-11-27:
@me-no-dev Any suggestions on how to proceed on this?

me-no-dev, 2017-11-28:
[this](https://github.com/me-no-dev/EspExceptionDecoder/blob/master/src/EspExceptionDecoder.java#L362) line parses the addresses. Maybe we need to change it to something else for esp32? Can you compile the plugin yourself? Know how?

lonerzzz, 2017-11-28:
@me-no-dev I will have a go of it and let you know if I get stuck. I am very familiar with Java, just not sure of dependencies and the like

me-no-dev, 2017-11-28:
no need to worry about dependencies :) clone the repo to ```[ARDUINO_INSTALL]/tools``` (you will see a couple of other modules there) and run the build script.

lonerzzz, 2017-11-28:
@me-no-dev Unfortunately what I have found is that the xtensa address tool cannot find the address within the code so I am no better off as of yet. Any suggestions on where to go if the toolkit is not being helpful?

At the same time, please add me to the decoder project and I will push an update that extends the address range for decoding and prints out a message in all cases.

tablatronix, 2018-03-07:
Whats the status of this ?

closed 2019-08-15 by stale[bot]