espressif/arduino-esp32 #849: Proposal for analogWrite implementation

bmitov, 2017-11-20:
Hi everyone,

I am experimenting with this simple implementation for analogWrite. Please let me know what you think?
For now I am adding it as a temporary solution in the Visuino libraries, until official support is implemented.
Please feel free to use it in any way you want and include it or something similar in the ESP32 libraries.

### Code:
```cpp
namespace Mitov
{
	uint8_t	CESP32PWMTimers[ 2 ] = {0};
	uint8_t	CESP32PWMPinMap[ 40 ] = {0};
}
	void analogWrite( uint8_t APin, uint16_t AValue )
	{
//		Serial.println( "analogWrite" );
//		Serial.println( AValue );
		if( AValue == 0 || AValue == 255 )
		{
			if( Mitov::CESP32PWMPinMap[ APin ] != 0 )
			{
				ledcWrite( Mitov::CESP32PWMPinMap[ APin ] - 1, AValue );
//				Serial.println( "ledcDetachPin" );
				digitalWrite( APin, (AValue == 255 ) ? HIGH : LOW );
				ledcDetachPin( APin );
				uint8_t AChannel = Mitov::CESP32PWMPinMap[ APin ] - 1;
				Mitov::CESP32PWMTimers[ AChannel >> 3 ] &= ~ ( 1 << ( AChannel & 7 ) );
				Mitov::CESP32PWMPinMap[ APin ] = 0;
			}
			
			digitalWrite( APin, (AValue == 255 ) ? HIGH : LOW );
			return;
		}

		if( Mitov::CESP32PWMPinMap[ APin ] == 0 )
		{
			uint8_t AChannel = 0;
			for( AChannel = 0; AChannel < 16; ++AChannel )
				if( ! ( Mitov::CESP32PWMTimers[ AChannel >> 3 ] & ( 1 << ( AChannel & 7 ) ) ))
					break;

			if( AChannel == 16 )
				return; // Can't allocate channel

			Mitov::CESP32PWMPinMap[ APin ] = AChannel + 1;
			Mitov::CESP32PWMTimers[ AChannel >> 3 ] != ( 1 << ( AChannel & 7 ) );
			ledcSetup( AChannel, 1000, 8 ); // 1KHz 8 bit
			ledcWrite( AChannel, AValue );
			ledcAttachPin( APin, AChannel );
//			Serial.print( "ledcAttachPin: " ); Serial.print( APin ); Serial.print( " " ); Serial.println( AChannel );
			return;
		}

		ledcWrite( Mitov::CESP32PWMPinMap[ APin ] - 1, AValue );
//		Serial.print( "ledcWrite: " ); Serial.print( Mitov::CESP32PWMPinMap[ APin ] - 1 ); Serial.print( " " ); Serial.println( AValue );
	}

imayoda, 2017-12-23:
nice abstraction :)

bmitov, 2017-12-23:
Thank you imayoda! :-) I think it maybe better to implement it with Sigma/Delta. Is there any documentation or explanation on the Sigma/Delta values?

imayoda, 2017-12-27:
hasn't read more than this on the argument, still waiting for my first esp32 breakouts to come :)
https://esp-idf.readthedocs.io/en/v2.0/api/peripherals/sigmadelta.html
kudos

arthurzennig, 2018-09-08:
Hi Mitov, 
your implementation works just fine!

bmitov, 2018-09-08:
Thank you arthurzennig,
I showed it here as proposal to be included in the official library. I would also appreciate any suggestions for improvements. For now this is what I use in the Visuino support for ESP32.
Cheers,
  Boian

ERROPiX, 2018-10-28:
Hi @bmitov 

I have create a small library that provide the classic analogWrite for ESP32, please test it and give me your feedback

Repository https://github.com/ERROPiX/ESP32_AnalogWrite

bmitov, 2018-10-28:
Thank you ERROPiX,
Looks good :-) . Seems like you are using some ideas similar to my implementation :-)

closed 2019-08-15 by stale[bot]