espressif/arduino-esp32 #851: Dallas ds18b20 fails when WIFI_OFF

gbiohazard, 2017-11-20:
### Hardware:
Board:							Heltec ESP32 Lora Oled
Core Installation/update date:			15/11/17
IDE name:							Arduino 1.8.1
Flash Frequency:					80Mhz
Upload Speed:						921600

### Description:

Hi, 
When using the OneWire and DallasTemperature libraries to get ds18b20 measures, it works perfectly fine.
If I include WiFi.h and set the WiFi.mode(WIFI_OFF), the ds18b20 fails about 70% of the time (and gives me -127.00).

When set to Precision 12, the conversion should take about 750ms (and this is what I get when I don't change the WiFi mode. But with WIFI_OFF, the conversion take about 96ms and fails.

No trouble with btStop();
4.7kOhm pullup.

Thanks !!

### Sketch:
#include <OneWire.h>
#include <DallasTemperature.h>
#include <WiFi.h>

#define ONE_WIRE_BUS 22
OneWire oneWire(ONE_WIRE_BUS);
DallasTemperature sensors(&oneWire);

void setup(void)
{
   //WiFi.mode(WIFI_OFF); //<<<<<<<<<<< Uncomment the line en enjoy the fail!

  Serial.begin(115200);
  Serial.println("Dallas Temperature Control Library - Async Demo");
  Serial.println("\nDemo shows the difference in length of the call\n\n");
  sensors.begin();
}

void loop(void)
{ 
  unsigned long start = millis();    
  sensors.requestTemperatures();
  unsigned long stop = millis();
  Serial.print("Time used: ");
  Serial.println(stop - start);
    Serial.print("Temperature: ");
  Serial.println(sensors.getTempCByIndex(0));  
  Serial.println("\n");
  
  delay(5000);
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

rel1ct, 2017-11-20:
[link](https://github.com/espressif/arduino-esp32/issues/755#issuecomment-341684954)

stickbreaker, 2017-12-29:
@gbiohazard I have a modified OneWire library that is working with WiFi here:

[stickbreaker/OneWire](https://github.com/stickbreaker/OneWire)

Chuck.

closed 2019-08-15 by stale[bot]