espressif/arduino-esp32 #852: Implementation of Serial.availableForWrite

Snap1202, 2017-11-20:
is there any chance to implement Serial.availableForWrite for ESP32 under Arduino IDE

me-no-dev, 2017-11-21:
What is the use case for that? It is used in ESP8266 because it uses ring buffer and ISR to send bytes out. ESP32 writes directly to the hardware.

JamesGKent, 2017-11-21:
as it's available in the arduino api: [https://www.arduino.cc/en/Serial/AvailableForWrite](https://www.arduino.cc/en/Serial/AvailableForWrite) would it not be worth implementing for code portability?

me-no-dev, 2017-11-21:
It generally is, but I am not sure how to properly make it work here. We are running multi-cpu chip, so what you get as result and what is the reality once you start writing are two different things. I can return the free bytes in the fifo, but that will not be the reality when you start writing. It might have sent more bytes (thus have more space) or another thread might have started writing to the UART and to have actually less space.

Snap1202, 2017-11-21:
Thanks for the answere. I just tried to port something from arduino to esp and got stuck at this point. I know maybe not the right place to ask but can maybe somehow show me a work around for this command? The originall project has multiple Arduinos connected via rx tx ring so i habe no idea how to get it working without it.If not appropriate to ask here please delete this post.

JamesGKent, 2017-11-21:
@Snap1202 I'm guessing that in your sketch you are porting it is checking whether there is enough space in the buffer to send something, if what you are sending is small in size (10s of bytes?) you could make the possibly faulty assumption that there will always be enough room in the buffer and eliminate those checks?

@me-no-dev i understand what you are saying, but surely it would be better to return the free bytes in the fifo and document (in code for now, would be best to add some proper docs to the repo) that due to the multi threaded nature of the chip that the value returned could change/be incorrect when multiple threads are using the serial interface

Snap1202, 2017-11-21:
its not my Code but here the link to the code i try to port.

https://github.com/voroshkov/Chorus-RF-Laptimer/blob/master/Arduino/chorus_rf_laptimer/sendSerialHex.h

JamesGKent, 2017-11-21:
@Snap1202 in that case all calls to Serial.availableForWrite are to check if there is enough space to send the message, of which the biggest is 14 bytes. so you could probably assume that there will always be space and remove (comment out?) those checks to allow it to work on the ESP32, or alternatively you could implement the method in the core to always return a value bigger than 14 (for this particular use case)

stickbreaker, 2017-11-21:
instead of `availableForWrite()` an atomic write could be implemented `bool success=writeAllOrNothing(char* buff,uint16_t buffLen)` as part of `Serial()`.  If `success` is false, none of the bytes were added to the `Serial` queue.

Chuck.

schlaegerz, 2018-02-09:
I am also stuck at this. I need some form of atomic write. My system right now involves my writing to serial to another board that processes it very slowly, so I always will get to a point where I am not able to write and I need to know when it is full so I can continue processing.

kubark42, 2018-04-04:
I have an inverse problem to many here, where  I want to wait to send until I know that the buffer has been cleared (or almost cleared). This is to give the downstream receiver time to toggle a CTS pin before its RX buffer gets swamped. 

I understand the problems with multiple actors being able to write to the UART, but that doesn't really affect my use case. While I could wait an arbitrary amount of time to resend (if I know the baud rate and message size I can calculate PHY time), this sounds like a brittle workaround.

JamesGKent, 2018-04-04:
@kubark42 that file is for the esp8266 not the esp32, so this issue is not resolved...
it's simpler on the 8266 as there is only one core so there are no race conditions to worry about.

kubark42, 2018-04-04:
Sincerest apologies, I had googled `esp8266` and missed that this was ESP32. Won't happen again!

closed 2019-08-15 by stale[bot]