espressif/arduino-esp32 #853: interrupt causes core panic and reboot

sonic2wb, 2017-11-20:
using a NRF24 radio to receive weather data and display it on an i2c oled display 
using an interrupt on pin 17 

sometimes it will go for hours (sending a read every 60secs)  or some times it will fail with every reading 
Guru Meditation Error: Core  1 panic'ed (Coprocessor exception)
Register dump:
PC      : 0x400d18f5  PS      : 0x00060731  A0      : 0x80080e20  A1      : 0x3ffc0b00
A2      : 0x3ffc3c10  A3      : 0x00000001  A4      : 0x3ffc34d8  A5      : 0x00000000
A6      : 0x3ffc3374  A7      : 0x3ffc1160  A8      : 0x3ffc3470  A9      : 0x3ffc0b90
A10     : 0x3ffc34d8  A11     : 0x3ffc3470  A12     : 0x80085e72  A13     : 0x3ffd1010
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000006  EXCCAUSE: 0x00000004
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff

Backtrace: 0x400d18f5:0x3ffc0b00 0x40080e1d:0x3ffc0bc0 0x40080e1d:0x3ffc0bf0 0x40082625:0x3ffc0c10 0x400d71ce:0x00000000

ecoding 8 results
0x400d18f5: radioInterrupt() at C:\Users\shado\Documents\Arduino\The-Extractor/A20Temperrorcheck.ino line 82
0x400d18f5: radioInterrupt() at C:\Users\shado\Documents\Arduino\The-Extractor/A20Temperrorcheck.ino line 82
0x40080e1d: __onPinInterrupt at C:\Users\shado\Documents\Arduino\hardware\espressif\esp32\cores\esp32/esp32-hal-gpio.c line 211
0x40080e1d: __onPinInterrupt at C:\Users\shado\Documents\Arduino\hardware\espressif\esp32\cores\esp32/esp32-hal-gpio.c line 211
0x40082625: _xt_lowint1 at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./xtensa_vectors.S line 1105
0x400d71ce: i2cWrite at C:\Users\shado\Documents\Arduino\hardware\espressif\esp32\cores\esp32/esp32-hal-i2c.c line 421

----------------------------- Remove above -----------------------------

### Hardware:
Board: DOIt esp32 sevkit 1
Core Installation/update date:			?11/jul/2017?
IDE name:							Arduino IDE
Flash Frequency:					40Mhz
Upload Speed:						115200

```

stickbreaker, 2017-11-21:
It might be related to I2C,  I have a repo here [stickbreaker/arduino-esp32](https://github.com/stickbreaker/arduino-esp32) that you may want to try.  A few people have found that it solves their I2C issues.

BUT, read the readme.md It is not quite a drop in replacement.  a `uint8_t err=Wire.endTransmission(false);` will not return 0 as does the Arduino (AVR). But a new Error Message of 7 (ERROR_CONTINUE).  This is because of the hardware differences between the ESP32 and the AVR chip, a Wire transactions cannot be paused.  So  any Wire transaction that ends with a ReSTART (endTransmission(false) or requestFrom(id,len,false)) is queued until a transaction with a STOP is encountered (endTransmission(), endTransmission(true),requestFrom(id,len,true),requestFrom(id,len)).

There are examples,  I am pushing you to try it as a guinea pig.  I have only had 3 people give me feed back on it so far.  With all of them I have been able to correct any problems, or stability issues they have encountered. I would like enough testing that I could reasonably consider submitting it for inclusion into the main Espressif/arduino-esp32 repo.

Chuck.

sonic2wb, 2017-11-21:
Thanks Will def look into it. I think i solved my own issue. I did not know that ISR's have restrictions of what code they will run and that millis() does not work.

sonic2wb, 2017-11-21:
ok so i disabled i2c and its still crashing with a new error

Decoding 8 results
0x400d1984: radioInterrupt() at C:\Users\shado\Documents\Arduino\The-Extractor/A20Temperrorcheck.ino line 82
0x400d1984: radioInterrupt() at C:\Users\shado\Documents\Arduino\The-Extractor/A20Temperrorcheck.ino line 82
0x40080e21: __onPinInterrupt at C:\Users\shado\Documents\Arduino\hardware\espressif\esp32\cores\esp32/esp32-hal-gpio.c line 211
0x40080e21: __onPinInterrupt at C:\Users\shado\Documents\Arduino\hardware\espressif\esp32\cores\esp32/esp32-hal-gpio.c line 211
0x40082629: _xt_lowint1 at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./xtensa_vectors.S line 1105
0x400d5097: loop() at C:\Users\shado\Documents\Arduino\The-Extractor/A20Temperrorcheck.ino line 82

sonic2wb, 2017-11-21:
another new error:

Decoding 8 results
0x400d0fb8: _flash_cache_start at ?? line ?
0x400d0fb8: _flash_cache_start at ?? line ?
0x400824fd: _xt_coproc_exc at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./xtensa_vectors.S line 999

me-no-dev, 2017-11-21:
ok, where is your code? Did you check power and cable?

sonic2wb, 2017-11-21:
my orginal code was very long and complicated so i created a new sketch that just listens for the ISR and serial prints when its triggered.  i have attached the code and and the latest stack dump
 
Data Packet Recieved
Guru Meditation Error: Core  1 panic'ed (Coprocessor exception)
Register dump:
PC      : 0x400d0c98  PS      : 0x00060c31  A0      : 0x80080d68  A1      : 0x3ffc0b10
A2      : 0x3ffc22a8  A3      : 0x00000001  A4      : 0x00000000  A5      : 0x00000000
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x800d0c98  A9      : 0x3ffc0ba0
A10     : 0x3ffc22a8  A11     : 0x3ffc2274  A12     : 0x3ffc2450  A13     : 0x00000001
A14     : 0x00000008  A15     : 0x00000000  SAR     : 0x00000006  EXCCAUSE: 0x00000004
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff

Backtrace: 0x400d0c98:0x3ffc0b10 0x40080d65:0x3ffc0bd0 0x40080d65:0x3ffc0bf0 0x400824fd:0x3ffc0c10 0x4000bfed:0x00000000

Decoding 8 results
0x400d0c98: radioInterrupt() at C:\Users\shado\Documents\Arduino\isr/isr.ino line 29
0x400d0c98: radioInterrupt() at C:\Users\shado\Documents\Arduino\isr/isr.ino line 29
0x40080d65: __onPinInterrupt at C:\Users\shado\Documents\Arduino\hardware\espressif\esp32\cores\esp32/esp32-hal-gpio.c line 211
0x40080d65: __onPinInterrupt at C:\Users\shado\Documents\Arduino\hardware\espressif\esp32\cores\esp32/esp32-hal-gpio.c line 211
0x400824fd: _xt_lowint1 at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./xtensa_vectors.S line 1105

[isr.zip](https://github.com/espressif/arduino-esp32/files/1490976/isr.zip)

me-no-dev, 2017-11-21:
OK I looked at the code and can say a few things:
1. Interrupt should run in RAM and not call functions that are not in RAM (Serial.X, _radio.X and so on).
2. The whole routine needs to be rewritten to actually be a TASK and the interrupt should set a flag/bits to signal the task that there is data to be read from the radio.

sonic2wb, 2017-11-21:
ok that makes sense .  Here is the example from the NRF24Lite libaray that i was following but I think i understand what ur saying. 
https://github.com/dparson55/NRFLite/blob/master/examples/IRQ_ACK_RX/IRQ_ACK_RX.ino

sonic2wb, 2017-11-22:
@me-no-dev Thank you!   followed your advice and just had the ISR set a flag and then had the code act on that flag and reset the flag once it was acted on. Works perfectly now!  This was the first time ever using an interrupt and had some learning to do, Cheers!

closed 2017-11-22 by sonic2wb

Nickbuster, 2023-04-29:
But then it's not interrupt function.
If only flag has been set via interrupt, code acts based on the flag but when?
For example, you have alarm sound playing and you want to switch it off while it's playing, this is not the solution.