espressif/arduino-esp32 #856: Bluethoot Mac

totovr, 2017-11-22:
### Hardware: MacOsx

Board: dev32 Board						
Core Installation/update date:			november/2017
IDE name:							Arduino IDE
Flash Frequency:					        40Mhz
Upload Speed:						115200

### Description:

esp32 support bluethoot with MacOsx?

me-no-dev, 2017-11-23:
Yes, why wouldn't it?

closed 2017-11-23 by me-no-dev

totovr, 2017-11-23:
Can you provide me an example to prove it? please.

me-no-dev, 2017-11-23:
? I do not get your request? What have you tried so far and how are you trying to connect to the device?

totovr, 2017-11-23:
I'm currently using [this board](http://espressif.com/en/products/hardware/esp32-devkitc/overview) with the Arduino IDE.

I tried this [example](https://github.com/nkolban/ESP32_BLE_Arduino/blob/cb1ab4ea76af17d347cdadcc003d1fe53af12aa7/examples/BLE_client/BLE_client.ino), but OS doesn't recognize the "bluetooth" device.

me-no-dev, 2017-11-23:
this is a BLE client and not server. How are you trying to connect to a device on you Mac? You need some sort of BLE app on your Mac if you want to see BLE devices (and another sketch that is not a Client)