espressif/arduino-esp32 #858: Arduino as IDF-Component Broken Because of FATFS changes in IDF

Curclamas, 2017-11-22:
### Hardware:
Board:							All
IDE name:							IDF component

### Description:
Some FATFS defines have changed naming (see https://github.com/espressif/esp-idf/commit/1fbd3ffac8ddf7e7796fbdece9b9ebe664f57761) now arduino can't be compiled against latest IDF. 

### Debug Messages:
```
AR build/spiffs/libspiffs.a
/home/ubuntu/workspace/esp/logger-firmware/components/arduino/libraries/SD/src/sd_diskio.cpp:60:31: error: '_VOLUMES' was not declared in this scope
 static ardu_sdcard_t* s_cards[_VOLUMES] = { NULL };
                               ^
/home/ubuntu/workspace/esp/logger-firmware/components/arduino/libraries/SD/src/sd_diskio.cpp: In function 'bool sdWait(uint8_t, int)':
/home/ubuntu/workspace/esp/logger-firmware/components/arduino/libraries/SD/src/sd_diskio.cpp:72:16: error: 's_cards' was not declared in this scope
         resp = s_cards[pdrv]->spi->transfer(0xFF);
                ^
/home/ubuntu/workspace/esp/logger-firmware/components/arduino/libraries/SD/src/sd_diskio.cpp: In function 'void sdStop(uint8_t)':
/home/ubuntu/workspace/esp/logger-firmware/components/arduino/libraries/SD/src/sd_diskio.cpp:80:5: error: 's_cards' was not declared in this scope
     s_cards[pdrv]->spi->write(0xFD);
     ^
/home/ubuntu/workspace/esp/logger-firmware/components/arduino/libraries/SD/src/sd_diskio.cpp: In function 'void sdDeselectCard(uint8_t)':
/home/ubuntu/workspace/esp/logger-firmware/components/arduino/libraries/SD/src/sd_diskio.cpp:85:28: error: 's_cards' was not declared in this scope
     ardu_sdcard_t * card = s_cards[pdrv];
                            ^
/home/ubuntu/workspace/esp/logger-firmware/components/arduino/libraries/SD/src/sd_diskio.cpp: In function 'bool sdSelectCard(uint8_t)':
/home/ubuntu/workspace/esp/logger-firmware/components/arduino/libraries/SD/src/sd_diskio.cpp:91:28: error: 's_cards' was not declared in this scope
     ardu_sdcard_t * card = s_cards[pdrv];
                            ^
/home/ubuntu/workspace/esp/logger-firmware/components/arduino/libraries/SD/src/sd_diskio.cpp: In function 'char sdCommand(uint8_t, char, unsigned int, unsigned int*)':
/home/ubuntu/workspace/esp/logger-firmware/components/arduino/libraries/SD/src/sd_diskio.cpp:100:28: error: 's_cards' was not declared in this scope
     ardu_sdcard_t * card = s_cards[pdrv];
                            ^
/home/ubuntu/workspace/esp/logger-firmware/components/arduino/libraries/SD/src/sd_diskio.cpp: In function 'bool sdReadBytes(uint8_t, char*, int)':
/home/ubuntu/workspace/esp/logger-firmware/components/arduino/libraries/SD/src/sd_diskio.cpp:169:28: error: 's_cards' was not declared in this scope
     ardu_sdcard_t * card = s_cards[pdrv];
                            ^
/home/ubuntu/workspace/esp/logger-firmware/components/arduino/libraries/SD/src/sd_diskio.cpp: In function 'char sdWriteBytes(uint8_t, const char*, char)':
/home/ubuntu/workspace/esp/logger-firmware/components/arduino/libraries/SD/src/sd_diskio.cpp:187:28: error: 's_cards' was not declared in this scope
     ardu_sdcard_t * card = s_cards[pdrv];
                            ^
/home/ubuntu/workspace/esp/logger-firmware/components/arduino/libraries/SD/src/sd_diskio.cpp: In function 'bool sdReadSector(uint8_t, char*, long long unsigned int)':
/home/ubuntu/workspace/esp/logger-firmware/components/arduino/libraries/SD/src/sd_diskio.cpp:219:50: error: 's_cards' was not declared in this scope
         if (!sdCommand(pdrv, READ_BLOCK_SINGLE, (s_cards[pdrv]->type == CARD_SDHC) ? sector : sector << 9, NULL)) {
                                                  ^
/home/ubuntu/workspace/esp/logger-firmware/components/arduino/libraries/SD/src/sd_diskio.cpp: In function 'bool sdReadSectors(uint8_t, char*, long long unsigned int, int)':
/home/ubuntu/workspace/esp/logger-firmware/components/arduino/libraries/SD/src/sd_diskio.cpp:240:52: error: 's_cards' was not declared in this scope
         if (!sdCommand(pdrv, READ_BLOCK_MULTIPLE, (s_cards[pdrv]->type == CARD_SDHC) ? sector : sector << 9, NULL)) {
                                                    ^
/home/ubuntu/workspace/esp/logger-firmware/components/arduino/libraries/SD/src/sd_diskio.cpp: In function 'bool sdWriteSector(uint8_t, const char*, long long unsigned int)':
/home/ubuntu/workspace/esp/logger-firmware/components/arduino/libraries/SD/src/sd_diskio.cpp:275:51: error: 's_cards' was not declared in this scope
         if (!sdCommand(pdrv, WRITE_BLOCK_SINGLE, (s_cards[pdrv]->type == CARD_SDHC) ? sector : sector << 9, NULL)) {
                                                   ^
/home/ubuntu/workspace/esp/logger-firmware/components/arduino/libraries/SD/src/sd_diskio.cpp: In function 'bool sdWriteSectors(uint8_t, const char*, long long unsigned int, int)':
/home/ubuntu/workspace/esp/logger-firmware/components/arduino/libraries/SD/src/sd_diskio.cpp:304:28: error: 's_cards' was not declared in this scope
     ardu_sdcard_t * card = s_cards[pdrv];
                            ^
/home/ubuntu/workspace/esp/logger-firmware/components/arduino/libraries/SD/src/sd_diskio.cpp: In function 'DSTATUS ff_sd_initialize(uint8_t)':
/home/ubuntu/workspace/esp/logger-firmware/components/arduino/libraries/SD/src/sd_diskio.cpp:457:28: error: 's_cards' was not declared in this scope
     ardu_sdcard_t * card = s_cards[pdrv];
                            ^
/home/ubuntu/workspace/esp/logger-firmware/components/arduino/libraries/SD/src/sd_diskio.cpp: In function 'DSTATUS ff_sd_status(uint8_t)':
/home/ubuntu/workspace/esp/logger-firmware/components/arduino/libraries/SD/src/sd_diskio.cpp:573:12: error: 's_cards' was not declared in this scope
     return s_cards[pdrv]->status;
            ^
/home/ubuntu/workspace/esp/logger-firmware/components/arduino/libraries/SD/src/sd_diskio.cpp: In function 'DRESULT ff_sd_read(uint8_t, uint8_t*, DWORD, UINT)':
/home/ubuntu/workspace/esp/logger-firmware/components/arduino/libraries/SD/src/sd_diskio.cpp:578:28: error: 's_cards' was not declared in this scope
     ardu_sdcard_t * card = s_cards[pdrv];
                            ^
/home/ubuntu/workspace/esp/logger-firmware/components/arduino/libraries/SD/src/sd_diskio.cpp: In function 'DRESULT ff_sd_write(uint8_t, const uint8_t*, DWORD, UINT)':
[... 35 lines cut ...]
/home/ubuntu/workspace/esp/logger-firmware/components/arduino/libraries/SD/src/sd_diskio.cpp: In function 'uint32_t sdcard_num_sectors(uint8_t)':
/home/ubuntu/workspace/esp/logger-firmware/components/arduino/libraries/SD/src/sd_diskio.cpp:750:28: error: 's_cards' was not declared in this scope
     ardu_sdcard_t * card = s_cards[pdrv];
                            ^
/home/ubuntu/workspace/esp/logger-firmware/components/arduino/libraries/SD/src/sd_diskio.cpp:751:16: error: '_VOLUMES' was not declared in this scope
     if(pdrv >= _VOLUMES || card == NULL){
                ^
/home/ubuntu/workspace/esp/logger-firmware/components/arduino/libraries/SD/src/sd_diskio.cpp: In function 'uint32_t sdcard_sector_size(uint8_t)':
/home/ubuntu/workspace/esp/logger-firmware/components/arduino/libraries/SD/src/sd_diskio.cpp:759:16: error: '_VOLUMES' was not declared in this scope
     if(pdrv >= _VOLUMES || s_cards[pdrv] == NULL){
                ^
/home/ubuntu/workspace/esp/logger-firmware/components/arduino/libraries/SD/src/sd_diskio.cpp:759:28: error: 's_cards' was not declared in this scope
     if(pdrv >= _VOLUMES || s_cards[pdrv] == NULL){
                            ^
/home/ubuntu/workspace/esp/logger-firmware/components/arduino/libraries/SD/src/sd_diskio.cpp: In function 'sdcard_type_t sdcard_type(uint8_t)':
/home/ubuntu/workspace/esp/logger-firmware/components/arduino/libraries/SD/src/sd_diskio.cpp:767:28: error: 's_cards' was not declared in this scope
     ardu_sdcard_t * card = s_cards[pdrv];
                            ^
/home/ubuntu/workspace/esp/logger-firmware/components/arduino/libraries/SD/src/sd_diskio.cpp:768:16: error: '_VOLUMES' was not declared in this scope
     if(pdrv >= _VOLUMES || card == NULL){
                ^
make[1]: *** [libraries/SD/src/sd_diskio.o] Error 1
make[1]: *** Waiting for unfinished jobs....
CC build/xtensa-debug-module/trax.o
```

atanisoft, 2017-11-22:
_VOLUMES has been renamed to FF_VOLUMES as part of https://github.com/espressif/esp-idf/commit/8e180f6155b4fe3b18818f3b498d2991c44d2c13.

Curclamas, 2017-11-22:
Yes, I tried this locally and it works. Yet I'm reluctant to propose a PR because I am sure @me-no-dev will take care of this during the regular idf-update process. Just wanted to note this down here, so it gets tracked :)

me-no-dev, 2017-11-23:
yup :) will bake a new release

everslick, 2018-04-19:
I think I can close this, no?

closed 2018-04-19 by everslick