espressif/arduino-esp32 #859: Battery powered lolin32 deep sleep issue with ext1_wakeup

henricazottes, 2017-11-22:
### Hardware:
Board:							Wemos lolin32-lite
Core Installation/update date:			22/11/17
IDE name:							VSCODE Platform.io
Flash Frequency:					40Mhz
Upload Speed:						115200

### Description:
I'm testing the deepsleep modes just to mesure some consumption data. To do so, I have a program that blink the builtin led twice before going to sleep. Powered with the usb cable, everything is fine. But when I power the board with the battery connector, it just won't go to sleep.

Notice that the same program used with ``esp_deep_sleep_enable_ext0_wakeup`` works well, here I'm trying the ``esp_deep_sleep_enable_ext1_wakeup`` method instead, to see if it consumes less power.

Note: I added a pulldown resitor on GPIO0 and I mesured a fixed 0v on GPIO0.

### Sketch:
```cpp
#include <Arduino.h>

uint64_t mask = 1;

void print_wakeup_reason(){
  esp_deep_sleep_wakeup_cause_t wakeup_reason;

  wakeup_reason = esp_deep_sleep_get_wakeup_cause();

  switch(wakeup_reason)
  {
    case 1  : Serial.println("Wakeup caused by external signal using RTC_IO"); break;
    case 2  : Serial.println("Wakeup caused by external signal using RTC_CNTL"); break;
    case 3  : Serial.println("Wakeup caused by timer"); break;
    case 4  : Serial.println("Wakeup caused by touchpad"); break;
    case 5  : Serial.println("Wakeup caused by ULP program"); break;
    default : Serial.println("Wakeup was not caused by deep sleep"); break;
  }
}

void setup() {
    Serial.begin(9600);
    Serial.println("I waked up !");
    print_wakeup_reason();
    /* set up RTC_IO for GPIO12 */
    esp_deep_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_AUTO);
    /* disable pullup resister */
    gpio_pullup_dis(GPIO_NUM_0);
    //pinMode(0, PULLDOWN);
    /* enable pulldown resister */
    gpio_pulldown_en(GPIO_NUM_0);
    /* anable deep sleep with external wakeup
    and input level which will trigger wakeup is LOW on GPIO12 */
    esp_deep_sleep_enable_ext1_wakeup(mask, ESP_EXT1_WAKEUP_ANY_HIGH);
    pinMode(LED_BUILTIN, OUTPUT);
}

void loop() {
    // put your main code here, to run repeatedly:
    digitalWrite(LED_BUILTIN, HIGH);
    delay(1000);
    digitalWrite(LED_BUILTIN, LOW);
    delay(1000);
    digitalWrite(LED_BUILTIN, HIGH);
    delay(1000);
    digitalWrite(LED_BUILTIN, LOW);
    delay(1000);
    Serial.println("Bring ESP to sleep mode until get external signal");
    esp_deep_sleep_start();
}
```

- If I plug the battery while powered in with USB, and then remove the USB, it stays in deepsleep
- If I plug the USB while powered with the battery, it goes in deepsleep

Any idea ? :)

And it's off topic but is a 0.87mA in deepsleep and 43mA in waken state a normal consumption ? :)

me-no-dev, 2017-11-23:
pulled down GPIO0? this would boot into download mode!

henricazottes, 2017-11-24:
Hum, maybe that's the problem. But why does it work with USB then ?

henricazottes, 2017-11-25:
Ok, I tried with GPIO4 and it works great, thanks :)

But no idea why it was working when powered with usb :/

closed 2017-11-25 by henricazottes