espressif/arduino-esp32 #860: Installation issue:  get.exe UnicodeDecodeError

Cpt-Hook, 2017-11-22:
After cloning the repository I exected get.exe and this is the error message.
I have no idea what to do now.

System: Windows, Info: Windows-7-6.1.7600-SP0
Platform: i686-mingw32
Traceback (most recent call last):
  File "get.py", line 138, in <module>
    get_tool(tool)
  File "get.py", line 76, in get_tool
    local_path = dist_dir + archive_name
UnicodeDecodeError: 'ascii' codec can't decode byte 0xf8 in position 12: ordinal not in range(128)
Failed to execute script get

copercini, 2017-11-22:
Your folder or file name has any ⊘ somewhere?
[image]

Cpt-Hook, 2017-11-22:
Oh this is possible, I am from the Czech republic, so in my username there is a "ř" symbol. Since I cannot simply change my username in win10 (I think at least), is there a way to at least change install folder of esp32 files? ESP8266 installed its files somewhere else if I remember correctly.

me-no-dev, 2017-11-23:
done ;)

closed 2017-11-23 by me-no-dev

Cpt-Hook, 2017-11-23:
Thank you for a quick fix, amazing.

me-no-dev, 2017-11-23:
:) you welcome