espressif/arduino-esp32 #861: InstrFetchProhibited/IllegalInstruction on X509 cert verification

tedder, 2017-11-23:
### Hardware:
Board:							Sparkfun dev board, DOIT kit
Core Installation/update date:		10 Nov 2017
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:					115200, 921600

### libraries
* pubsubclient: 2.6.0 (current)
* esp32 commit 80a7d6084bc0297d371198cbf4d0eab0b98ddd0c from Nov 10 2017, verified it includes the [larger memory/stack size](https://github.com/espressif/arduino-esp32/commit/4495659ac53bcdc4f8e59d6918b3e472903d63a7#diff-9b0caa90c75b300921be10a08d10ff95)

### Description:
X509 certificate connections crash the board with `InstrFetchProhibited` (sparkfun) or `IllegalInstruction` (DOIT). It seems maybe this is a cross-core communication issue, given that wifi is on core0 and the code runs on core1?

I'm not confident with opening this here nor in Pubsubclient, but I think it fits here much better.

I don't think it's a WDT issue, based on the trace.

Noting #657, as shown in the commit above it isn't the small mem/stack issue in there, also noting [pubsubclient #84](https://github.com/knolleary/pubsubclient/issues/84). Both are similar but I don't think they have the same root cause.

### Sketch:
Snippet. I can give a full example if it helps.
```
    espClient.setCertificate(__9a4_public_bin_key);
    espClient.setPrivateKey(__9a4private_bin_key);
    client.setServer(mqtt_server, 8883);

  while (!client.connected()) {
    Serial.println("Attempting MQTT connection...");
    // Create a random client ID
    String clientId = "ESP32Client-";
    clientId += String(random(0xffff), HEX);
    Serial.print("clientid: ");
    Serial.println(clientId.c_str());
    // Attempt to connect
    if (client.connect(clientId.c_str())) { // crashes here (this is line 65)
```

### Debug Messages:
Sparkfun doesn't have the debug menu, but the DOIT does. However I didn't see any more debugging in the logs even with "core debug level=debug".

### errors

first, the decoded exception:
```
Decoding 8 results
0x400d2120: PubSubClient::readPacket(unsigned char*) at /Users/ted/Documents/Arduino/libraries/pubsubclientdiegopx/src/PubSubClient.cpp line 706
0x400d23a8: PubSubClient::connect(char const*, char const*, char const*, char const*, unsigned char, bool, char const*) at /Users/ted/Documents/Arduino/libraries/pubsubclientdiegopx/src/PubSubClient.cpp line 706
0x400d2436: PubSubClient::connect(char const*) at /Users/ted/Documents/Arduino/libraries/pubsubclientdiegopx/src/PubSubClient.cpp line 706
0x400d1e8b: reconnect() at /Users/ted/Documents/Arduino/esp32_mqtt/esp32_mqtt.ino line 65
0x400d1feb: loop() at /Users/ted/Documents/Arduino/esp32_mqtt/esp32_mqtt.ino line 151
0x40134e5c: loopTask(void*) at /Applications/Arduino_183.app/Contents/Java/hardware/espressif/esp32/cores/esp32/main.cpp line 18 (discriminator 1)
```

and the guru exception:
```
Guru Meditation Error of type InstrFetchProhibited occurred on core  1. Exception was unhandled.
Register dump:
PC      : 0x0d2dfc40  PS      : 0x00060430  A0      : 0x800d2123  A1      : 0x3ffd5c10  
A2      : 0x3ffc32e8  A3      : 0x3ffd5c32  A4      : 0x0d2dfc40  A5      : 0x000031af  
A6      : 0x3f401079  A7      : 0x3f401c03  A8      : 0x800d20c9  A9      : 0x3ffd5bf0  
A10     : 0x3ffc33a4  A11     : 0x3ffc32ec  A12     : 0xffff877f  A13     : 0x000031af  
A14     : 0x3f401079  A15     : 0x0000001d  SAR     : 0x00000008  EXCCAUSE: 0x00000014  
EXCVADDR: 0x0d2dfc40  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace: 0x0d2dfc40:0x3ffd5c10 0x400d2120:0x3ffd5c30 0x400d23a8:0x3ffd5c70 0x400d2436:0x3ffd5cb0 0x400d1e8b:0x3ffd5ce0 0x400d1feb:0x3ffd5d20 0x40134e5c:0x3ffd5d60
```

copercini, 2017-11-23:
>     espClient.setCertificate(__9a4_public_bin_key);

try with your certificate, not the public key, also enable verbose debug level to see what is happening

tedder, 2017-11-23:
I actually swapped the cert, just didn't rename the variable, in the above. Here's the first few lines showing that:

    const char* __9a4_public_bin_key = \
      "-----BEGIN CERTIFICATE-----\n" \

For verbose debugging, you're talking about tools->core debug level, yes? I put notes for that above. If there's some other way, let me know.

copercini, 2017-11-23:
> I actually swapped the cert, just didn't rename the variable, in the above. 
ok

> For verbose debugging, you're talking about tools->core debug level, yes? 
yep, I think just "ESP32 Dev Module" have this option

tedder, 2017-11-26:
FWIW, I read the diffs of #865 and reran. Still failing.

```Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited)
. Exception was unhandled.
Register dump:
```

everslick, 2018-04-19:
@tedder Is this issue still unsolved for you? Do you feel this is a bug in the Arduino core? Can we close this?

closed 2019-08-15 by stale[bot]