espressif/arduino-esp32 #863: Brownout during wifi startup

tehKaiN, 2017-11-23:
### Hardware:
Board:							ESP-WROOM-32 + breakout board
Core Installation/update date:			8/jun/2016
IDE name:							Platform.io
Flash Frequency:					?40Mhz?
Upload Speed:						115200

### Description:
I'm developing ESP32-based module for board which has 500mA 3v3 power supply.

After using code attached below, each time during WiFi startup I get brownout. On scope there is quite big voltage drop. This doesn't happen on a breadboard with larger capacitor attached (470uF), so you could assume that's the problem, however AT firmware from https://github.com/espressif/esp32-at works flawlessly under same electrical conditions and starts WiFi without any problems. Unfortunately this firmware is closed source, so my question is: what can I do differently to reduce init voltage drop to have same results as in AT firmware?

I've tried `esp_wifi_set_max_tx_power` but it works only after `esp_wifi_start()` but then it's too late to make any difference.

Also, at first I've tried using Arduino functions for Wifi, but they result in brownouts too.

### Sketch:
```cpp

#include <Arduino.h>
#include "esp_wifi.h"

#define SSID "ESP32AP"

void setup() {
	wifi_init_config_t wifiInitializationConfig = WIFI_INIT_CONFIG_DEFAULT();

	esp_wifi_init(&wifiInitializationConfig);

	esp_wifi_set_storage(WIFI_STORAGE_RAM);

	esp_wifi_set_mode(WIFI_MODE_AP);

	wifi_config_t ap_config;
	strcpy((char *)ap_config.ap.ssid, SSID);
	ap_config.ap.channel = 0;
	ap_config.ap.authmode = WIFI_AUTH_OPEN;
	ap_config.ap.ssid_hidden = 0;
	ap_config.ap.max_connection = 1;
	ap_config.ap.beacon_interval = 100;

	esp_wifi_set_config(WIFI_IF_AP, &ap_config);

	esp_wifi_start();
}

void loop()
{
}
```

### Debug Messages:
```
ets Jun  8 2016 00:22:57
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1                         
load:0x3fff0010,len:4                     
load:0x3fff0014,len:588                    
load:0x40078000,len:0                
load:0x40078000,len:9768                          
entry 0x400789b8
```

me-no-dev, 2017-11-23:
The Arduino WIFi.* call the same IDF functions above, so difference will not matter. In AT probably the power is lowered in another way.

I can advise you to do two things:
- get better LDO that can handle more current so it does not lead to such voltage drop
- submit an issue in IDF's repository, so engineers can look at the AT firmware and see what is different (the firmware might also be compiled with older IDF/sdkconfig and not having the brownout code inside)

PRO2XY, 2017-11-27:
Hey! I was facing the same issue with brownout detector restarting the ESP32. 
I'm pretty sure it might be a hardware issue, even though I have ruled out the power supply capacity, solder joints, capacitors, etc. I am thinking maybe it's some bad solder joint inside the ESP-WROOM-32 module itself. I intend to take the metal shield off and check for the same, and if nothing else, then solder the VDD3P3_RTC up to 3V3 pad on the module directly. Or maybe some other black magic. 

But, until then I used this as a workaround, and thought it may be of help to you. 
`#include "soc/soc.h"`
`#include "soc/rtc_cntl_reg.h"`
`void setup(){`
`    WRITE_PERI_REG(RTC_CNTL_BROWN_OUT_REG, 0); //disable brownout detector`
`   .... other stuff ...`

Needless to say that this will disable the brownout detector completely, so if you do have power related issue you will just get GURU MEDITATION ERROR(s) of sorts. (I wonder how the name came to be so..!)

Redferne, 2017-11-28:
Amiga 4ever! 😎

tehKaiN, 2017-12-01:
@PRO2XY That solved a problem for me. Thanks!

closed 2017-12-01 by tehKaiN

Bartimaeus-, 2018-04-01:
I had a similar problem when trying to use the WiFi when powered from USB on a dev board that had inputs for USB power and 12v. The inputs went to the same LDO through some OR'ing diodes:
[image]
When the board was powered from 5v, there was enough of a drop across the OR'ing diode to cause the 3.3v output to dip to 2.3v:
[image]

If your board has a similar setup then choosing different diodes could fix the problem, but as @me-no-dev suggested getting a better LDO (or better yet a switching power supply) is a  better long term solution. My short term solution was to power the board from a bench supply set to 5.5v so the drop across the diode wouldn't affect the output of the LDO.

The workaround posted by @PRO2XY is not an actual fix to this problem, it just hides some of the symptoms. Turning off the brownout detector is like taking the batteries out of a carbon monoxide alarm because you're getting a headache. There are all sorts of unpredictable behavior that can start showing up in a brownout condition that may not be caught be other alarm functions (that's why the brownout detection exists). Just a word of caution for those who might not know any better reading this thread.

what78what, 2018-04-08:
@PRO2XY, great post, thanks! Helped.

PRO2XY, 2018-04-09:
What @Bartimaeus- has written about BOD is true. It is only going to mask the problem and let you go a little further. It is not going to solve anything at all!
I can say with almost 100% certainty that anyone looking for a solution in this thread has an issue with the power supply circuit, and most probably it is what @Bartimaeus- noted in his comment.

szuercher, 2018-04-20:
If you soldered by your self, also check the GND connection, they are hard to solder. Soldering GND on the thermal pad solve my problem.

bobesp32, 2018-08-02:
I HAD THE SAME EXACT PROBLEM, MY FIX WAS EASY, ALL I DID WAS USE A NEW (AND BETTER) USB CABLE - MY PROBLEM WAS GONE!
HOPE THIS HELPS.

cybergogo, 2018-10-04:
Hi all,
just wanted to share my experience. 
i'm using an ESP-32S, classic, with the white breakout PCB which only has 2 buttons and resistors. 
Had huge difficulty to make it work, crashing at each wifi init. 
Was using an LM2596 voltage regulator, no luck. Added various capacitors, no luck.
In the end, an AMS1117-3.3v solved the issue. maybe the variable LM2596 was too variable. I don't know, but it works now...

philbowles, 2018-10-28:
@PRO2XY : The Guru Meditation is an error notice displayed by early versions of the Commodore Amiga computer when they crashed. It is analogous to the "Blue Screen of Death" in Microsoft Windows operating systems, or a kernel panic in Unix

PRO2XY, 2018-10-29:
@philbowles 
How interesting! Thanks :+1:

Erik84750, 2018-11-20:
I use a bare-bones ESP32 WROOM with 0.1mm enamelled copper soldered to Vcc, GND, TX, RX, EN and IO00, Arduino IDE and boards installed with in Preferences Additional Board Manager https://dl.espressif.com/dl/package_esp32_index.json added.
Power supply is a lab supply.
Brown out detection prevented anything with wifi to start up after download. Added 3x 470uF caps, no help.
What did help:
1. PRO2XY advise of Nov 27 2017
2. increase voltage from 3.20V to 3.42V (3.30V not enough)
But at issue too may be the fact that this test setup is on a breadboard, so ground and supply wiring are not ideal.
Anyway, at issue is probably not actually thee brown out detection by itself but the underlying causes of inadequate power supply wiring.

stickbreaker, 2018-11-20:
I am totally surprised it wold boot at all with wire that small.
here is a table of current capacity of small wires used in model railroad [here](http://www.sumidacrossing.org/LayoutElectricity/MRRWire/WireAmpacity/)
[image]

this table recommends wire larger than 28ga.  The Esp32 can use 500mA when radio is active.

Chuck.

Erik84750, 2018-11-20:
Well I should have known better Chuck! Thanks for that input.

gittyhubble, 2018-12-02:
I had the same problem with ESP32 WROOM-32. Min voltage check with Keysight DMM showed some 40mV. The problem was instantly solved when I added a 47uF cap between 3.3V and GND near the ESP32 pin 1 and 2.

rkistner, 2018-12-30:
When using one of the dev boards powered by USB, the issue could be caused by a 0.5V drop over a diode between USB and Vin. This is similar to the case described by @Bartimaeus. If an external power supply won't be used, this diode is optional and can be bypassed, which should sole this issue.

Some more details here: https://github.com/nkolban/esp32-snippets/issues/168#issuecomment-437592150

gulliverrr, 2019-01-03:
My 2 cents... Plain ESP32 WROOM-32 without any board. Lab PSU used with adequate Amps. My problem was my (chinese?) breadboard with my (chinese?) dupont wires. Never had issues before with any of my other (non-ESP32) projects. Soldered directly 2 wires cut from my spools to 3V3 and GND (pin position 1 and 2), added the [3 lines](https://github.com/espressif/arduino-esp32/issues/863#issuecomment-347179737) to disable brownout detection, no cap used. Worked 10 out of 10 after that.
Tried without the 3 lines but with cap (either 47u or 100n) didn't work. So *my* finding is caps didn't help and the 3 lines are mandatory.

darkfader, 2019-02-05:
Just my 2 cents... I have seen that the ESP32-SOLO-1 didn't brown-out on a board where ESP32-WROOM did; both using 1 core only. Probably a combination of LDO type and low voltage (2.8V). It could also be that the WROOM was not the new version with improved layout. Adding more caps didn't help too much.

afulloa, 2019-06-11:
SOLVED, I had the same problem. I resolved with a cable with more section diameter. Check your cable and length. Thanks for yours coment.

SaurabhGirpunje, 2019-06-29:
> ### Hardware:
> Board:	ESP-WROOM-32 + breakout board
> Core Installation/update date:	8/jun/2016
> IDE name:	Platform.io
> Flash Frequency:	?40Mhz?
> Upload Speed:	115200
> 
> ### Description:
> I'm developing ESP32-based module for board which has 500mA 3v3 power supply.
> 
> After using code attached below, each time during WiFi startup I get brownout. On scope there is quite big voltage drop. This doesn't happen on a breadboard with larger capacitor attached (470uF), so you could assume that's the problem, however AT firmware from https://github.com/espressif/esp32-at works flawlessly under same electrical conditions and starts WiFi without any problems. Unfortunately this firmware is closed source, so my question is: what can I do differently to reduce init voltage drop to have same results as in AT firmware?
> 
> I've tried `esp_wifi_set_max_tx_power` but it works only after `esp_wifi_start()` but then it's too late to make any difference.
> 
> Also, at first I've tried using Arduino functions for Wifi, but they result in brownouts too.
> 
> ### Sketch:
> ```c++
> #include <Arduino.h>
> #include "esp_wifi.h"
> 
> #define SSID "ESP32AP"
> 
> void setup() {
> 	wifi_init_config_t wifiInitializationConfig = WIFI_INIT_CONFIG_DEFAULT();
> 
> 	esp_wifi_init(&wifiInitializationConfig);
> 
> 	esp_wifi_set_storage(WIFI_STORAGE_RAM);
> 
> 	esp_wifi_set_mode(WIFI_MODE_AP);
> 
> 	wifi_config_t ap_config;
> 	strcpy((char *)ap_config.ap.ssid, SSID);
> 	ap_config.ap.channel = 0;
> 	ap_config.ap.authmode = WIFI_AUTH_OPEN;
> 	ap_config.ap.ssid_hidden = 0;
> 	ap_config.ap.max_connection = 1;
> 	ap_config.ap.beacon_interval = 100;
> 
> 	esp_wifi_set_config(WIFI_IF_AP, &ap_config);
> 
> 	esp_wifi_start();
> }
> 
> void loop()
> {
> }
> ```
> 
> ### Debug Messages:
> ```
> ets Jun  8 2016 00:22:57
> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:1                         
> load:0x3fff0010,len:4                     
> load:0x3fff0014,len:588                    
> load:0x40078000,len:0                
> load:0x40078000,len:9768                          
> entry 0x400789b8
> ```

Instead of using #include "esp_wifi.h" , you should try only #include <WiFi.h> for initializing the wifi on the esp32 board.

vk1996, 2019-07-30:
I had brownout problem with ESP32 Wroom in my custom design. EN pin of ESP32 Wroom with pull up resistor 10K solved my problem.

byq13, 2019-09-20:
Do not disable brownout detector. If you need disable it, then do it like this:

`      uint32_t brown_reg_temp = READ_PERI_REG(RTC_CNTL_BROWN_OUT_REG); //save WatchDog register
       WRITE_PERI_REG(RTC_CNTL_BROWN_OUT_REG, 0); //disable brownout detector
       WiFi.mode(WIFI_MODE_STA); // turn on WiFi
       WRITE_PERI_REG(RTC_CNTL_BROWN_OUT_REG, brown_reg_temp); //enable brownout detector
`

jPerotto, 2019-12-16:
I managed to solve it like this!

https://lang-ship.com/reference/ESP-IDF/3.1.3/brownout_8c_source.html

jullrey, 2020-01-18:
> Do not disable brownout detector. If you need disable it, then do it like this:
>
> `uint32_t brown_reg_temp = READ_PERI_REG(RTC_CNTL_BROWN_OUT_REG); //save WatchDog register` [...]

byq13, is correct.  I disabled brownout detector on a Wemos Lolin32 with a Li-ion battery connected to it and the thing ran the battery down to a dangerously low level (below 2v) before it wouldn't wake up anymore.  Thank God I didn't burn down the house!

RickTracer, 2020-06-03:
Just posting this if it helps anyone else as I too struggled with WiFi startup causing brownout errors and panic attacks, even with a decent LDO. In my case I had the T5 Touch 5 (GPIO 12 MTDI which impacts boot mode and Flash voltage) hooked up as a capacitive touch sensor. If I unhooked this then the startup worked (so I just switched to the T2 GPIO 2 pin instead, could have played with setting the pin as input after wifi start but this was simplest for my case). 

Hope that helps someone!

dexop, 2020-06-04:
also having restarts on bluetooth classin level 9 power output.
disabling brownout didnt help
Solved temporarily by adding capacitors to 3v3 and Vin ,
Finaly removed diode, and everuthing works without added capacitors. on long cable.

quincisoft, 2020-08-09:
> Hey! I was facing the same issue with brownout detector restarting the ESP32.
> I'm pretty sure it might be a hardware issue, even though I have ruled out the power supply capacity, solder joints, capacitors, etc. I am thinking maybe it's some bad solder joint inside the ESP-WROOM-32 module itself. I intend to take the metal shield off and check for the same, and if nothing else, then solder the VDD3P3_RTC up to 3V3 pad on the module directly. Or maybe some other black magic.
> 
> But, until then I used this as a workaround, and thought it may be of help to you.
> `#include "soc/soc.h"`
> `#include "soc/rtc_cntl_reg.h"`
> `void setup(){`
> ` WRITE_PERI_REG(RTC_CNTL_BROWN_OUT_REG, 0); //disable brownout detector`
> ` .... other stuff ...`
> 
> Needless to say that this will disable the brownout detector completely, so if you do have power related issue you will just get GURU MEDITATION ERROR(s) of sorts. (I wonder how the name came to be so..!)
> 
> Edit: Maybe #675 could use this too.

Great. thanks

marg200x, 2020-11-08:
jullrey Sin duda alguna tu tenias la respuesta desde un principio....... Es un hecho que el ESP32 NO TIENE DEFECTOS DE FABRICACIÓN cuando el BROWNOUT DETECTOR se llega a disparar desde un principio. Se trata de un simple problema de configuración que se resuelve en el sketch PRIMERO DESACTIVAR EL BOD ANTES DE ENCENDER EL WIFI Y ACTIVAR EL BOD CUANDO EL WIFI ESTE ACTIVADO. Esto es debido a que al activar el WIFI el circuito sufre una caída de corriente mínima y después se vuelve a estabilizar, todo esto en un periodo de tiempo muy corto. Pero si el BOD esta activado desde el principio este sera disparado y llevara a RESET SOFTWARE al dispositivo ESP32. MUY SIMPLE. MicroChip explica muy bien el funcionamiento de un BOD aqui la liga https://www.youtube.com/watch?v=G90Xd2ssxog.

Burathar, 2020-11-19:
I was having the same problem, if I might add something for other people who come across this issue while looking for a solution:
- Use a proper (5V) powersupply, usb power might not be enough. Just make sure the usb power and external powersupply are both connected to ground, if you need the usb serial.
- Try adding a 100nF capacitor between 5V and ground.
- Don't use a crappy breadboad to test the ESP32 unit. I was using a cheap breadboard and kept getting brownout errors, due to unstable connections and probably weird impedance / capacitance effects.

I know this doesn't resolve the issues of the board itself, but it makes it more likely to be able to cope with them..

wuujiann, 2021-03-01:
I understand that this is old topic.  I would like to put in my two cents.
From what I've read, Brown Out Detection is triggered when voltage dropped to a certain level (e.g., 2.5V).  This could happen when BLE or WIFI, which draws a huge amount of current, is turned on.  This causes a sudden drop in voltage, leading to BOD trigger being reported.  
In my project which runs on 3x AA batteries, I needed a way to know that battery is too low to drive the WiFi to send data.  When battery level is low, I would like to sound the buzzer to let users know that battery needs to be replaced.  Below is the workaround I did to check if battery is too low to drive WiFi.

- Upon boot / reset, get the BOD count from NVRAM (Preferences library)
- If BOD count  exceeds threshold:
  * Reset BOD count to 0 and save to NVRAM.
  * Sound low battery buzzer.
  * Go to deep sleep, say... for 1 hour.
- If BOD count does not exceed threshold:
  * Increase BOD count and save to NVRAM
  * Turn on WiFi
    => If battery capacity is insufficient, this will lead to BOD 
  * Send Data via WiFi
  * Reset BOD count to 0 and save to NVRAM
  * Go to deep sleep

I know this does not prevent BOD, which is actually not avoidable.  However, it provides a way to handle BOD due to enabling WiFi/BLE.

josef2600, 2023-07-05:
i found out how to solve it!
https://github.com/josef2600/Brownout-detection-problem-in-Arduino-and-ESP32-C3

idzerve, 2023-12-21:
OMG, I was sceptical for the breakout board, but just decided to try to ommit it as I run out of ideas. 
ESP32-WROOM-32
LP3855-3.3 very low dropout voltage ( 24mV at 150 mA)
3xNiMh 1.25v/2600 mAh
Constant cycle of boot with the breakout board, tried all suggestions, big capacitor, 10K from GPIO0 and/or EN to 3.3v
Soldered directly started to work properly without any capacitor and 10k resistors.