espressif/arduino-esp32 #866: Compile error on latest commit

daytona16520, 2017-11-24:
### Hardware:
Board:							        ESP32 Dev Module
Core Installation/update date:			24/Nov/2017
IDE name:							Arduino IDE
Flash Frequency:					       80Mhz
Upload Speed:					       921600

### Description:
IDE reports compile error.

### Sketch:
Not applicable.  Error occurs with any sketch.

### Debug Messages:
Arduino: 1.8.5 (Windows 10), Board: "ESP32 Dev Module, QIO, 80MHz, 4MB (32Mb), 921600, None"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\tom00\AppData\Local\Arduino15\packages -hardware E:\OneDrive\Arduino\My Projects\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\tom00\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries E:\OneDrive\Arduino\My Projects\libraries -fqbn=espressif:esp32:esp32:FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -ide-version=10805 -build-path C:\Users\tom00\AppData\Local\Temp\arduino_build_534155 -warnings=none -build-cache C:\Users\tom00\AppData\Local\Temp\arduino_cache_589833 -prefs=build.warn_data_percentage=75 -verbose E:\OneDrive\Arduino\My Projects\ESP32\WiFi_thermometer_OLED_ESP32_v1.1\WiFi_thermometer_OLED_ESP32_v1.1.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\tom00\AppData\Local\Arduino15\packages -hardware E:\OneDrive\Arduino\My Projects\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\tom00\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries E:\OneDrive\Arduino\My Projects\libraries -fqbn=espressif:esp32:esp32:FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -ide-version=10805 -build-path C:\Users\tom00\AppData\Local\Temp\arduino_build_534155 -warnings=none -build-cache C:\Users\tom00\AppData\Local\Temp\arduino_cache_589833 -prefs=build.warn_data_percentage=75 -verbose E:\OneDrive\Arduino\My Projects\ESP32\WiFi_thermometer_OLED_ESP32_v1.1\WiFi_thermometer_OLED_ESP32_v1.1.ino
Using board 'esp32' from platform in folder: E:\OneDrive\Arduino\My Projects\hardware\espressif\esp32
Using core 'esp32' from platform in folder: E:\OneDrive\Arduino\My Projects\hardware\espressif\esp32
Detecting libraries used...
"E:\OneDrive\Arduino\My Projects\hardware\espressif\esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE="mbedtls/esp_config.h" -DHAVE_CONFIG_H "-IE:\OneDrive\Arduino\My Projects\hardware\espressif\esp32/tools/sdk/include/config" "-IE:\OneDrive\Arduino\My Projects\hardware\espressif\esp32/tools/sdk/include/bluedroid" "-IE:\OneDrive\Arduino\My Projects\hardware\espressif\esp32/tools/sdk/include/app_trace" "-IE:\OneDrive\Arduino\My Projects\hardware\espressif\esp32/tools/sdk/include/app_update" "-IE:\OneDrive\Arduino\My Projects\hardware\espressif\esp32/tools/sdk/include/bootloader_support" "-IE:\OneDrive\Arduino\My Projects\hardware\espressif\esp32/tools/sdk/include/bt" "-IE:\OneDrive\Arduino\My Projects\hardware\espressif\esp32/tools/sdk/include/driver" "-IE:\OneDrive\Arduino\My Projects\hardware\espressif\esp32/tools/sdk/include/esp32" "-IE:\OneDrive\Arduino\My Projects\hardware\espressif\esp32/tools/sdk/include/esp_adc_cal" "-IE:\OneDrive\Arduino\My Projects\hardware\espressif\esp32/tools/sdk/include/ethernet" "-IE:\OneDrive\Arduino\My Projects\hardware\espressif\esp32/tools/sdk/include/fatfs" "-IE:\OneDrive\Arduino\My Projects\hardware\espressif\esp32/tools/sdk/include/freertos" "-IE:\OneDrive\Arduino\My Projects\hardware\espressif\esp32/tools/sdk/include/heap" "-IE:\OneDrive\Arduino\My Projects\hardware\espressif\esp32/tools/sdk/include/jsmn" "-IE:\OneDrive\Arduino\My Projects\hardware\espressif\esp32/tools/sdk/include/log" "-IE:\OneDrive\Arduino\My Projects\hardware\espressif\esp32/tools/sdk/include/mdns" "-IE:\OneDrive\Arduino\My Projects\hardware\espressif\esp32/tools/sdk/include/mbedtls" "-IE:\OneDrive\Arduino\My Projects\hardware\espressif\esp32/tools/sdk/include/mbedtls_port" "-IE:\OneDrive\Arduino\My Projects\hardware\espressif\esp32/tools/sdk/include/newlib" "-IE:\OneDrive\Arduino\My Projects\hardware\espressif\esp32/tools/sdk/include/nvs_flash" "-IE:\OneDrive\Arduino\My Projects\hardware\espressif\esp32/tools/sdk/include/openssl" "-IE:\OneDrive\Arduino\My Projects\hardware\espressif\esp32/tools/sdk/include/spi_flash" "-IE:\OneDrive\Arduino\My Projects\hardware\espressif\esp32/tools/sdk/include/sdmmc" "-IE:\OneDrive\Arduino\My Projects\hardware\espressif\esp32/tools/sdk/include/spiffs" "-IE:\OneDrive\Arduino\My Projects\hardware\espressif\esp32/tools/sdk/include/tcpip_adapter" "-IE:\OneDrive\Arduino\My Projects\hardware\espressif\esp32/tools/sdk/include/ulp" "-IE:\OneDrive\Arduino\My Projects\hardware\espressif\esp32/tools/sdk/include/vfs" "-IE:\OneDrive\Arduino\My Projects\hardware\espressif\esp32/tools/sdk/include/wear_levelling" "-IE:\OneDrive\Arduino\My Projects\hardware\espressif\esp32/tools/sdk/include/xtensa-debug-module" "-IE:\OneDrive\Arduino\My Projects\hardware\espressif\esp32/tools/sdk/include/console" "-IE:\OneDrive\Arduino\My Projects\hardware\espressif\esp32/tools/sdk/include/soc" "-IE:\OneDrive\Arduino\My Projects\hardware\espressif\esp32/tools/sdk/include/newlib" "-IE:\OneDrive\Arduino\My Projects\hardware\espressif\esp32/tools/sdk/include/coap" "-IE:\OneDrive\Arduino\My Projects\hardware\espressif\esp32/tools/sdk/include/wpa_supplicant" "-IE:\OneDrive\Arduino\My Projects\hardware\espressif\esp32/tools/sdk/include/expat" "-IE:\OneDrive\Arduino\My Projects\hardware\espressif\esp32/tools/sdk/include/json" "-IE:\OneDrive\Arduino\My Projects\hardware\espressif\esp32/tools/sdk/include/nghttp" "-IE:\OneDrive\Arduino\My Projects\hardware\espressif\esp32/tools/sdk/include/lwip" -std=gnu++11 -fno-exceptions -Os -g3 -Wpointer-arith -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -w -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-sign-compare -fno-rtti  -c -w -x c++ -E -CC -DF_CPU=240000000L -DARDUINO=10805 -DARDUINO_ESP32_DEV -DARDUINO_ARCH_ESP32 -DARDUINO_BOARD="ESP32_DEV"  -DESP32 -DCORE_DEBUG_LEVEL=0 "-IE:\OneDrive\Arduino\My Projects\hardware\espressif\esp32\cores\esp32" "-IE:\OneDrive\Arduino\My Projects\hardware\espressif\esp32\variants\esp32" "C:\Users\tom00\AppData\Local\Temp\arduino_build_534155\sketch\WiFi_thermometer_OLED_ESP32_v1.1.ino.cpp" -o "nul"
"E:\OneDrive\Arduino\My Projects\hardware\espressif\esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE="mbedtls/esp_config.h" -DHAVE_CONFIG_H "-IE:\OneDrive\Arduino\My Projects\hardware\espressif\esp32/tools/sdk/include/config" "-IE:\OneDrive\Arduino\My Projects\hardware\espressif\esp32/tools/sdk/include/bluedroid" "-IE:\OneDrive\Arduino\My Projects\hardware\espressif\esp32/tools/sdk/include/app_trace" "-IE:\OneDrive\Arduino\My Projects\hardware\espressif\esp32/tools/sdk/include/app_update" "-IE:\OneDrive\Arduino\My Projects\hardware\espressif\esp32/tools/sdk/include/bootloader_support" "-IE:\OneDrive\Arduino\My Projects\hardware\espressif\esp32/tools/sdk/include/bt" "-IE:\OneDrive\Arduino\My Projects\hardware\espressif\esp32/tools/sdk/include/driver" "-IE:\OneDrive\Arduino\My Projects\hardware\espressif\esp32/tools/sdk/include/esp32" "-IE:\OneDrive\Arduino\My Projects\hardware\espressif\esp32/tools/sdk/include/esp_adc_cal" "-IE:\OneDrive\Arduino\My Projects\hardware\espressif\esp32/tools/sdk/include/ethernet" "-IE:\OneDrive\Arduino\My Projects\hardware\espressif\esp32/tools/sdk/include/fatfs" "-IE:\OneDrive\Arduino\My Projects\hardware\espressif\esp32/tools/sdk/include/freertos" "-IE:\OneDrive\Arduino\My Projects\hardware\espressif\esp32/tools/sdk/include/heap" "-IE:\OneDrive\Arduino\My Projects\hardware\espressif\esp32/tools/sdk/include/jsmn" "-IE:\OneDrive\Arduino\My Projects\hardware\espressif\esp32/tools/sdk/include/log" "-IE:\OneDrive\Arduino\My Projects\hardware\espressif\esp32/tools/sdk/include/mdns" "-IE:\OneDrive\Arduino\My Projects\hardware\espressif\esp32/tools/sdk/include/mbedtls" "-IE:\OneDrive\Arduino\My Projects\hardware\espressif\esp32/tools/sdk/include/mbedtls_po [... cut ...]
Using library esp8266-oled-ssd1306-master at version 3.2.7 in folder: E:\OneDrive\Arduino\My Projects\libraries\esp8266-oled-ssd1306-master 
Using library DHT22_Universal in folder: E:\OneDrive\Arduino\My Projects\libraries\DHT22_Universal (legacy)
Using library BLE at version 0.4.7 in folder: E:\OneDrive\Arduino\My Projects\hardware\espressif\esp32\libraries\BLE 
Using library WiFiClientSecure at version 1.0 in folder: E:\OneDrive\Arduino\My Projects\hardware\espressif\esp32\libraries\WiFiClientSecure 
exit status 1
Error compiling for board ESP32 Dev Module.

CIPop, 2017-11-24:
```
Arduino: 1.8.5 (Windows 10), Board: "ESP32 Dev Module, QIO, 80MHz, 4MB (32Mb), 921600, None"

Build options changed, rebuilding all
Archiving built core (caching) in: C:\Users\crist\AppData\Local\Temp\arduino_cache_721338\core\core_espressif_esp32_esp32_FlashMode_qio,FlashFreq_80,FlashSize_4M,UploadSpeed_921600,DebugLevel_none_4c840121b7dc969297fc43efc6f7f3ea.a
c:/users/crist/documents/arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe:C:\Users\crist\Documents\Arduino\hardware\espressif\esp32/tools/sdk/lib\libidf_test.a: file format not recognized; treating as linker script

c:/users/crist/documents/arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe:C:\Users\crist\Documents\Arduino\hardware\espressif\esp32/tools/sdk/lib\libidf_test.a:1: syntax error

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board ESP32 Dev Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

The `libidf_test.a` file contains only 

```
!<arch>
```

CIPop, 2017-11-24:
/cc @me-no-dev

ifrew, 2017-11-24:
@CIPop Thanks for the quick post on this.. I was pulling what was left of my hair out after spending 10 hours trying to get wifi/ble/i2c going...a lot of moving parts and issues there..then this issue happened.. think in future I will look here first.. just assumed it was visual studio caching issues :-)

CIPop, 2017-11-24:
I found https://github.com/espressif/esp-idf/tree/e43ac33b7c592f6c9ffc45523f04719934e3ad68/components/idf_test 
Looks like the lib is used for internal testing so it should be irrelevant for Arduino.

I was able to workaround by copying another lib from the same folder and renaming it (I created a new copy of libxtensa-debug-module.a and renamed it as libidf_test.a). This is a hack and may not work for projects actually using the debug-module lib.

me-no-dev, 2017-11-24:
Done! I wonder why this did not break on my end.. I compiled several sketches and the CI build also went fine. Maybe Windows complaining only?

closed 2017-11-24 by me-no-dev

daytona16520, 2017-11-24:
@me-no-dev   Thank you for the quick fix!

Cheers.

ifrew, 2017-11-24:
Hmm..I just deleted and recloned the core esp32 git repository and I am now getting a slightly different error. I am using visual studio with visual micro add-in as my IDE.

Linking it all together ...
"D:\Documents\Arduino\hardware\espressif\esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc" -nostdlib "-LD:\Documents\Arduino\hardware\espressif\esp32/tools/sdk/lib" "-LD:\Documents\Arduino\hardware\espressif\esp32/tools/sdk/ld" -T esp32_out.ld -T esp32.common.ld -T esp32.rom.ld -T esp32.peripherals.ld -T esp32.rom.spiram_incompatible_fns.ld -u ld_include_panic_highint_hdl -u call_user_start_cpu0 -Wl,--gc-sections -Wl,-static -Wl,--undefined=uxTopUsedPriority -u __cxa_guard_dummy -Wl,--start-group "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\airwhereesp32_1_1.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\airPKT.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\airwhere_ota.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\CalcTools.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\fileOps.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\sounds.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\timeFunctions.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\TimerObject.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\WebHelper.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\wifi_client_connect.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\WiFiHelper.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\WifiTools.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\ESP8266WebServer\src\ESP8266WebServer.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\ESP8266WebServer\src\Parsing.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\fanet_stack\fmac.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\fanet_stack\sx1272.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\vario\airwhere_esp32_vario.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\vario\cct.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\vario\I2Cdev.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\vario\KalmanVario.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\vario\MahonyAHRS.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\vario\MPU6050.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\vario\MPU9250.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\vario\MS5611.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\vario\VarioAudio.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\Wire\Wire.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\WiFi\ETH.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\WiFi\WiFi.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\WiFi\WiFiAP.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\WiFi\WiFiClient.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\WiFi\WiFiGeneric.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\WiFi\WiFiMulti.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\WiFi\WiFiScan.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\WiFi\WiFiServer.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\WiFi\WiFiSTA.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\WiFi\WiFiUdp.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\Update\Updater.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\SPI\SPI.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\Preferences\Preferences.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\FS\FS.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\FS\vfs_api.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\BLE\BLE2902.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\BLE\BLEAddress.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\BLE\BLEAdvertisedDevice.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwheree [... cut ...] ip_adapter -lc_nano -lrtc -lspi_flash -lwpa2 -lesp32 -lapp_update -lnghttp -lspiffs -lespnow -lnvs_flash -lesp_adc_cal -llog -lexpat -lm -lc -lheap -lmbedtls -llwip -lnet80211 -lpthread -ljson -lstdc++ -Wl,--end-group -Wl,-EL -o "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug/airwhereesp32_1_1.ino.elf"
ld.exe: cannot find -lidf_test

Error linking for board Node32s
Build failed for project 'airwhereesp32_1_1'
 
collect2.exe*: error: ld returned 1 exit status

me-no-dev, 2017-11-24:
did you restart the IDE?