espressif/arduino-esp32 #867: Sketch doesn't run on latest build

daytona16520, 2017-11-24:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:		24/Nov/2017
IDE name:						Arduino IDE 1.8.5 under Windows 10 Home 64bit
Flash Frequency:					80Mhz
Upload Speed:					115200

### Description:
Compile and upload works, showing no errors, but the sketch doesn't run.   Tried two different boards to rule out hardware.

Is it just me or is anyone else having same problems?

### Sketch:
Not applicable.  Tried different sketches.

### Debug Messages:
Archiving built core (caching) in: C:\Users\tom00\AppData\Local\Temp\arduino_cache_624837\core\core_espressif_esp32_esp32_FlashMode_qio,FlashFreq_80,FlashSize_4M,UploadSpeed_115200,DebugLevel_none_cc8d03a82408ab89806eebc3d4cf6de4.a
Sketch uses 144193 bytes (11%) of program storage space. Maximum is 1310720 bytes.
Global variables use 11224 bytes (3%) of dynamic memory, leaving 283688 bytes for local variables. Maximum is 294912 bytes.
E:\OneDrive\Arduino\My Projects\hardware\espressif\esp32/tools/esptool.exe --chip esp32 --port COM8 --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0xe000 E:\OneDrive\Arduino\My Projects\hardware\espressif\esp32/tools/partitions/boot_app0.bin 0x1000 E:\OneDrive\Arduino\My Projects\hardware\espressif\esp32/tools/sdk/bin/bootloader_qio.bin 0x10000 C:\Users\tom00\AppData\Local\Temp\arduino_build_205099/DHT.ino.bin 0x8000 C:\Users\tom00\AppData\Local\Temp\arduino_build_205099/DHT.ino.partitions.bin 
esptool.py v2.1
Connecting........_
Chip is ESP32D0WDQ6 (revision 1)
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 8192 bytes to 47...

Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 1057.0 kbit/s)...
Hash of data verified.
Flash params set to 0x022f
Compressed 14144 bytes to 9238...

Writing at 0x00001000... (100 %)
Wrote 14144 bytes (9238 compressed) at 0x00001000 in 0.9 seconds (effective 130.5 kbit/s)...
Hash of data verified.
Compressed 145344 bytes to 81708...

Writing at 0x00010000... (20 %)
Writing at 0x00014000... (40 %)
Writing at 0x00018000... (60 %)
Writing at 0x0001c000... (80 %)
Writing at 0x00020000... (100 %)
Wrote 145344 bytes (81708 compressed) at 0x00010000 in 7.3 seconds (effective 159.9 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 122...

Writing at 0x00008000... (100 %)
Wrote 3072 bytes (122 compressed) at 0x00008000 in 0.1 seconds (effective 384.0 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting...

me-no-dev, 2017-11-24:
did you pull the commit I pushed a couple of hours ago?

daytona16520, 2017-11-24:
Certainly did.  The previous issue I had was it wouldn't compile.

Cheers.

me-no-dev, 2017-11-24:
what do you get in the console? anything?

kerikun11, 2017-11-24:
Hi:)
I have the same problem.
the boot log is 

~~~
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13088
entry 0x40078ad4
~~~

Thank you.

daytona16520, 2017-11-24:
This is from a simple sketch with single Serial.println("Hello world!");

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13088
entry 0x40078ad4

me-no-dev, 2017-11-24:
can you please pull the last change and give it a go?

daytona16520, 2017-11-24:
Hmmm ..... unfortunately, it's still no go.

BTW, I did a clean install by deleting the esp32 folder first and re-cloning.

Cheers.

me-no-dev, 2017-11-24:
hmm.. I went through all possible flags and changes since last lib build. this is the only thing I found. Got the new toolchain right?
I'll keep  going through my modules hoping to find one that will fail

daytona16520, 2017-11-24:
Yep.  I did run get.exe after cloning.

me-no-dev, 2017-11-24:
can you please try flash freq 40Mhz? I notice that I get that issue when booting at 80Mhz

daytona16520, 2017-11-24:
Eureka!  Confirmed working at 40MHz flash frequency.

I've always used 80MHz before and never had a problem so it's very strange to me, but I'm no expert on this issue.

I guess you'll find a solution for this eventually and upload a new commit.

Thanks for your prompt attention and all the hard work.  Much appreciated.

Cheers.

me-no-dev, 2017-11-24:
that is sad though... means that some things need separate builds for different flash frequencies. I currently support 2 bootloaders and something like that would make the need for 4 bootloaders and a couple other libs.

macrojames, 2017-11-24:
Can confirm 40MHz solves that issue. Thanks a lot.

I just recieved 3 new modules and almost got crazy because they all failed after the flashing.

me-no-dev, 2017-11-24:
OK Guys! Give it a go. All configurations should work

closed 2017-11-24 by me-no-dev

gerardwr, 2017-11-24:
With previous version using 80Mhz  the Arduino sketch uploaded OK , but did not run.

Reading this topic I tried 40Mhz and then it DID run.

Just download the latest repo. Can confirm that it now uploads AND runs also at 80Mhz.

Thx @me-no-dev

vimalswaroopj, 2020-04-25:
this is only because of bad usb cable, I've changed the cable and the problem got solved.