espressif/arduino-esp32 #868: Debian 9 installation error

4rk4ik, 2017-11-24:
$ ./get.py 
System: Linux, Info: Linux-4.9.0-4-amd64-x86_64-with-debian-9.2
Platform: x86_64-pc-linux-gnu
Downloading xtensa-esp32-elf-linux64-1.22.0-75-gbaf03c2-5.2.0.tar.gz
Done
Extracting xtensa-esp32-elf-linux64-1.22.0-75-gbaf03c2-5.2.0.tar.gz
Traceback (most recent call last):
  File "./get.py", line 138, in <module>
    get_tool(tool)
  File "./get.py", line 103, in get_tool
    unpack(local_path, '.')
  File "./get.py", line 55, in unpack
    tfile = tarfile.open(filename, 'r:gz')
  File "/usr/lib/python2.7/tarfile.py", line 1693, in open
    return func(name, filemode, fileobj, **kwargs)
  File "/usr/lib/python2.7/tarfile.py", line 1751, in gzopen
    raise ReadError("not a gzip file")
tarfile.ReadError: not a gzip file

The file seems to be corrupted as I tried with tar and also failed

$ tar xvf dist/xtensa-esp32-elf-linux64-1.22.0-75-gbaf03c2-5.2.0.tar.gz -C dist/

gzip: stdin: not in gzip format
tar: Child returned status 1
tar: Error is not recoverable: exiting now

4rk4ik, 2017-11-24:
Just to precise that copying xtensa-esp32-elf from github repo to arduino-1.8.5/hardware/espressif/esp32/tools/ allow me to compile.

maydaymiao, 2018-07-16:
Got same error on Mac, any idea?

zanderzhng, 2018-07-26:
@maydaymiao download them manually.
you will need to download 3 packages in total, and the download links are available in below link.
https://github.com/espressif/arduino-esp32/blob/8b01b9e187a99fa54226ab5f42f024b9c3fab9b8/package/package_esp32_index.template.json.

- esptool-2.3.1-macos.tar.gz
- mkspiffs-0.2.3-arduino-esp32-osx.tar.gz
- xtensa-esp32-elf-osx-1.22.0-80-g6c4433a-5.2.0.tar.gz

lbernstone, 2018-07-26:
You can now install the latest release through Arduino boards manager. https://github.com/lbernstone/arduino-esp32/blob/pr1/docs/arduino-ide/boards_manager.md

closed 2018-07-28 by copercini