espressif/arduino-esp32 #869: Linking error on latest commit

ifrew, 2017-11-24:
### Hardware:
Board:							Node32S
Core Installation/update date:			11/24/17
IDE name:						Visual Micro for Arduino (Visual Studio)
Flash Frequency:					80Mhz
Upload Speed:						921600

#866 not quite working yet for me. Couldn't reopen issue so referencing that issue and listing the new error.  Looks like its trying to link -lidf_test and can't find it. This happened after I deleted and recloned the esp32 core.

Cheers
Iain

 
Linking it all together ...
"D:\Documents\Arduino\hardware\espressif\esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc" -nostdlib "-LD:\Documents\Arduino\hardware\espressif\esp32/tools/sdk/lib" "-LD:\Documents\Arduino\hardware\espressif\esp32/tools/sdk/ld" -T esp32_out.ld -T esp32.common.ld -T esp32.rom.ld -T esp32.peripherals.ld -T esp32.rom.spiram_incompatible_fns.ld -u ld_include_panic_highint_hdl -u call_user_start_cpu0 -Wl,--gc-sections -Wl,-static -Wl,--undefined=uxTopUsedPriority -u __cxa_guard_dummy -Wl,--start-group "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\airwhereesp32_1_1.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\airPKT.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\airwhere_ota.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\CalcTools.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\fileOps.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\sounds.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\timeFunctions.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\TimerObject.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\WebHelper.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\wifi_client_connect.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\WiFiHelper.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\WifiTools.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\ESP8266WebServer\src\ESP8266WebServer.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\ESP8266WebServer\src\Parsing.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\fanet_stack\fmac.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\fanet_stack\sx1272.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\vario\airwhere_esp32_vario.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\vario\cct.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\vario\I2Cdev.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\vario\KalmanVario.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\vario\MahonyAHRS.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\vario\MPU6050.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\vario\MPU9250.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\vario\MS5611.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\vario\VarioAudio.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\Wire\Wire.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\WiFi\ETH.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\WiFi\WiFi.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\WiFi\WiFiAP.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\WiFi\WiFiClient.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\WiFi\WiFiGeneric.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\WiFi\WiFiMulti.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\WiFi\WiFiScan.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\WiFi\WiFiServer.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\WiFi\WiFiSTA.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\WiFi\WiFiUdp.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\Update\Updater.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\SPI\SPI.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\Preferences\Preferences.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\FS\FS.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\FS\vfs_api.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\BLE\BLE2902.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\BLE\BLEAddress.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\BLE\BLEAdvertisedDevice.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\BLE\BLEAdvertising.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\BLE\BLECharacteristic.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\BLE\BLECharacteristicMap.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\BLE\BLEClient.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\BLE\BLEDescriptor.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\BLE\BLEDescriptorMap.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\BLE\BLEDevice.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\BLE\BLERemoteCharacteristic.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\BLE\BLERemoteDescriptor.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\BLE\BLERemoteService.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\BLE\BLEScan.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\BLE\BLEServer.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\BLE\BLEService.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\BLE\BLEServiceMap.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\BLE\BLEUtils.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\BLE\BLEUUID.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\BLE\BLEValue.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\BLE\FreeRTOS.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug\BLE\GeneralUtils.cpp.o" "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug/core.a" -lgcc -lopenssl -lbtdm_app -lfatfs -lwps -lcoexist -lidf_test -lwear_levelling -lhal -lnewlib -ldriver -lbootloader_support -lpp -lsmartconfig -ljsmn -lwpa -lethernet -lphy -lapp_trace -lconsole -lulp -lwpa_supplicant -lfreertos -lbt -lmicro-ecc -lcxx -lxtensa-debug-module -lmdns -lvfs -lsoc -lcore -lsdmmc -lcoap -ltcpip_adapter -lc_nano -lrtc -lspi_flash -lwpa2 -lesp32 -lapp_update -lnghttp -lspiffs -lespnow -lnvs_flash -lesp_adc_cal -llog -lexpat -lm -lc -lheap -lmbedtls -llwip -lnet80211 -lpthread -ljson -lstdc++ -Wl,--end-group -Wl,-EL -o "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhereesp32_1_1\espressif_node32s\Debug/airwhereesp32_1_1.ino.elf"
ld.exe: cannot find -lidf_test

Error linking for board Node32s
Build failed for project 'airwhereesp32_1_1'
 
collect2.exe*: error: ld returned 1 exit status

me-no-dev, 2017-11-24:
you did not restart the IDE ;) old board config with new sdk files

closed 2017-11-24 by me-no-dev

ifrew, 2017-11-24:
Many thanks! Yes I copied my old board.txt file into the tree rather than edit the new one to increase file upload size! Lesson learnt! cheers!