espressif/arduino-esp32 #870: O

extreme-sysadminning, 2017-11-24:
Please fill the info fields, it helps to get you faster support ;)

If you have a Guru Meditation Error, please decode it:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?ESP32 Dev Module?
Core Installation/update date:			?11/jul/2017?
IDE name:							?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
Upload Speed:						?115200?

### Description:
Describe your problem here

### Sketch:
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

closed 2017-11-24 by me-no-dev

extreme-sysadminning, 2017-11-24:
Did i just open a bug with just an ,,O'' while fighting with my lockscreen, only to find that i typed into the wrong keyboard?

Let's never talk about this again. Nothing happened, ok?

stickbreaker, 2017-11-24:
@extreme-sysadminning No once, but Twice!