espressif/arduino-esp32 #872: BLE exception during init - esp core 32 build11/24/17

ifrew, 2017-11-24:
I have been trying to get wifi and ble working together unsuccessfully. I have been following the threads about the issues closely trying hard to keep in step. Many moving parts! I spent 11 hours yesterday making progress to overcome the heap issues that nkolban reported (I'm a Glaswegian too btw!) by downloading the latest esp32 core and using the latest build from your ble code yesterday However, that brought in new I2C issues that I just resolved now by downloading the esp32 from this morning (Seattle PST) again and using new wire and hal-i2c files but nowusing the BLE that came with the core this morning.

So I re-enabled my ble  code and immediately I get an exception when calling init. exception decoding gives this information.  Is this another stack issue of not enough memory?

0x4008bfe8: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./panic.c line 572
0x4008c0e7: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./panic.c line 572
0x40088f1f: xQueueCreateCountingSemaphore at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./queue.c line 1995
0x400f27e9: osi_sem_new at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/osi/semaphore.c line 31
0x4012dcdc: fixed_queue_new at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/osi/fixed_queue.c line 56
0x40100c05: btm_ble_init at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/stack/btm/btm_ble_gap.c line 3638
0x4013100b: btu_init_core at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/stack/btu/btu_init.c line 102
0x4010bcc7: btu_task_start_up at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/stack/btu/btu_task.c line 559
0x4010bd07: btu_task_thread_handler at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/stack/btu/btu_task.c line 559

### Hardware:
Board:							Node esp32S
Core Installation/update date:			11/24/17
IDE name:						Visual Micro(Visual Studio) built on top of Arduino 1.8.4
Flash Frequency:					80Mhz
Upload Speed:						921600

daytona16520, 2017-11-24:
Hi,

I'm also using BLE and WiFi at the same time and I have no issues.

Have you changed the partition size?

The BLE libraries do use a lot of program memory so the default APP partition size of approx. 1.3M may not be enough.  I set my partition to 2M to be on the safe side.

Cheers.

daytona16520, 2017-11-25:
OK.  I just changed my code to add WiFi.begin() just before calling BLEDevice::init("myDevice") and it still works fine for me.

ifrew, 2017-11-25:
Yes. I have over 2MB in my partition but my code is only 1.35MB. I have a simple example of enabling wifi and ble that work together with no errors.. So I'm expecting it is due to other issues. I am using spi and i2c at the same time. 

Just hoping someone with knowledge of IDF components can comment since the failure is in there. Wasn't sure whether to post in the idf forum or here.

ifrew, 2017-11-25:
Created an idf issue too at https://github.com/espressif/esp-idf/issues/1315

nkolban, 2017-11-25:
Do you have a minimal sketch that we can look at to try and recreate the issue?  If so, maybe post a pastebin link to it?    I'd also try and build the app using the latest Arduino IDE and using the latest releases of the Arduino-ESP32 libraries.  This will give us a base line that is common to all that we can work against.

ifrew, 2017-11-25:
Unfortunately I can't get it to replicate in a minimal sketch yet. I have all the latest esp32 and core libraries since yesterday. I can move up to Arduino ide 1.8.5 from 1.8.4 and see what happens. Hoping someone in IDF land might know what could cause an exception making the xQueueCreateCountingSemaphore call.

Just wondered if you had came across it. What's frustrating Neil is that my project code had started to run with the esp32 core from Oct 19th using your ble library but then failed with the heap errors. Had to move to the latest esp32 core to overcome the heap allocation limit you found. However fixing that by using the new core brought in i2c issues. Got that fixed and now just can't get ble to init.

I'll move up to latest Arduino build and see what happens. Otherwise I'm up to date.

Thanks for replying!

ifrew, 2017-11-25:
Ok its a RAM issue. I traced the code back to figure out a MALLOC was failing. So I went to see if I could release more RAM in the BLEDevice.cpp file. And there I found a commented out statement made ny @nkolban 

//esp_bt_controller_mem_release(ESP_BT_MODE_CLASSIC_BT);  //FIXME waiting for response from esp-idf issue

I uncommented this as I believe IDF fixed the issue nkolban was referring to and sure enough it gets past this exception now, the ble server starts up and waits  but fails later. So maybe another ram issue

0x4008b321: uxListRemove at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./list.c line 214
0x4008a6a2: xTaskRemoveFromEventList at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./tasks.c line 3485
0x40088fed: xQueueGenericSend at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./queue.c line 1995
0x4013a745: hci_hal_h4_task_post at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/hci/hci_hal_h4.c line 180
0x4013a7c6: host_recv_pkt_cb at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/hci/hci_hal_h4.c line 274
0x401a0469: vhci_send at ?? line ?
0x4019a87a: r_eif_send at ?? line ?
0x4019ae19: hci_tx_start at hci_tl.c line ?
0x4019b002: r_hci_tl_send at ?? line ?
0x4019eea2: r_llm_con_req_ind at ?? line ?
0x4019f725: r_llm_pdu_defer at ?? line ?

me-no-dev, 2017-11-25:
how much heap do you have after you boot BLE?

nkolban, 2017-11-26:
Apparently we have some duplicate issues ... I'm following what appears to be the same issue here:

https://github.com/nkolban/esp32-snippets/issues/227

ifrew, 2017-11-26:
Sorry for having two, issues open regards the same problem. I can close this one or the one in esp32-snippets. I wasn't sue if it was a core issue or nkolban ble class issue. Which one do you want left open?

What's the best way to find out how much heap is left?

ifrew, 2017-11-26:
Printed out heap availability at beginning of setup, before BLE and after BLE. The code I have can run for a while then reboots at same place each time, or some times it reboots quickly after ble is started and waiting a connection but always at the same place, ie at the last stack trace above 

here is my trace log..
https://pastebin.com/tyj55sJu

nkolban, 2017-11-26:
When working in Arduino ESP32, when we get a back trace, we can decode the source that through the exception using the recipe found here:

https://github.com/me-no-dev/EspExceptionDecoder

have a read at that story, get that component installed, recreate the failure and then plugin the back trace into the tool to determine where in the source the exception was thrown.

ifrew, 2017-11-26:
Already done that @nkolban its in this thread a few posts back.. But here it is again

0x4008b321: uxListRemove at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./list.c line 214
0x4008a6a2: xTaskRemoveFromEventList at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./tasks.c line 3485
0x40088fed: xQueueGenericSend at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./queue.c line 1995
0x4013a745: hci_hal_h4_task_post at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/hci/hci_hal_h4.c line 180
0x4013a7c6: host_recv_pkt_cb at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/hci/hci_hal_h4.c line 274
0x401a0469: vhci_send at ?? line ?
0x4019a87a: r_eif_send at ?? line ?
0x4019ae19: hci_tx_start at hci_tl.c line ?
0x4019b002: r_hci_tl_send at ?? line ?
0x4019eea2: r_llm_con_req_ind at ?? line ?
0x4019f725: r_llm_pdu_defer at ?? line ?

ifrew, 2017-11-27:
I'm going to close this issue and use https://github.com/nkolban/esp32-snippets/issues/227 as the main working thread as I have now discovered its not due to a ram issue but there is a crash in the bluedroid scheduler when an event is received.

closed 2017-11-27 by ifrew

hanadjerbi1, 2018-08-14:
https://github.com/espressif/arduino-esp32/issues/1756