espressif/arduino-esp32 #876: Please change "this" in guide to something specific

mindfuucker, 2017-11-25:
Hello, 

I tried installing the Arduino ESP32 but could not get the boards working. 
The windows installation instructions were faulty. I

It stated that:

> Target Directory: _C:/Users/[YOUR_USER_NAME]/Documents/Arduino/hardware/espressif/esp32_
> 
> Change this to your Sketchbook Location if you have a different directory listed underneath the "Sketchbook location" in Arduino preferences.

So I assumed **this** to be C:/Users/[YOUR_USER_NAME]/Documents/Arduino/hardware/espressif/esp32. **this** did not work so I changed it to _C:/Users/[YOUR_USER_NAME]/Documents/Arduino_

Please update **this** because it took me about an hour to figure out.

me-no-dev, 2017-11-25:
wait a second... where did you install the core? What version IDE? All custom hardware definitions (not through package manager) can be on two places:
- [SKETCHBOOK_DIR]/hardware/[manufacturer]/[chip]
- [ARDUINO_INSTALL_DIR]/hardware/[manufacturer]/[chip]

The first one is preferred, because it's independent from the IDE installation

mindfuucker, 2017-11-26:
- IDE version: 1.8.4 and tried 1.8.3 when the newest failed.
- Core folder is in C:\Users\[user]\Documents\Arduino\hardware\espressif\esp32

me-no-dev, 2017-11-28:
oh god :D ok

closed 2017-11-28 by me-no-dev