espressif/arduino-esp32 #880: New version of Espressif Docs missing I2C Fifo

stickbreaker, 2017-11-26:
The latest release of the [ESP32 Hardware Reference](https://espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf) had gutted most references for the I2C Fifo. 

* fifo_conf(0x0018) register is missing bits 0..9
    * 0..4 rx_fifo_full_thrd
    * 5..9 tx_fifo_empty_thrd
* fifo_data(0x001C) is totally missing
* The Int_*(0x0020..0x002C) registers are missing  0,1,2,4
    * 0 rx_fifo_full
    * 1 tx_fifo_empty
    * 2 rx_fifo_ovf
    * 4 slave_tran_comp

Is this a purposeful removal?  Is the next silicon going to remove fifo hardware?

Is this the beginning of the end of I2C support? 

Should I stop wasting time on My I2C rewrite?

Chuck.

me-no-dev, 2017-11-26:
certainly not ;) I do not know what those are removed, will check, but there is nothing to worry about

stickbreaker, 2017-11-26:
I realize Oriental mindset places high value on 'face', never admitting wrong, but hiding problems (UART FIFO) does not fix things.  Publish the know faults and how to work around them. Leaving UXO's (unexploded ordinance) lying around does not make a safe work environment.

Chuck.

me-no-dev, 2017-11-26:
Don't overreact now :) it's probably a mistake or something ;) I am sure nobody meant to hide anything.

stickbreaker, 2017-11-28:
vent off

closed 2017-11-28 by stickbreaker