espressif/arduino-esp32 #882: how to made a web server sleep and wake up

skyladle, 2017-11-27:
Please fill the info fields, it helps to get you faster support ;)

If you have a Guru Meditation Error, please decode it:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?ESP32 Dev Module?
Core Installation/update date:			?Nov/2017?
IDE name:							?Arduino IDE?   component?
Flash Frequency:					?40Mhz?
Upload Speed:						?115200?

### Description:
how to let a esp32 mini web server enter sleep mode, and wake up if there is a client issue a request.
a mini web server power consumption is about 0.15A, it is not good for alkaline battery application. 
thanks.

copercini, 2017-11-27:
For now, you can use [modem sleep mode](https://github.com/espressif/esp-idf/tree/master/examples/wifi/power_save), that keep connected to the AP using DTIM beacon mechanism (basically wifi is turned off and wake up every 100ms to 1000ms to check if it has new packets), or depending on the case, may use a button or wake up on LAN (ethernet cable only) can archive more aggressive energy saves

In the future, there is IEEE 802.11ba with Wake-Up Radio that looks like a good solution in these cases

skyladle, 2017-11-29:
thansk,I simply insert  esp_wifi_set_ps(WIFI_PS_MODEM); in setup(), the current reading toggle between 0.15A and 0.06A. is it all?

#include "WiFi.h"
#include "esp_wifi.h"

const char* ssid="OTMEME";
const char* password="******";
WiFiServer server(80);

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  WiFi.begin(ssid,password);
  while(WiFi.status() !=WL_CONNECTED){
      delay(500);
      Serial.print(".");      
  }
  Serial.println(WiFi.localIP());
  server.begin();

  esp_wifi_set_ps(WIFI_PS_MODEM);
}

void loop() {
  // put your main code here, to run repeatedly:
  WiFiClient client=server.available();
  if(client){
    Serial.println("New Client.");
    String currentLine="";
    client.println("<!doctype html><html>" "<head><meta charset=\"utf-8\">" "<title>see </title>");
    client.println("</head><body>");
    client.println("<p> press F12 to see the current value of dimming</p>\n"); 
    client.println("</body></html>");
    while(client.connected()){
        if(client.available()){
          char c=client.read();
          Serial.write(c);
          if(c=='\n'){
            if(currentLine.length()==0){
              client.println("HTTP/1.1 200 OK");
              client.println("Content-type:text/html<br>");
              client.println();
              client.print("Click <a href=\"/On\">On</a> to turn on pin 5 on.<br>");
              client.print("Click <a href=\"/L\">here</a> to turn on pin 5 off.<br>");
              client.println();
              break;
            }else {
              currentLine="";              
            }
          }else if(c!='\r'){
             currentLine+=c;
          }
          
        }
    }   

  }

}

closed 2019-08-15 by stale[bot]